z88dk/z88dk #2566: Unexpected z80asm macro formal parameter behavior?

ceharris, 2024-06-12:
I'm trying to write a macro that happens to include an expansion of another macro with formal parameters of the same name. 
Here's a simplified example of what I'm trying to do.

```
	macro setTOS x, y
	ld C, y
	ld B, x
	endm

	macro useSetTOS x, y
 	setTOS x, y
	endm

	org 0
	useSetTOS H, L
```
Assembling this example, I get these errors.
```
$ z88dk-z80asm macro-example1.asm
macro-example1.asm:13: error: undefined symbol: y
  ^---- y
macro-example1.asm:13: error: undefined symbol: x
  ^---- x
```

After experimenting a bit, I realized that if the formal parameter names used by the two macros are not the same, it works as expected, producing the expected machine code.
```
        macro setTOS x1, y1
        ld C, y1
        ld B, x1
        endm

        macro useSetTOS x, y
        setTOS x, y
        endm

        org 0
        useSetTOS H, L
```

Is this expected behavior? It seems like an odd limitation, and it makes it difficult to convert code for other assemblers that apparently don't have this issue.

suborb, 2024-06-23:
That does sound undesirable, I think @pauloscustodio is busy offline at the moment so please be patient!

pauloscustodio, 2024-06-24:
Definitively a bug. I'll look into it.

pauloscustodio, 2024-06-24:
Bug solved, thanks for reporting.

closed 2024-06-24 by pauloscustodio

ceharris, 2024-06-26:
Confirmed solved. Thanks for the quick turnaround!

commit 3f05c420 (2024-06-24): Fix #2566: error in expansion of nested macros with same parameter names

The preprocessor was creating a new layer of macro expansion and in that
layer was defining the arguments as new macros. This worked well,
except when macro A has an argument x, and calls macro B that also has
an argument x.

Now a textual replacement of all arguments by the parameters is made
before creating a new layer of macro expansion.