z88dk/z88dk #2569: 8085 - am9511 - planets

feilipu, 2024-06-14:
Tracking a strange runtime error.

I'm using the [planet motion](https://github.com/feilipu/planets) repository, which is used for testing different math libraries.

The code is working with other math libraries, and particularly with math32, but not in the combination of 8085 and am9511.

Testing on a real 8085 and am9511 produces three lines of output, but then it jumps off somewhere and doesn't return.
It is very odd that three iterations of calculations are done, and only then it leads jumps off into oblivion.
A [correct run](https://github.com/feilipu/planets/blob/main/planetapu.txt) looks like this.

```sh
B>planet85

Geocentric Coordinates

Solar Day 7855.000
       Sun x  -0.199427 y   0.996939 z   0.000000
      Moon x   0.002463 y   0.001046 z  -0.000174
   Mercury x   0.137410 y   0.790757 z  -0.047798
```
The interesting thing is that this real life behaviour is replicable with `z88dk-ticks`.

```sh
    zcc +test -clib=8085 -v -m --list -O2 -DAMALLOC -DPRINTF --am9511 @planets.lst -o planet85_ticks.bin
    z88dk-ticks -m8085 -counter 99999999999 planet85_ticks.bin
```

Also produces...

```sh
host:~/Z80/src/planets$ z88dk-ticks -m8085 -counter 99999999999 -x planet85_ticks.map planet85_ticks.bin

Geocentric Coordinates

Solar Day 7855.000
       Sun x  -0.199435 y   0.996937 z   0.000000
      Moon x   0.002463 y   0.001046 z  -0.000174
   Mercury x   0.137401 y   0.790755 z  -0.047798

```
The same result.

Currently trying to debug where it is going wrong. Thoughts welcome.

feilipu, 2024-06-14:
I think there's something suspicious in the comparative operators, specifically `l_f32_ge`.
Everything otherwise looks normal.

feilipu, 2024-06-14:
This is a bit subtle. The [8085 Am9511 math test](https://github.com/z88dk/z88dk/blob/master/test/suites/math/Makefile#L53) results are correct, and adding more test variations catches nothing.

suborb, 2024-06-14:
I guess it's good that it's real hardware as well as ticks showing the behaviour! 

Eliminating things:

* Compiling as if 8080
* Applying 8080 opt rules only
* Not applying any rules
* Linking 8080 test_clib and 8080_crt instead of the 8085 variants

So it's in the am9511 library somewhere.

I've not looked at the source code, but I'm assuming it's just looping for convergence and the failed comparison means that the loop doesn't break? - Normally when this sort of thing happens ticks -trace reports a load of nops, but here it's genuine code running.

suborb, 2024-06-14:
Z80:

```
Geocentric Coordinates

Solar Day 7855.000
       Sun x  -0.199435 y   0.996937 z   0.000000
Error 1.870611
Error 0.000994
      Moon x   0.002463 y   0.001046 z  -0.000174
Error 11.705850
Error 0.238823
Error 0.000119
   Mercury x   0.137401 y   0.790755 z  -0.047798
Error 0.207069
Error 0.000000
     Venus x  -0.893031 y   1.182404 z   0.042568
```

8085:

```
Geocentric Coordinates

Solar Day 7855.000
       Sun x  -0.199435 y   0.996937 z   0.000000
Error 1.870611
Error 0.000994
      Moon x   0.002463 y   0.001046 z  -0.000174
Error 11.705850
Error 0.238823
Error 0.000119
   Mercury x   0.137401 y   0.790755 z  -0.047798
Error 0.207069
Error 0.000000
```

suborb, 2024-06-14:
```
#include <stdio.h>

int main()
{
        float x = 0.0;

        if ( x >= 1.0e-3 ) {
                printf("Fail\n");
        }
}
```

z80:

```
z88dk-ticks a.bin

Ticks: 7121
```

8085:

```
z88dk-ticks -m8085  a.bin
Fail

Ticks: 14530
```

Might be worth checking the zero comparison path?

feilipu, 2024-06-15:
> For the return from `l_f32_ge`, on the z80, the zero flag is reset, on 8085 it isn't. The `sla e`and `rl d` in `zero_left` clears the flag. Very subtle.

Yes. Very subtle. Obvious once you know. But anyway... fixed now I hope.

commit f34e8894 (2024-06-15): am9511 - comparatives closes #2569
closed 2024-06-15 by feilipu