z88dk/z88dk #2580: How is ADD DE, buffer assembled?

Candid-Moe, 2024-06-24:
I'm learning Z80 and made this program to test how to code and access parts of a struct via a pointer.

```
defvars 0 {
    size     ds.b   32
    name     ds.w   0x1234
    status   ds.b  15
    }

org 0x4000

    LD DE, buffer
    ADD DE, name
    LD A, (DE)

buffer:
    db 4
    dw 0x5051
    db 3

```
When I look at the generated code, I found that `ADD DE, name` was compiled to this

`0003  e521200019545de1      ADD DE, name`

or
[image]

which make sense: `ADD DE, nn` is not a Z80 code and Z80asm translate it into several machine ops.

How does it works? What other "instructions" is Z80asm able to convert to machine codes? Where I can read more about this?

dbolli, 2024-06-24:
You could do
```
	LD DE,(buffer+name)	; DE = buffer + 1
	LD A,(DE)	; A = contents of address $5051
```
Regards,
Derek.

feilipu, 2024-06-24:
> Which make sense: ADD DE, nn is not a Z80 code and Z80asm translate it into several machine ops.
> How does it works?

The `z88dk-z80asm` assembler is both a "synthetic" assembler and a "MACRO" assembler. By that I mean it is able to synthesise a variety of different opcodes into equivalent binary assembly, as well as to translate macros into opcodes, which may or may not also be synthetic.

Usually synthetic opcodes are only created where there are no side effects generated from the code swap, such as setting flags or trashing other registers, but that is not always the case.

z88dk-z80asm does this synthetic assembly for several reasons.

1. There are multiple machine variants supported and often makes it easier to write assembly for these different variants without conditional assembly, if opcodes supported on the Z80 for example are synthesised for the GBZ80, or vice versa.

2. Avoiding stupid typos. It easy to write something like `ld h,b` `ld l,e` and have that do entirely the wrong thing, when you really meant to do `ld hl,bc` (or `ld hl,de`). By having these as synthetic 16 bit instructions it avoids silly mistakes, and makes code generally easier to read.

> What other "instructions" is Z80asm able to convert to machine codes? Where I can read more about this?

The complete list of synthetic operations is contained within the full list of opcodes supported across all variants.
The list is [here](https://raw.githubusercontent.com/z88dk/z88dk/master/src/z80asm/dev/cpu/opcodes.txt), and it is found within the [z88dk-z80asm wiki page](https://github.com/z88dk/z88dk/wiki/Tool---z80asm).

feilipu, 2024-06-26:
Please reopen if you've follow on questions.

closed 2024-06-26 by feilipu