z88dk/z88dk #2581: macOS - zcc being flagged for virus

feilipu, 2024-06-26:
Noted from yesterday's nightly.

[image]

But, I haven't checked whether things on z88dk end have changed, because I guess it is a change from the Apple.

suborb, 2024-06-26:
I probably need to accept the t&cs again.

dbolli, 2024-06-26:
I think this is due to an older macOS version. I have macOS 14.5 running on M1 Apple silicon and I downloaded and built the 2024-06-25 03:27 nightly build and I don't get this error message.

feilipu, 2024-06-26:
It's using the current (supported) version for older Macs. Since mine came out of a skip and is 2017 vintage, it is a bit older.

At least I'm not asking for it to run on my G4 Power Mac (on 10.4.11).

suborb, 2024-06-26:
I did have to accept the t&cs, as a result the notarisation failed and downloaded binaries would be untrusted:

```
#########################################################################

Notarising osx kit

#########################################################################
creating Notary API submission for z88dk-osx-20240626-3f05c420f1-22742.zip (sha256: a5845588301d05519a7703a3a89970c65fba5c15524190689288406a5d4f33df)
HTTP error from https://appstoreconnect.apple.com/notary/v2/submissions
{
  "errors": [
    {
      "code": "FORBIDDEN.REQUIRED_AGREEMENTS_MISSING_OR_EXPIRED",
      "detail": "This request requires an in-effect agreement that has not been signed or has expired.",
      "id": "PPJHHFFUGXKV2SPMORNPX25ECU",
      "links": {
        "see": "/agreements"
      },
      "status": "403",
      "title": "A required agreement is missing or has expired."
    }
  ]
}
Error: error interacting with Notary API
```

feilipu, 2024-06-27:
Resolved with `zcc - Frontend for the z88dk Cross-C Compiler - v22743-900fec2205-20240627`.

closed 2024-06-27 by feilipu