z88dk/z88dk #2586: 8085 - _z88dk_callee return improvement

feilipu, 2024-07-19:
I've been working on a function that has a lot of parameters, and when compiled as` `__z88dk_callee` and with  `-m8085 the compiler chooses to use `pop bc` to remove the dead parameters from the stack. But in `-mz80` it uses `pop af`.

Using `pop bc` means that `bc` can't be used to hold the return address, which further necessitates using `saved_hl` to store the `int` return value, while the return address is held temporarily in `hl`.

Would it be better to use `pop af` to blow off the dead parameters, as is done in the z80 code path?

```C
int collision_asteroids(struct asteroid asteroids[], uint8_t size, struct vector2d* v, FLOAT radius) __z88dk_callee;
```

8085

```
   258  1378  220000            	ld	(saved_hl),hl
   258  137b  e1                	pop	hl
   258  137c  c1                	pop	bc
   258  137d  c1                	pop	bc
   258  137e  c1                	pop	bc
   258  137f  c1                	pop	bc
   258  1380  c1                	pop	bc
   258  1381  e5                	push	hl
   258  1382  2a0000            	ld	hl,(saved_hl)
   258  1385  c9                	ret
```

z80

```
   258  1449  c1                	pop	bc
   258  144a  f1                	pop	af
   258  144b  f1                	pop	af
   258  144c  f1                	pop	af
   258  144d  f1                	pop	af
   258  144e  f1                	pop	af
   258  144f  c5                	push	bc
   258  1450  c9                	ret
```

feilipu, 2024-07-19:
Possibly, this below line could have `vartype != KIND_NONE` removed from the decision, but I'm not sure of the flow on effect.

https://github.com/z88dk/z88dk/blob/32bbfef91bb598b3ca4bf26624626ff1e5a65b3c/src/sccz80/codegen.c#L1502

suborb, 2024-07-19:
I think it's meant to be `KIND_LONG`

feilipu, 2024-07-19:
Added note to commit.
The `double` return looks suspicious.

closed 2024-07-20 by feilipu