z88dk/z88dk #2587: Best practices for creating local labels in z80asm (question)

ifilot, 2024-07-30:
Let me first of all thank this community for building and supporting such a great software suite. I have been using it with much delight.

I have a brief questions I am hoping somebody is able to offer their advice upon. Suppose I have two subroutines in a single assembly (`.asm`) file

```
subroutine1:
  ld b,2
next:
  ; do something
  djnz next
ret

subroutine2:
  ld b,2
next:
  djnz next
  ; do something
ret
```

The code above would not assemble since there is a duplicate label `next`. In some assemblers, it is possible to indicate that a label is locally scoped - typically by prepending it with a dot (`.`) such that the label belongs to the first non-local label encountered earlier. Can something like this be done in `z80asm` and what are recommended good practices? 

What I am currently doing is manually generating unique labels like `subroutine1_next` and `subroutine2_next`, which works, but seems like a inferior solution in terms of elegance.

ifilot, 2024-07-30:
@pauloscustodio Thank you for this detailed answer and quick response. Appreciated!

If you allow me to ask two follow-up questions:

1. So is it considered good practice then to place each subroutine in its own `.asm` file? It seems to me that this would result in a lot of small files and since certain subroutines are tied together in terms of functionality, it makes sense to combine them into a single file rather than store them in separate files.
2. Are there any plans to support local labels in the future or are there good reasons not to so?

pauloscustodio, 2024-07-30:
> So is it considered good practice then to place each subroutine in its own .asm file?

Not necessarily, but is is good practise to keep the module size small and self-contained, to limit the public interface and promote informarion hiding.

> Are there any plans to support local labels in the future or are there good reasons not to so?

Supporting local labels is not hard to do, but it clashes with the legacy syntax of z80asm of allowing both ```label:``` and ```.label``` to mean the same. Implementing local labels with ```.label``` would break legacy code.

suborb, 2024-07-30:
> Supporting local labels is not hard to do, but it clashes with the legacy syntax of z80asm of allowing both `label:` and `.label` to mean the same. Implementing local labels with `.label` would break legacy code.

The other way I've seen it done is with a trailing `::`

ifilot, 2024-07-30:
From my personal side, I would sincerely appreciate supporting local labels by `z80asm`. It can be in the form of two trailing colons as suggested by @suborb or perhaps using something like `@label:`, assuming this does not clash with other features of the assembler. 

I also understand that making feature requests is simple, and I recognize that developers have limited time (a valuable commodity!) or other interests. There may be more urgent or relevant issues to address and I have little insight in how much of an effort this feature request would be. Ultimately, the decision lies with the developers.

@pauloscustodio and @suborb : Thank you both for answering/commenting upon my questions.

pauloscustodio, 2024-07-30:
Clarification of requirements:
- when a label is used (e.g. ```jr next```) the assembler needs to verify if there are any local labels of that name, and if yes resolve to the one closest in terms of line numbers;
- if there are no local labels of the same name, a non-local label is assumed;
- local labels are introduced with the syntax ```@next:```

First requirement is the most difficult to implement, as the assembler is currently single-pass and the local label may appear further down in the code.

A simpler alternative would be to require that local labels, when referenced, are also prefixed with the ```@``` sign, e.g. ```jr @next```.

Any comments on the second alternative?

dbolli, 2024-07-30:
My 2 cents:

Implement #local and #endlocal

e.g. 

```
subroutine1:
#local
  ld b,2
next:
  ; do something
  djnz next
ret
#endlocal

subroutine2:
#local
  ld b,2
next:
  djnz next
  ; do something
ret
#endlocal
```
so that subroutine1: and subroutine2: are global and each occurance
of next: is local

Regards,
Derek.

pauloscustodio, 2024-07-30:
> Implement #local and #endlocal

Thanks for the suggestion @dbolli, but the assembler does not have multi-level symbol tables; at the end it has to generate one object file with a flat space of symbols. The changes to implement this are greater than the suggestion by @ifilot .

pauloscustodio, 2024-08-16:
Closed.

Local Labels
A label started with a @ is a local label, whose full name is the name of the previous non-local label, the @ sign and the local name. Local labels can be repeated in the same source file, and are resolved to the nearest definition. They can also be referenced by the full name.

    subroutine:                 ; previous non-local label
    @here:                      ; defines label subroutine@here
        jr @here                ; jumps to subroutine@here
    
    sub2:                       ; another non-local label
    @here:                      ; defines label sub2@here
        jr @here                ; jumps to sub2@here

    main:
        call subroutine@here    ; refer to a local label full name

ped7g, 2024-08-16:
just curious, is it (usage of local label) literally "to the nearest definition" like by bytes or LoCs?

IMHO that sounds open for confusing situations. In sjasmplus the local context is opened/closed by "main" labels, so any usage of local label is implicitly extended by the last main label.

From the example here it's not clear what happens when:
```
subroutine:
@here:
  rra ; some
  rra ; long
  rra ; inner-loop
  rra ; implementation
  rra ; code
   jr @here ; does this jump to subroutine@here or sub2@here ? (in sj+ it goes up)
sub:
@here:
   nop ; short inner loop
   jr @here
```
 Or in this case it's explicitly subroutine@here because it's inside that block, and the "nearest" search happens only if it would be some other main label like "innerloop:" without @here defined ahead of `jr @here` (that's error in sj+ then)?

pauloscustodio, 2024-08-16:
> just curious, is it (usage of local label) literally "to the nearest definition" like by bytes or LoCs?

Nearest definition is in lines of code. Of course there are edge cases where you dont get what intended, as the example above, but it should be fit for most purposes.

pauloscustodio, 2024-08-17:
Question to the community: 

shall we keep local labels as implemented (i.e. resolved to the nearest line of code of occurence), or shall we follow sjasmplus (i.e. local labels only exist in the space between two non-local labels)?

The way it is implemented is more flexible, but may cause unexpected results, as shown above by @ped7g.

The way of sjasmplus is more restrictive, i.e. local labels cannot repeat between non-local labels, but is more deterministic. In terms of implementation it is also easier, but the more complex implementation is already in place.

feilipu, 2024-08-17:
The way you have it currently, `GLOBAL` entry into a loop (for example), originated with a local label, does not have to be at the top of the code.

This is more flexible, and better IMHO, for `do while` style loops.

```asm
PUBLIC enter

@loop 
  ; test `while`
  ret NZ
.enter
  ;
  ; stuff, exit fail on C
  ;
  jr NC, @loop 

  ; handle failure case
  ret
```

Though, I'd probably not use local labels anyway except for the simple loop cases.

suborb, 2024-08-18:
Personally I prefer the sjasmplus way - having converted code which used the current implementation it's quite confusing to read.

pauloscustodio, 2024-08-18:
two votes in favour of sjasmplus, one vote in favour of the current implementation, I'll change it to match sjasmplus.

pauloscustodio, 2024-08-19:
Implement local labels with the same use case as sjasmplus:
a local label is prefixed with @ and is valid in a block between
two non-local labels. A local label name cannot be repeated in the
same block, but can be repeated in another block.

The local label full name is <last-non-local-label>@<local-label>.

A local label can be referred to as its short name @Label or long name
prev_non_local@label.

commit 1d375834 (2024-08-19): Fix #2587 local labels

Implement local labels with the same use case as sjasmplus:
a local label is prefixed with @ and is valid in a block between
two non-local labels. A local label name cannot be repeated in the
same block, but can be repeated in another block.

The local label full name is <last-non-local-label>@<local-label>.

A local label can be referred to as its short name @label or long name
prev_non_local@label.
closed 2024-08-19 by pauloscustodio