z88dk/z88dk #2590: (z80asm) Disabling synthetic opcodes

suborb, 2024-09-02:
z80asm makes development of z88dk libraries that work across multiple CPUs easier due to the synthetic opcodes, indeed we rely on this quite a lot for index register loading and non-performance critical emulation on the gbz80.

However, when using z80asm as a vanilla assembler these can be confusing and lead to unexpected code being silently inserted.

So, would be possible to add an option to disable/enable the synthetics @pauloscustodio ?

pauloscustodio, 2024-09-03:
Yes, it is possible, but not straitforward. I'll have to think about the best way.

feilipu, 2024-09-03:
Out of interest, is the use case to avoid calling helper functions, especially `gbz80`, that wouldn't be provided where just the assembler is in use? Or is it more related to (loose) syntax from other sources that produces wrong opcodes with `z80asm`?

Xalior, 2024-09-03:
The case I saw this come up recently is neither of those -- it was someone
"learning Z80 by doing" and was struggling to differentiate between what
was real, and what was the assembler "doing magic".

-Dx

feilipu, 2024-10-16:
For "learning by doing" I like to use the nice [z80/z180 tables by ClrHome](https://clrhome.org/table/), and the [gbz80 one by Pastraiser](https://www.pastraiser.com/cpu/gameboy/gameboy_opcodes.html).

But I've an allergy to the Intel mnemonics, so I made my own [8085 version with Zilog mnemonics](https://gitlab.com/feilipu/8085-opcodes), with all the undocumented additions fully documented.

Because I can never remember which flags work and when, I always have them open in a tab.

fitch, 2024-12-17:
If it's harder to have an option to prevent this from happening, could it be at least somehow outputted to the log with some "very verbose" logging mode, so that it can be catched. I have just this "learning Z80 by doing" use case, and I spent _a very long time_ figuring out what was going on, before I understood that synthetic opcodes are replaced with something else.

feilipu, 2024-12-17:
Just reiterating Paulo's comment above, using `-l` to list the assembled results will show a call instruction `cd0000` or a longer hex string (4 or 5 bytes) where synthetic instructions are being used. That is a fast and direct way to see this in action today.

fitch, 2024-12-17:
Yes, I agree - this is a good start, thanks!

fitch, 2024-12-19:
@feilipu The problem is not necessarily that some of the synthetic opcodes are replaced _in place_ with a set of real opcodes, but then - for example - some synthetic opcodes generate the call which can be unexpected when you can't make calls (for example, no RAM available for stack). However, it's a good question should the assembler point this out or leave it to the programmer. I would've just benefited from the assembler warning that I've used non-standard Z80 instructions in the assembly file without an intention to do so.

feilipu, 2024-12-19:
If you look through the [generated opcode list](https://raw.githubusercontent.com/z88dk/z88dk/master/src/z80asm/dev/cpu/opcodes.txt), searching for `cd `, most examples of a `call` are for deficiencies of 8080, 8085, or GBZ80 with respect to the Z80. These are important to enable writing generalised libraries without constant conditional assembly.

However there are some `call` examples that intend to create an opcode that is not present in the base family at all, like `add hl,a` for example, and they may have been added before z88dk-z80asm had macro features as an aid.

(Noting`add hl,a` is found in the Spectrum Next z80n, but I'd guess it is never used in any application otherwise).

🤔 Probably these `call` opcodes could (depending on a check of the code base first) just be pruned out entirely, along with their helper functions. As a thought...

pauloscustodio, 2024-12-19:
These additional opcodes were created to simulate on a z80 opcodes of the z80n.

feilipu, 2024-12-19:
Yes. But considering that the eZ80 and kc160 also have many opcodes that are not "backported" to Z80, is it really relevant to do backporting just for the z80n opcodes?

Just raising the question, as these extensions using `call`  instructions do require correct stack to work and seem to be the main concern.

feilipu, 2024-12-20:
Looking through the [z80asm helper functions](https://github.com/z88dk/z88dk/tree/master/src/z80asm/dev/z80asm_lib), it seems that mainly the z80n `add nn,a` instructions are relevant as a trap for z80 new users, in that they're not available for any "original" CPU. They're not found in any library functions they so could be an easy consistency prune.

I guess the `sub hl,nn` instructions could also be a trap as only the 8085 undocumented `sub hl,bc` is a real opcode (which obviously should remain for the 8085), except on r4k. So these instructions could also be easily confusing for the z80 new users too, and I'd be in favour of removing them too.

fitch, 2024-12-20:
Yes, @feilipu I think you got it right: it's the `add hl, a` that got me lost when it was replaced by a `call`. But as per using `-l` I've now created a makefile step to check that none of the instructions are being replaced with `cd0000` to prevent this error happening again. :)

pauloscustodio, 2024-12-20:
> Yes, it is possible, but not straightforward. I'll have to think about the best way.

The change is under way, I will create a -strict option to z80asm to disable all undocumented opcodes and all synthetic instructions.

feilipu, 2024-12-20:
> a `-strict` option to z80asm to disable all undocumented opcodes and all synthetic instructions.

While a good option and the original request. I fear that anyone who has not RTFM (or learned enough) to know about undocumented opcodes will also not find the `-strict` command line option.

My Local Council sometimes just grinds the edges of concrete pathways to prevent tripping falls. Much cheaper than digging up and replacing the whole pavement. Perhaps the same applies here?

pauloscustodio, 2024-12-31:
> While a good option and the original request, I fear that anyone who has not RTFM (or learned enough) to know about synthetic opcodes will also not find the -strict command line option.

I do like the -strict option because is allows the use of ```cp``` and ```jp``` in their original meaning with ```-m8080 -strict```.

feilipu, 2025-01-01:
> I do like the -strict option because it allows the use of `cp` and `jp` in their original meaning with `-m8080 -strict`.

Yes. Same applies to `-m8085` too. A very good thing for antique code support.

Still it would be advantageous for new users to make the simple traps (z80n `add` and 8085 `sub`) automatically fail on the z80 default, I believe.

pauloscustodio, 2025-05-01:
Closed through pull request #2707

feilipu, 2025-05-01:
I notice that the 8085 Extended Opcodes are not included in the 8085 "strict" opcode set.

The Extended Opcodes are documented in Table 2.10 of the [Tundra Semiconductor 8000 Series data sheet](https://github.com/z88dk/techdocs/blob/master/intel/Tundra%20Semiconductor%208000%20Series%20Components%20Manual.pdf) and therefore they are "strictly correct" documented 8085 opcodes, as their products are licensed from Intel.

The 10 Extended Opcodes should be included in the same code class as the `RIM` and `SIM` opcodes.

pauloscustodio, 2025-05-02:
> The 10 Extended 8085 Opcodes should be included in the same code class as the RIM and SIM opcodes.

Done.

commit 45e3021d (2025-05-02): Fix #2590: 8085: The 10 Extended 8085 Opcodes should be included in the same code class as the RIM and SIM opcodes.
closed 2025-05-02 by pauloscustodio