z88dk/z88dk #2591: +msx -subtype=rom --no-crt works incorrectly

avysk, 2024-09-10:
If I create `test.asm` with the following content

```
loop: jr loop
```

and compile it as `zcc +msx -subtype=rom --no-crt test.asm -o test.rom` I am not getting the correct MSX ROM cartridge file: it is missing ROM header.

If I modify the `test.asm` file to read

```
global _main
_main: jr _main
```

and compile as above, but without `--no-crt`, the ROM header is there. Unfortunately, in this case, the result is bloated (168 bytes instead of 16 bytes header + instruction for `jr`).

suborb, 2024-09-10:
That seems like the correct behaviour. The ROM header is in the crt0 file.

Re size, see here: https://github.com/z88dk/z88dk/wiki/Classic--Homebrew#why-is-the-generated-executable-so-big and here: https://github.com/z88dk/z88dk/wiki/Classic--Pragmas I suspect the relevant bit might be here: https://github.com/z88dk/z88dk/wiki/Classic--Pragmas#startup-behaviour

avysk, 2024-09-10:
So, it does not contradict the title of the issue. The combination of `-subtype=rom --no-crt` does not work.

avysk, 2024-09-10:
I have added to the source

```
#pragma define CRT_INITIALIZE_BSS = 0
#pragma define CLIB_EXIT_STACK_SIZE = 0
#pragma define CRT_ENABLE_STDIO = 0
#pragma define CRT_FOPEN_MAX = 0
#pragma define CRT_OPEN_MAX = 0
#pragma define CRT_ENABLE_COMMANDLINE = 0
#pragma define CRT_COMMANDLINE_REDIRECTION = 0
#pragma define CRT_ENABLE_EIDI = 0
```

and I am still getting 168 bytes with crt. What am I doing wrong?

avysk, 2024-09-10:
(Sorry for the multiple add/delete comment -- there was some strange problem with github.)

suborb, 2024-09-10:
So for size, I've made a few minor changes, I've got these files:

```
small.asm

SECTION code_user

global _main

_main:  jr _main
```

```
zpragma.inc

#pragma define CRT_INITIALIZE_BSS = 0
#pragma define CLIB_EXIT_STACK_SIZE = 0
#pragma define CRT_ENABLE_STDIO = 0
#pragma define CRT_FOPEN_MAX = 0
#pragma define CRT_OPEN_MAX = 0
#pragma define CRT_ENABLE_COMMANDLINE = 0
#pragma define CRT_COMMANDLINE_REDIRECTION = 0
#pragma define CRT_ENABLE_EIDI = 0

#pragma export fputc_cons = 0
```

Compiling with:

```
zcc +msx -subtype=rom -m small.asm -pragma-include:zpragma.inc -create-app
```

We can then disassemble the ROM file:

```
start:

;;;;; ROM fixing
                    di                                      ;[4010] f3
                    ld        sp,($fc4a)                    ;[4011] ed 7b 4a fc
                    call      $0138                         ;[4015] cd 38 01
                    rrca                                    ;[4018] 0f
                    rrca                                    ;[4019] 0f
                    and       $03                           ;[401a] e6 03
                    ld        c,a                           ;[401c] 4f
                    ld        b,$00                         ;[401d] 06 00
                    ld        hl,$fcc1                      ;[401f] 21 c1 fc
                    add       hl,bc                         ;[4022] 09
                    ld        c,a                           ;[4023] 4f
                    ld        a,(hl)                        ;[4024] 7e
                    and       $80                           ;[4025] e6 80
                    or        c                             ;[4027] b1
                    ld        c,a                           ;[4028] 4f
                    inc       hl                            ;[4029] 23
                    inc       hl                            ;[402a] 23
                    inc       hl                            ;[402b] 23
                    inc       hl                            ;[402c] 23
                    ld        a,(hl)                        ;[402d] 7e
                    and       $0c                           ;[402e] e6 0c
                    or        c                             ;[4030] b1
                    ld        h,$80                         ;[4031] 26 80
                    call      $0024                         ;[4033] cd 24 00

;;;;; Copies data section into RAM
                    call      crt0_init                     ;[4036] cd 43 40

;;;;;  Call main
                    call      _main                         ;[4039] cd 59 40

;;;;; Cleanup anything that needs to be cleaned up
__Exit:
                    call      crt0_exit                     ;[403c] cd 4f 40

;;;;; Terminate behaviour
crt_exit_loop_forever:
                    halt                                    ;[403f] 76
                    jr        crt_exit_loop_forever         ;[4040] 18 fd
l_dcal:
                    jp        (hl)                          ;[4042] e9

;;;;; Copy data section
crt0_init:
                    ld        hl,$405b                      ;[4043] 21 5b 40
                    ld        de,$c003                      ;[4046] 11 03 c0
                    ld        bc,$0001                      ;[4049] 01 01 00
                    ldir                                    ;[404c] ed b0
                    ret                                     ;[404e] c9

crt0_exit:
                    ret                                     ;[404f] c9

;;;;; Hook to call into the bios
msxbios:
                    ld        iy,($fcc0)                    ;[4050] fd 2a c0 fc
                    call      $001c                         ;[4054] cd 1c 00
                    ei                                      ;[4057] fb
                    ret                                     ;[4058] c9

_main:
                    jr        _main                         ;[4059] 18 fe
                    nop                                     ;[405b] 00
```

You can of course use -crt0=[file.asm] to specify your own crt0 file.

closed 2025-01-31 by suborb