z88dk/z88dk #2592: +msx -subtype=rom produces files of the wrong size

avysk, 2024-09-10:
MSX ROM cartrige files should be 8k in size; z88dk produces unusable output as it is not padded to the right size.

`zcc +msx -subtype=rom ... -o tmp.rom` -- the result in unusable. Workaround is

```
dd if=/dev/zero of=out.rom bs=8k count=1
dd if=tmp.rom of=out.rom conv=notrunc
```

Now `out.rom` is 8k file (`tmp.rom` padded with zeroes in the end) and is perfectly usable as MSX ROM file.

suborb, 2024-09-10:
The final creation of the rom is performed by -create-app. This stage also concatenates the data section onto the code so that the ROM will run correctly.

avysk, 2024-09-10:
I have added to the source

```
#pragma define CRT_INITIALIZE_BSS = 0
#pragma define CLIB_EXIT_STACK_SIZE = 0
#pragma define CRT_ENABLE_STDIO = 0
#pragma define CRT_FOPEN_MAX = 0
#pragma define CRT_OPEN_MAX = 0
#pragma define CRT_ENABLE_COMMANDLINE = 0
#pragma define CRT_COMMANDLINE_REDIRECTION = 0
#pragma define CRT_ENABLE_EIDI = 0
```

and I am still getting 168 bytes with crt. What am I doing wrong?

avysk, 2024-09-10:
I believe that `-create-app` indeed fixes the things. The fault was on my end.

closed 2024-09-10 by avysk