z88dk/z88dk #2597: Mattel Aquarius vs. Aquarius+ console input/output

iratahack, 2024-09-19:
The existing _fputc_cons_native_ and _fgetc_cons_native_ routines rely on calling routines in the original aquarius ROM. This works fine with the Aquarius+ until banking is enabled. When banking is enable, the ROM routines need to be called indirectly via _ext_call_ routine which handles calling into the ROM while saving and restoring the banking registers.

The changes to _fputc_cons_native_ and _fgetc_cons_native_ are simple but, these changes are not compatible with the original Aquarius.

My question is, how to handle different _fputc_cons_native_ and _fgetc_cons_native_ routines for the Aquarius and Aquarius+? Should a new target be created aqplus or is there a better way?

Thanks.

suborb, 2024-09-19:
There's a few ways, to do it.

Either we use trampoline routines, so that library code looks like this:

```
call firmware
defw address
```

Then `firmware` is implemented in the crt0. eg the +cpc version restores all the registers the firmware needs.

Or if you put `fputc_cons_native` etc in a library and link that library it will override what's there (that's how the CP/M targets work).

Or you can do a redirect on the command line (see what `--generic-console` maps to).

Normally by the time banking is involved it's time to call in `--generic-console` and `--hardware-keyboard` and bypass the firmware as much as possible!

iratahack, 2024-09-19:
@suborb, my plan would be to copy the existing routines and modify them for the aqplus, creating the 2 files below.

- target/aquarius/stdio/fgetc_cons_aqplus.asm
- target/aquarius/stdio/fputc_cons_aqplus.asm

then modify target/aquarius/aqplus.lst to include them. When using -clib=aqplus the new routines would be picked up, else the original ones will be used. If you are OK with this I'll create a PR.

suborb, 2024-09-19:
That's perfect - don't forget about getk.asm as well though - it has a call to $1f01 buried in it.

iratahack, 2024-09-19:
KEYASC (address $1f00/$1f01) is part of the standard ROM and is available in both Aq and Aq+ since there is no bank switching by KEYASC there is no need for an indirect call, but IX does need protecting as it is clobbered. I will add this change also for the aqplus library.

closed 2024-09-20 by iratahack