z88dk/z88dk #2599: Local labels getting assigned to wrong scope?

ceharris, 2024-09-22:
Consider this simple example:
```
foo:
	nop
@loop1:
	jp @loop1
@loop2:
	call bar
	jp @loop2

bar:
	ret
```

The label `@loop1` works as expected. However, the assembler complains about the second label:
```
demo1.asm:8: error: undefined symbol: bar@loop2
  ^---- bar@loop2
```

It seems to think the scope for `@loop2` is `bar`, not `foo`. 

Experimenting a bit, it seems to get confused by any non-local symbol referenced between the local label definition and a subsequent reference to the local label. For example:
```
FOOBAR = 1

foo:
@loop:
	ld a,(hl)
	cp FOOBAR
	inc hl
	jr nz,@loop
```

This results in a similar error.
```
demo2.asm:8: error: undefined symbol: FOOBAR@loop
  ^---- FOOBAR@loop
```

closed 2024-09-23 by pauloscustodio

ceharris, 2024-09-24:
Thanks for the quick fix, @pauloscustodio!  👍

dbolli, 2024-09-24:
Agreed. Retested successfully in nightly build. Thanks @pauloscustodio 👍👋

commit 17e09f01 (2024-09-23): Fix #2599: usage of global label was initializing a new scope for local labels
commit b44eb140 (2024-09-23): Merge pull request #2600 from z88dk/issue_2599

Fix #2599: usage of global label was initializing a new scope for