z88dk/z88dk #2606: [z88-z80asm]: Library file suffix changed(!) to ".lib"

SvenMichaelKlose, 2024-09-25:
Unnecessarily limiting and confusing where it could have blended in easily (multi-compiler build set-up).

pauloscustodio, 2024-09-25:
This is a legacy feature.

Question to the group: shall we change it to write library files with the name supplied without changing the extension to .lib? I suspect a great part of the tool chain will fail with this change.

dbolli, 2024-09-26:
In my opinion library files should have a .lib extension for the same reason text files have a .txt extension and some image files have a .jpg extension.

suborb, 2024-09-26:
Agree that they should have an extension. 

Maybe rules should be:

1. Use extension if provided. 
2. If no extension use .lib

Which works for creation.  

For usage the same rules could be applied.

SvenMichaelKlose, 2024-09-26:
If that breaks builds: hell no! Was able to fix my build stuff quickly any way. Never mind.

pauloscustodio, 2024-09-26:
The change is done in a branch and the CI is testing. Up to now everything is fine

pauloscustodio, 2024-09-26:
Fixed and integrated. Thanks for reporting.

commit b9a60128 (2024-09-26): Fix #2606: allow different lib extensions
closed 2024-09-26 by pauloscustodio