z88dk/z88dk #2607: [feature request] Could you please support the Hector computer (aks Victor Lambda)

Fabrizio-Caruso, 2024-09-27:
Could you please support the Hector computer (aks Victor Lambda)?
 
 https://fr.wikipedia.org/wiki/Hector_(micro-ordinateur)

suborb, 2024-10-13:
Do you know of an emulator and ROMs?

Fabrizio-Caruso, 2024-10-13:
Hi
There is an emulator and docs here: 
http://dchector.free.fr/
http://hectorvictor.free.fr/

Fabrizio-Caruso, 2024-10-14:
@suborb I see the roms here:
http://hectorvictor.free.fr/index.php?page=pNaq0x.HVa1AE

MAME/MESS is also an alternative emulator.

hectorvictor.free.fr/ contains tools such as wav2hec and hec2wav.

If you are blocked by the French language, I can give you a hand.

Fabrizio-Caruso, 2024-10-20:
The project in https://github.com/tomconte/dontwakecthulhu has Assembly code that call ROM routines.

tomconte, 2024-10-20:
Hi folks, I am happy to help, I am by no means an Hector (nor Z88dk) expert and I got started by cheating a little bit (I am not using the library, just a custom crt0 to jump to my code). But now that I understand a little bit more about both systems, I think I am starting to understand what would be needed to support this computer.

On thing to keep in mind is that there is actually a line of similar, but somewhat incompatible, machines. I think that the Victor 1 (aka Interact) might be out of scope since it is based on a 8080 processor. We could start with two targets: 

- HR == Hector 2 HR and HR+, and 
- HRX == Hector HRX, similar to the 2 HR but based on the Forth language instead of BASIC

Fabrizio-Caruso, 2024-10-20:
@tomconte Thanks!! 
8080 is *not* out of score. The two compilers in Z88DK (SCCZ80 and ZSDCC) can build binaries for the Intel 8080 and its derivaties such as the Intel 8085 and GBZ80 (GameBoy) as well as Z80 and its derivatives. There are already several Intel 8080 targets with library support in Z88DK.

suborb, 2024-10-20:
It's a tangled web...so if I understand correctly:

Apart from the CPU, these have the same hardware:

* Interact = Victor 1 = Vector Lambda = 8080 @ 2Mhz
* Hector 1, Victor Lambda 2 = z80 @ 1.75Mhz

Hardware wise:

* There's 4k of ROM from 0->0x1fff.
* Max of 48k of RAM from 0x4000 -> 0xffff, I'm guessing probably shipped with 8 or 16k?
* VRAM is at 0x4000 -> 0x49ff

It looks like there's some write only memory mapped hardware registers:

* 0x1000, 0x1800 - palette settings
* 0x2000 -> 0x2003 and 0x2800 -> 0x2803 = sound chip
* 0x3000 = read write for cassette
* 0x3800 -> 0x3807 = keyboard reading

Joysticks overlay certain keyboard keys.

Given the size of the VRAM, I guess it's a bitmap display with 2 bpp, looks like it's arranged with bit 0,1 displayed on the left. Given the clock speed I'm suspect it wasn't wonderfully speedy.

So it looks like we could support.

* [ ] Generic console
* [ ] Text attributes (bold, underline, colour)
* [ ] UDGs
* [ ] Inkey keyboard
* [ ] Joystick support
* [ ] Full graphics support

We'll drive the hardware directly as much as possible to avoid ROM calls and the inevitable differences between models.

The generated output will be a k7 file.

Then there's HR and HRX models.

suborb, 2024-10-22:
First screenshot (enigma.c):

[image]

suborb, 2024-10-27:
I've not had much time this week, but I've got direct screen access working now: 

[image]

But I'm not happy with the text resolution so I think I'm going to switch it over to using 4x8 fonts.

suborb, 2024-11-04:
Bit of work to do on the vpeek, but HR+ seems to work with most of the same code:

[image]

suborb, 2024-11-05:
Waiting for #2638 to go through and then support for the HR will be merged.

Doc page is here: https://github.com/z88dk/z88dk/wiki/Platform---Hector

Have fun, any issues, raise another issue!

closed 2024-11-05 by suborb

Fabrizio-Caruso, 2024-11-11:
@suborb Could you please provide some instructions on how to load the binaries or k7 files. 
None of the binaries I build for my 9 games can be loaded. I have tried both VB Hector as well as DC Hector.

Fabrizio-Caruso, 2024-11-14:
Not sure this helps but maybe the VB emulator seems to provide some hints (registers? ports?) and/or debugging options for the sound chip:
[image]