z88dk/z88dk #2626: Far Pointer access to VRAM on Aquarius+

mgkaiser, 2024-10-18:
On the Aquarius+ VRAM appears at banks 20 and 21 (https://github.com/fvdhoef/aquarius-plus/wiki/TRM%3A-Overview).   Currently far pointer support allows easy access to all RAM banks at banks 32 - 63, but provides no access to VRAM.   Presently valid values for the upper 8 bits of a far pointer are $00 - $1f which are mapped to 32-63 on the Aquarius+ banks.  I propose that $20-$3f be mapped to banks 0-31 on the Aquarius+.  This would allow both access to VRAM and read only access to the ROM's.  If this were done Video RAM would appear at $42:0000 - $42:ffff and Character RAM would appear at $43:0000 - $43:ffff.

dbolli, 2024-10-18:
Please correct me if I'm wrong but VRAM is accessible via IO_BANK3

`    // Map video RAM (page 20) to $C000 (bank 3)
    IO_BANK3 = 20;
`

Please see init() in z88dk/examples/aquarius/tiles_and_sprites/main.c

Regards,
Derek.

suborb, 2024-10-18:
I believe  @iratahack allowed this in #2603 - RAM now starts at 0x80000

Banks 0-3 are inaccessible since 23:16 being 0 indicates local memory.

mgkaiser, 2024-10-19:
You can choose to map any RAM anywhere.  That example is mapping VRAM to bank 3.   But the __far and __banked directives also map stuff to bank 3 so this method might clash.

Sent from my Verizon, Samsung Galaxy smartphone

mgkaiser, 2024-10-19:
I'll give this a try.  Nothing in the docs mention this and my Z80 assembly is weak so I did not read the code to indicate this but I could just be weak on Z80.....

Sent from my Verizon, Samsung Galaxy smartphone

mgkaiser, 2024-10-20:
Ok, after some research I now get it.  Far pointers are 24 bits.  They are defined as such:

Bits 0-13 form the 14 bit (16k) offset.
Bits 14-17 form the bank.
Bits 18-23 are UNUSED.

So if I want a pointer to 0x0000 in bank 0x14 (decimal 20, video ram) the value of the pointer is 0x050000.   Video RAM occupies the addresses 0x050000 - 0x053fff.   Character RAM at bank 0x15 (decimal 21) is at 0x54000 - 0x57ffff.    A little more documentation on how the far pointers map to physical addresses would be helpful.  The only use case where this breaks down is that banks 0-3 make the upper 8 bits of the far pointer equal 0x00 which is interpreted to mean "use whatever is mapped in for the Z80" rather than to do banking.  Banks 1-3 contain nothing so no loss there.  Bank 0 is the boot ROM which really isn't super useful once the machine is running.

suborb, 2024-10-20:
For the aqplus that's correct, and it's actually fairly unusual for there to be such a direct logical -> physical mapping. Most of time far is setup just to target the RAM (as it was originally with aqplus).

The general principle is that a far pointer contiguous, however in most cases the hardware isn't. The far concept came out of the z88 where you could at most allocate 253 bytes in one chunk and the physical address would be wherever there was space in the RAM. As a result we need a mapping table.

This dynamic mapping table was carried over to MSXDOS (where you can allocated 16k pages, but again could be anywhere in memory) and a hardcoded mapping table for +zx where default memory usage isn't linear.

The +sam mapping has to handle both internal and external mapping and a hardware slot index as well as the bank.

suborb, 2024-10-20:
Back to the original point though - access to VRAM.

Far memory isn't a speed demon - there's a lot of banking involved and for something like VRAM the overhead might prove excessive, for example the memcpyf implementation is:

```
;far *memcpyf(far *dst,far *src,size_t n)
; copies n bytes from src to dest
.memcpyf
._memcpyf
    ld      ix,2
    add     ix,sp
    ld      c,(ix+6)
    ld      b,(ix+7)
    ld      e,(ix+8)    ; E'B'C'=dst
    exx
    ld      c,(ix+2)
    ld      b,(ix+3)
    ld      e,(ix+4)    ; EBC=src
    call    __far_start
    ex      af,af'      ; save seg 1 binding
    ld      l,(ix+0)
    ld      h,(ix+1)    ; HL=n
    push    ix
    push    hl
    pop     ix          ; IX=n
.memcpy1
    ld      a,ixl
    or      ixh
    jr      z,memcpy3   ; on if copied n chars
    call    __far_page
    ld      a,(hl)          ; char from src
    ld      iyl,a
    call    l_far_incptrs
    exx
    call    __far_page
    ld      a,iyl
    ld      (hl),a          ; place at dst
    call    l_far_incptrs
    exx
    dec     ix
    jr      memcpy1
.memcpy3
    pop     ix
    ex      af,af'
    call    __far_end
    ld      l,(ix+6)
    ld      h,(ix+7)
    ld      e,(ix+8)        ; EHL=dst
    ret
```

So for direct VRAM access it would be quicker to page in manually and just use a regular pointer.

closed 2024-10-23 by suborb