z88dk/z88dk #2629: [bug?] Has the Z88DK build beem broken very recently?

Fabrizio-Caruso, 2024-10-20:
When building z88dk I am getting this error. I am actually doing this through a Dockerfile. So maybe I am doing something wrong.
```
3632.5 z88dk-z80asm -mz180  -d -m4=-Itarget/scz180/obj -m4=-I../../src/m4 -Otarget/scz180/obj/sdcc_ix -Itarget/scz180/obj -xlib/sdcc_ix/scz180 -I/sdk/z88dk/libsrc/_DEVELOPMENT//../../lib/ -D__SDCC -D__SDCC_IX @target/scz180/library/scz180_sdcc_ix.lst
3633.3 failed to seek to 180 in file 'target/scz180/obj/sdcc_ix/adt/wv_priority_queue/c/sdcc_ix/wv_priority_queue_size.o~'
3633.4 make: *** [Makefile:112: lib/sdcc_ix/scz180.lib] Error 1
3633.4 make: Leaving directory '/sdk/z88dk/libsrc/_DEVELOPMENT'
```

suborb, 2024-10-20:
I thought we'd banished all of those...what are your make options? Is there a -j in there somewhere?

Fabrizio-Caruso, 2024-10-20:
I think I should close this issue before you get angry at me.
Maybe this is not an issue because it fails on one VM which may have run out of memory. My bad. Sorry for not seeing that.
We use a Dockerfile in Cross-Lib: https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/.devcontainer/Dockerfile

It looks ok for me:
```
RUN mkdir -p /sdk
WORKDIR /sdk
ENV BUILD_SDCC=1
ENV BUILD_SDCC_HTTP=1
RUN git clone --recursive https://github.com/z88dk/z88dk.git && \
    apt update && \
    apt install -y build-essential bison flex libxml2-dev subversion zlib1g-dev m4 ragel re2c dos2unix texinfo texi2html gdb curl perl cpanminus ccache libboost-all-dev libmodern-perl-perl libyaml-perl liblocal-lib-perl libcapture-tiny-perl libpath-tiny-perl libtext-table-perl libdata-hexdump-perl libregexp-common-perl libclone-perl libfile-slurp-perl pkg-config libgmp3-dev && \
    cpanm --local-lib=~/perl5 App::Prove CPU::Z80::Assembler Data::Dump Data::HexDump File::Path List::Uniq Modern::Perl Object::Tiny::RW Regexp::Common Test::Harness Text::Diff Text::Table YAML::Tiny && \
    eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib) && \
    cd z88dk && chmod 777 build.sh && ./build.sh
    #&& rm -R /sdk/z88dk
ENV PATH="$PATH:/sdk/z88dk/bin"
```

It works on a different VM. So it must be something wrong with the first VM.

So I am closing this issue for the moment. I will recheck it.

closed 2024-10-20 by Fabrizio-Caruso