z88dk/z88dk #2631: (ticks) 8085 ldsi, aka ld de, sp+n not setting carry flag

pauloscustodio, 2024-10-21:
Test code:
```
     1  0000  31ffff            	ld		sp, 65535
     2  0003  a7                	and a
     3  0004  3801              	ldsi 1
     4  0006  c30000            	jp 0
```
Ticks run:
```
$ z88dk-ticks.exe -m8085 -trace test.bin
...
pc=0004, [pc]=38,    bc=0000,  de=0000,  hl=0000,  af=0056, ix=0000, iy=0000
sp=FFFF, [sp]=3100, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=1 P/V=1 N=1 C=0
clockh=1C56, clockhl=C937
                    ld        de,sp+$01                     ;[0004] 38 01
pc=0006, [pc]=C3,    bc=0000,  de=0000,  hl=0000,  af=0056, ix=0000, iy=0000
sp=FFFF, [sp]=3100, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=1 P/V=1 N=1 C=0
```

pauloscustodio, 2024-10-21:
same problem with ldhi, aka ld de, hl+n

suborb, 2024-10-21:
I don't think they should do anything with carry - The 8085 opcode chart by @feilipu doesn't show it. And for a non-z88dk view, mame does this:

```
                case 0x38: // 8085: undocumented LDSI nn, otherwise undocumented NOP
                        if (is_8085())
                        {
                                m_WZ.d = read_arg();
                                m_DE.d = (m_SP.d + m_WZ.d) & 0xffff;
                        }
```

and

```
                case 0x28: // 8085: undocumented LDHI nn, otherwise undocumented NOP
                        if (is_8085())
                        {
                                m_WZ.d = read_arg();
                                m_DE.d = (m_HL.d + m_WZ.d) & 0xffff;
                        }
                        break;
```

feilipu, 2024-10-21:
Table 2-10 in the Tundra Semi data-sheet shows no flags are affected by those instructions.

https://gitlab.com/feilipu/8085-opcodes/-/blob/main/Tundra%20Semiconductor%208000%20Series%20Components%20Manual.pdf

pauloscustodio, 2024-10-22:
> but could test on some hardware when I return

That would be great. Also, the document [CA80C85B Extended Instruction Set](https://gitlab.com/feilipu/8085-opcodes/-/blob/main/Tundra%20Semiconductor%208000%20Series%20Components%20Manual.pdf) could be added to the techdocs github?

feilipu, 2024-10-22:
It is already there, but hidden within the entire [Tundra Semi 8000 Series](https://github.com/z88dk/techdocs/blob/master/intel/Tundra%20Semiconductor%208000%20Series%20Components%20Manual.pdf) catalog.

I can copy the excerpt in too.

pauloscustodio, 2024-10-23:
> Just as a [further reference](https://www.righto.com/2013/07/), to note that Pavel Zima built a transistor level simulator of 
> the 8085 based on examination of the silicon dies, and his Opcode table also shows that the flags are unaffected by these 
> instructions.

Given this information, I will close this issue and correct the failing test in z80asm by ignoring the flags.

closed 2024-10-23 by pauloscustodio

feilipu, 2024-10-24:
I've played around with this using a real Intel 8085.
Not exhaustive, just with the `ld de,hl+n` instruction, with some different iterators, across t

There's no evidence that the current view is incorrect.
But it does raise another issue.

```c
/*
   bug-2631.c
   check the 8085 ld de,hl+n and ld de,sp+n instructions don't influence carry
   clear flags, add 1 to input hl, and look at flag bits
*/

// zcc +cpm -clib=8085 -SO2 --list bug-2631.c -o bug-2631.com

#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>

#pragma printf = "%s %c %X"     // enables  %s %c %X only

uint32_t testBug( uint16_t input) __z88dk_fastcall;

uint32_t testBug( uint16_t input)
{
    __asm__(                    \
        "xor a              \n" \
        "ld de,hl+1         \n" \
        "push af            \n" \
        "pop hl             \n" \
        );                      \
}

void main (void)
{
    for (uint16_t test = 0x0001; test != 0x0000; ++test)
            fprintf(stdout, " Input HL: %X  Output DE,HL+1: %X AF: %X\n", test, (uint16_t)(testBug(test)>>16), (uint16_t)testBug(test));
}
```
Produces (in excerpt)...

```
 Input HL: F6  Output DE,HL+1: F7 AF: 44
 Input HL: F7  Output DE,HL+1: F8 AF: 44
 Input HL: F8  Output DE,HL+1: F9 AF: 44
 Input HL: F9  Output DE,HL+1: FA AF: 44
 Input HL: FA  Output DE,HL+1: FB AF: 44
 Input HL: FB  Output DE,HL+1: FC AF: 44
 Input HL: FC  Output DE,HL+1: FD AF: 44
 Input HL: FD  Output DE,HL+1: FE AF: 44
 Input HL: FE  Output DE,HL+1: FF AF: 44
 Input HL: FF  Output DE,HL+1: 100 AF: 44
 Input HL: 100  Output DE,HL+1: 101 AF: 44
 Input HL: 101  Output DE,HL+1: 102 AF: 44
 Input HL: 102  Output DE,HL+1: 103 AF: 44
 Input HL: 103  Output DE,HL+1: 104 AF: 44
 Input HL: 104  Output DE,HL+1: 105 AF: 44
 Input HL: 105  Output DE,HL+1: 106 AF: 44
 Input HL: 106  Output DE,HL+1: 107 AF: 44
 Input HL: 107  Output DE,HL+1: 108 AF: 44
 Input HL: 108  Output DE,HL+1: 109 AF: 44
 Input HL: 109  Output DE,HL+1: 10A AF: 44
 Input HL: 10A  Output DE,HL+1: 10B AF: 44
 Input HL: 10B  Output DE,HL+1: 10C AF: 44
 Input HL: 10C  Output DE,HL+1: 10D AF: 44
 Input HL: 10D  Output DE,HL+1: 10E AF: 44
 Input HL: 10E  Output DE,HL+1: 10F AF: 44
 Input HL: 10F  Output DE,HL+1: 110 AF: 44
 Input HL: 110  Output DE,HL+1: 111 AF: 44
```

pauloscustodio, 2024-10-24:
> There's no evidence that the current view is incorrect.
> But it does raise another issue.

What do you mean?

feilipu, 2024-10-24:
> What do you mean?

See #2632.