z88dk/z88dk #2632: z80asm - list doesn't (always) include immediate values for 8085 opcodes

feilipu, 2024-10-24:
From #2631, listings should include the immediate bytes, but this is only partially correct.

 - `ld de,hl+n` seems never to produce an immediate byte listing other than `00`.
 - `ld de,sp+n` seems to only produce the immediate byte listing for values greater than 5 (i.e. `ld de,sp+6` is correct).

The binary produced is correct, however. This only affects listings.

For example.

```asm

    18                          ; Function testBug flags 0x00000208 __smallc __z88dk_fastcall
    18                          ; unsigned long uint32_ttestBug(unsigned int input)
    18                          ; parameter 'unsigned int input' at sp+2 size(2)
    18                          	C_LINE	19,"bug-2631.c::testBug::0::0"
bug-2631.c::testBug::0::0:
    19                          ._testBug
    19  0000  e5                	push	hl
    19  0001  af                	xor a
    19  0002  2800              	ld de,hl+1
    19  0004  f5                	push af
    19  0005  e1                	pop hl
    19  0006  c1                	pop	bc
    19  0007  c9                	ret
    19                          
    19                          
    19                          	C_LINE	28,"bug-2631.c::testBug::0::1"
bug-2631.c::testBug::0::1:
    28                          
    28                          ; Function main flags 0x00000000 __stdc
    28                          ; void main()
    28                          	C_LINE	29,"bug-2631.c::main::0::2"
bug-2631.c::main::0::2:
    29                          ._main
    29  0008  210000            	ld	hl,1	;const
    29  000b  e5                	push	hl
    29  000c  c30000            	jp	i_4	;EOS
    29                          .i_2
    29  000f  e1                	pop	hl
    29  0010  23                	inc	hl
    29  0011  e5                	push	hl
    29                          .i_4
    29  0012  e1                	pop	hl
    29  0013  e5                	push	hl
    29  0014  7c                	ld	a,h
    29  0015  b5                	or	l
    29  0016  ca0000            	jp	z,ASMPC+4
    29  0019  37                	scf
    29  001a  d20000            	jp	nc,i_3	;
    29  001d  210000            	ld	hl,__sgoioblk+10
    29  0020  e5                	push	hl
    29  0021  210000            	ld	hl,i_1+0
    29  0024  e5                	push	hl
    29  0025  3800              	ld	de,sp+4
    29  0027  ed                	ld	hl,(de)	;l_gint
    29  0028  e5                	push	hl
    29  0029  3806              	ld	de,sp+6
    29  002b  ed                	ld	hl,(de)	;l_gint
    29  002c  cd0000            	call	_testBug
    29  002f  eb                	ex	de,hl
    29  0030  110000            	ld	de,0
    29  0033  e5                	push	hl
    29  0034  3808              	ld	de,sp+8
    29  0036  ed                	ld	hl,(de)	;l_gint
    29  0037  cd0000            	call	_testBug
    29  003a  e5                	push	hl
    29  003b  3e05              	ld	a,5
    29  003d  cd0000            	call	fprintf
    29  0040  c1                	pop	bc
    29  0041  c1                	pop	bc
    29  0042  c1                	pop	bc
    29  0043  c1                	pop	bc
    29  0044  c1                	pop	bc
    29  0045  c30f00            	jp	i_2	;EOS
    29                          .i_3
    29  0048  c1                	pop	bc
    29  0049  c9                	ret

```

pauloscustodio, 2024-11-12:
@feilipu, I cound not reproduce this with z80asm alone. Can you please provide the C code and the zcc command line?

pauloscustodio, 2024-11-15:
Found smallest code that triggers bug:
```
	SECTION	code
	ld de,hl+1
	SECTION data
	defs 8
```
assembles and lists to:
```
test.asm:
     1                          	SECTION	code
     2  0000  2800              	ld de,hl+1
     3                          	SECTION data
     4  0000  0001000000000000  	defs 8
     5                          
     6                          
```
i.e. the 0x01 immediate value of the code section is being writen by list in the data section.

Now for the bug hunt.

feilipu, 2024-11-15:
Without further testing, does your fix also handle the `ld de,sp+n` issue too?

It was odd that this instruction has/had a different (but still broken) list behaviour to the `ld de, hl+n` instruction.

pauloscustodio, 2024-11-18:
> Without further testing, does your fix also handle the ld de,sp+n issue too?

Thanks for the comment, added to the test case.

commit c8601573 (2024-11-15): Fix #2632 - map patch bytes to section+address
commit de91cdc2 (2024-11-18): Fix #2632 - map patch bytes to section+address

Bug fix uncovered wrong result in PHASE.t
closed 2024-11-18 by pauloscustodio