z88dk/z88dk #2635: error 9 FATAL Compiler Internal Error in file 'SDCCopt.c' line number '312'

TinfoilAsteroid, 2024-11-01:
When compiling by test code it seems to break the optimizer casting int16_t to float:

```
#include <stdint.h>

uint8_t IntNorms[2][04] ={ {0x1F, 0x00, 0x3E, 0x1F},
                           {0x9F, 0x12, 0x37, 0x10},  };

float normalsf[13][3];

void main(void)
{
    uint8_t i = 0;
    while (i < 2)
    {
        uint8_t value8x = IntNorms[i][0];
        int16_t value16x = (int16_t)IntNorms[i][0];
        normalsf[i][0] = (float)value16x;
        ++i;
    }
    while (1==1);
}     
```
Compiled on windows cmd: 
zcc +zxn -c -vn -compiler=sccz80 -clib=sdcc_iy --math32 -SO2 --max-allocs-per-node3000 CodeTest.c
Gives CodeTest.c:20: error 9: FATAL Compiler Internal Error in file 'SDCCopt.c' line number '312' : 0
If I change the while loop to cast the uint8_t to float 
"normalsf[i][0] = (float)value8x;"
that compiles fine
and if I change the loop to

    while (i < 2)
    {
        uint8_t value8x = IntNorms[i][0];
        int16_t value16x = 2;
        normalsf[i][0] = (float)value16x;
        ++i;
    }
that compiles fine
but this fails:
    while (i < 2)
    {
        uint8_t value8x = 2;
        int16_t value16x = (int16_t)IntNorms[i][0];
        normalsf[i][0] = (float)value16x;
        ++i;
    }
    while (1==1);

I think its an issue with casting int16_t to float tripping up the optimiser. It looks to have happened at least in the nightly builds since around 10-Sept according to when I last downloaded a nightly build. Also happens with last nights.

Running under Win10 pro

feilipu, 2024-11-02:
Still looking at this, but a couple of initial thoughts.

Your compile line is asking for sccz80, and not sdcc.
This makes the `--max-allocs-per-node3000` request invalid.
It also makes the request for the `sdcc_iy` library invalid.
Basically the compiler request and the library request don't match.

> `zcc +zxn -c -vn -compiler=sccz80 -clib=sdcc_iy --math32 -SO2 --max-allocs-per-node3000 CodeTest.c`

TinfoilAsteroid, 2024-11-02:
Added #include <math.h> back in 
zcc +zxn -c -clib=sdcc_iy --math32 -SO2 CodeTest.c
CodeTest.c:21: error 9: FATAL Compiler Internal Error in file 'SDCCopt.c' line number '312' : 0
Contact Author with source code

zcc +zxn -c --list -clib=new --math32 -SO2  CodeTest.c
compiles without error

TinfoilAsteroid, 2024-11-02:
Code reads as below so fatal error is generated on while (1==1) (which I assume in reality is not the line tripping it given its actually line 18 tripping it up
---------------------

#include <stdint.h>
#include <math.h>

uint8_t IntNorms[2][04] ={ {0x1F, 0x00, 0x3E, 0x1F},

                           {0x9F, 0x12, 0x37, 0x10},  };
float normalsf[13][3];

void main(void)
{
    uint8_t i = 0;
    while (i < 2)
    {
        uint8_t value8x = 2;
        int16_t value16x = (int16_t)IntNorms[i][0];
        normalsf[i][0] = (float)value16x;
        ++i;
    }
    while (1==1);
}    
---------------------

feilipu, 2024-11-02:
Simplifying slightly, the issue seems dependent on whether float is declared global.
You can try it by just moving the comment to the `test_s` line, where the error is not present.

`'SDCCopt.c' line number '312`

```c
#include <stdint.h>

uint8_t IntNorms[2][4] =  { {0x1F, 0x00, 0x3E, 0x1F},
                            {0x9F, 0x12, 0x37, 0x10},  };
float normalsf[13][3];

float test_s;

void main(void)
{
    uint8_t i = 0;
    int16_t value16x;

    float test_d;

    while (i < 2)
    {
        value16x = (int16_t)IntNorms[i][0];
        test_s = (float)value16x;
//      test_d = (float)value16x;
        ++i;
    }
}
```

feilipu, 2024-11-02:
Reported upstream as [3795](https://sourceforge.net/p/sdcc/bugs/3795/).

I suggest you work around this issue as it will be quite a few months before we update `zsdcc` to the future v4.5.0 stable release.

PR #2659 (2025-01-29): zsdcc - update to 4.5.0 r15248
A patch to v4.5.0-rc1 (r15212) for zsdcc, for testing.

Will do a final version once v4.5.0 is released.

Closes #2658
Closes #2635
closed 2025-01-29 by suborb