z88dk/z88dk #2639: Setting color attribute for last char before \n is missing

kovzol, 2024-11-11:
[image]
I am facing a problem when using ANSI sequences via printf calls on Zx Spectrum:
```
#include <stdio.h>

int main() {

    printf("%c[%um", 27, 42);
    printf("12345678901234567890123456789012345678901234567890\n"); // wrong
    printf("1234567890123456789012345678901234567890123456789\n");
    printf("123456789012345678901234567890123456789012345678\n");
    printf("12345678901234567890123456789012345678901234567\n"); // wrong
    printf("1234567890123456789012345678901234567890123456\n");
    printf("123456789012345678901234567890123456789012345\n");
    printf("12345678901234567890123456789012345678901234\n"); // wrong
    printf("1234567890123456789012345678901234567890123\n");
    printf("123456789012345678901234567890123456789012\n"); // wrong
    printf("12345678901234567890123456789012345678901\n");
    printf("1234567890123456789012345678901234567890\n");
    printf("123456789012345678901234567890123456789\n"); // wrong

    return 0;
}
```
I compile and run this by entering `zcc +zx -vn ansibug.c -lndos -create-app -oansibug.bin -pragma-need=ansiterminal -Cl-v -pragma-define:ansicolumns=51 && fuse ansibug.tap`. It seems that the number of characters to color in the row is calculated incorrectly (there should be one more being colored in some cases, because the last character is not always colored: see the cases "wrong" in the code).

suborb, 2024-11-12:
I believe it only colours one attribute cell when it prints a character. 

Of course, a character may span 2. Normally this is covered by the next character colouring that cell. But at the end of the line that isn't the case.

I suspect that filling in multiple may end up with other cases looking wrong!

I'll finish off the Hector work and then take a look at this.

kovzol, 2024-11-13:
Thank you in advance.

zx70, 2024-12-07:
I think it is just the font you are using not fitting in 8 pixels. You can consider it an effect similar to the colour clash,  you shouldn't see this same problem with 4 or 8 bits font.

kovzol, 2024-12-08:
Thanks for checking!