z88dk/z88dk #2640: [hectorhr] No input ?

Fabrizio-Caruso, 2024-11-11:
I am able to load my games with VB Hector and more randomly with the one your page suggests (DC Hector).
They run but no keyboard input works.

Is this expected because inkey is not implemented yet ?

I  have attached some binaries

I use this Makefile target:
```
hectorhr: $(ASSETS_PATH)/z88dk_xchase.asm   
	$(Z88DK_PATH)$(MYZ88DK) +hector -clib=hectorhr $(SCCZ80_OPTS) $(INCLUDE_OPTS) \
	-pragma-redirect:fputc_cons=fputc_cons_generic  \
	-vn   -D__HECTOR__ -D_XL_NO_SOUND -D_XL_NO_JOYSTICK \
	-lndos \
	-create-app -o $(BUILD_PATH)/X$(GAME_NAME)_$@.bin \
	-D__CONIO_GRAPHICS   -DXSize=28 -DYSize=21 \
	-D_XL_SLOW_DOWN_FACTOR=$(_COMMON_SLOWDOWN_FACTOR)*$(_MSX_SLOWDOWN) \
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(FULL_FILES)
```
[build.zip](https://github.com/user-attachments/files/17709125/build.zip)

suborb, 2024-11-12:
I've not done any testing on the HRX - the ROM is different so the firmware entry points will be completely different.

However, I added inkey support this evening, so `--hardware-keyboard` should mean you get input on the HRX.

Next up is `joystick()`, though there doesn't seem to be a "fire" button on the hardware sticks which is a bit annoying.

Fabrizio-Caruso, 2024-11-12:
@suborb yes, https://github.com/mamedev/mame/blob/master/src/mame/hec2hrp/hec2hrp.cpp#L281 

In this case you can chose a reasonable key for fire, which IMHO would be the SPACE bar, because it is big and convenient or whatever you have chosen for a FIRE key in the past when mapping fake joysticks to keys.

suborb, 2024-11-12:
I've added joystick in - Joystick 1 uses space as fire, Joystick 2 uses tab. There's also AQOP-BN, 8246-05 and hjkl-sd

Let me think about what we can do for the palette - definitely a function allow it to be configured.

suborb, 2024-11-13:
I've added `hector_set_palette(()` to `<arch/hector.h>` which allows the pen colours to be chosen. That way you can choose which colours you'd like to use

Fabrizio-Caruso, 2024-11-16:
How do I use hector_set_palette()?

suborb, 2024-11-16:
It’s detailed here: https://github.com/z88dk/z88dk/blob/master/include/arch/hector.h