z88dk/z88dk #2641: [hectorhr] RED color not shown (shown as black?)

Fabrizio-Caruso, 2024-11-12:
On both the (emulated) Hector HR+ as well as the (emulated) Hector HRX, the RED color is shown as black (so it is not shown at all). It should be shown as red or anything other than black. The two Hectors by default have different 4-color palettes but I think they should both have red. Any color other than black would be OK.

suborb, 2024-11-12:
There's no mapping taking place, with the default palette we get this:

[image]

Fabrizio-Caruso, 2024-11-12:
@suborb OK, no real problem. 
If I am not wrong, you also define macros for some integer values that "usually" do a reasonable mapping of the real colors.
In this very case: GREEN and WHITE are OK, BLACK not sure, RED goes to real BLACK, BLUE goes to real RED...
It is not a real issue indeed. I am using my own macros anyway.

suborb, 2024-11-13:
With 4 colours, mapping is always going to be weird. I guess if we set up the palette to black, blue, red, white, then we could do this mapping:

* BLACK -> BLACK
* BLUE -> BLUE
* GREEN -> RED
* CYAN -> BLUE
* RED -> RED
* MAGENTA -> RED
* BROWN -> RED
* LIGHTGRAY -> WHITE
* DARKGRAY -> BLACK
* LIGHTBLUE -> BLUE
* LIGHTGREEN -> RED
* LIGHTCYAN -> BLUE
* LIGHTRED -> RED
* LIGHTMAGENTA -> RED
* YELLOW -> WHITE
* WHITE -> WHITE

Which is probably passable..

suborb, 2024-11-13:
I've added in that mapping now (thanks for the poke) so you can remove your special magic!

closed 2024-11-13 by suborb

Fabrizio-Caruso, 2024-11-13:
I realize that any mapping done by constant values (I suppose macros) has a problem if it assumes it is using the default colors because the default colors are different on the HRX vs HR2+.

suborb, 2024-11-13:
> I realize that any mapping done by constant values (I suppose macros) has a problem if it assumes it is using the default colors because the default colors are different on the HRX vs HR2+.

Yeah, the mapping goes hand-in-hand with having a "z88dk palette" which is setup on program start if gencon is used.