z88dk/z88dk #2644: [ntsc][m5] M5 targets works fine on real PAL Sort M5 (both standard M5 and M5 PRO) but fails on NTSC-J M5 Junion

Fabrizio-Caruso, 2024-11-19:
M5 targets works fine on real PAL Sort M5 (both standard M5 and M5 PRO) but fails on NTSC-J M5 Junior

This has been tested on real hardware by Roy Templeman

[image]

The same M5 Junior with the very same monitor works fine with other games.

Is Z88DK doing something that makes it only work on PAL systems?

suborb, 2024-11-22:
The VDP registers look like this:

```
Reg/Bit	7	6	5	4	3	2	1	0
0	-	-	-	-	-	-	M2	EXTVID
1	4/16K	BL	GINT	M1	M3	-	SI	MAG
2	-	-	-	-	PN13	PN12	PN11	PN10
3	CT13	CT12	CT11	CT10	CT9	CT8	CT7	CT6
4	-	-	-	-	-	PG13	PG12	PG11
5	-	SA13	SA12	SA11	SA10	SA9	SA8	SA7
6	-	-	-	-	-	SG13	SG12	SG11
7	TC3	TC2	TC1	TC0	BD3	BD2	BD1	BD0
```

The one that I can possibly see being problematic is EXTVID which controls whether external video is enabled. We always reset it, perhaps this machine needs it to be enabled?

Fabrizio-Caruso, 2024-11-23:
@suborb We could ask @Loudscotsbloke (Roy) to test again but I need to understand how I can change this setting. 
I wonder whether this issue is a general NTSC-J issue of the VDP and more targets with the same VDP could be impacted.

@Loudscotsbloke has an M5 Junior (NTSC-J) which may be a slightly different computer. I don't know anything about it. I know he has tested other M5 games with it and they have no issue.

Anyway how can I set the bit? Can I do it through some VPOKE or some embedded Assembly?

suborb, 2024-11-23:
Well, I knew there were bells ringing. Look at this code:

https://github.com/z88dk/z88dk/blob/master/libsrc/video/tms9918/__tms9918_set_tables.asm#L74

Where we explicitly set the bit for the m5, so maybe it needs to be reset.

It's a bit kludgey, but since we're figuring it out, there's a couple of ways to do this:

1. Comment out that set line and rebuild m5_clib.lib
2. Hexedit the generated binary to `nop` out that set line and then invoke appmake separately

I think 1. is easier.

We've had that line/similar for years, and I'm wondering where the information came from.

If you can report back on success/failure on the junior. If successful can we also check against the PAL versions.

Fabrizio-Caruso, 2024-11-24:
@suborb why not just do this in my own code instead of recompiling z88dk? If it makes sense, how?
By the way I don't see the relation between NTSC-J and external video but it is worth trying if we have nothing else.
The M5 Jr may have other differences I am not aware of.

suborb, 2024-11-24:
You can use `vdp_set_reg(0,2)` for mode 2. `vdp_set_reg(0,0)` for modes 0 and 1.

The extvid signal is used to overlay the VDP screen on top of an external video input - given the picture you've sent through it looks like the sync has gone and I suspect that setting that bit syncs to the external video.