z88dk/z88dk #2647: [msx/svi] Mono graphics not working

fu-sen, 2024-12-01:
I've recently been using the z88dk again.

I am rebuilding previously built sources and mono graphics are not working in MSX and SVI.
This issue occurred even if I built the `example/` directory.

This issue is visible in version 2.3 and the latest nightly build.

suborb, 2024-12-03:
Previously attempting to use graphics forced a switch to VDP mode 2.

When I rewrote the VDP library to support a lot more screen modes I removed that feature. So now you have to either select the screen mode in code `vdp_set_mode(2)` or set the default mode for the program using the command line option: `-pragma-define:CLIB_DEFAULT_SCREEN_MODE=2`

fu-sen, 2024-12-04:
When I used z88dk in the past, MSX required `-lmsxbios`. I changed it to `-pragma-define:CLIB_DEFAULT_SCREEN_MODE=2` and it worked.

svi is not working the same way. In the first place, `OK` is displayed on the screen without cutting.
However, I think this is a separate issue, so I will close this issue for now. Thank you for your response!

closed 2024-12-04 by fu-sen

fu-sen, 2024-12-05:
Thanks for the testing and the option tips.
I was trying some example/ folders.

I was trying out this program I created in the past with z88dk:
https://gist.github.com/fu-sen/4e079229d43a0aa6c9d40734c35fb2ab

I have BlueMSX and openMSX which support SV-318/SV-328 and am testing with them.
```
zcc +svi -lm -create-app -pragma-define:CLIB_DEFAULT_SCREEN_MODE=2 sin.c
```
`BLOAD"cas:",R` gives `Found:sin`, but the screen does not change and immediately outputs `Ok`.

fu-sen, 2024-12-05:
I understand the reason for this modification. It also works on `-pragma-define:CLIB_DEFAULT_SCREEN_MODE=3`.

```
zcc +msx -lm -create-app -pragma-define:CLIB_DEFAULT_SCREEN_MODE=3 sin.c
zcc +svi -lm -create-app -pragma-define:CLIB_DEFAULT_SCREEN_MODE=3 sin.c
```
[image]

suborb, 2024-12-05:
The SVI-318 has a different amount of RAM - I can see we have a rom318 target which moves the BSS section - from the same sourcecode, only the 0xc000 bank is populated.

So, I think we need to move the `CRT_ORG_CODE` for the other types if you want it to run on the 318. You can do this by adding `-pragma-define:CRT_ORG_CODE=51200` on the command line.

fu-sen, 2024-12-05:
```
>zcc +svi -pragma-define:CRT_ORG_CODE=51200 -lm -create-app -pragma-define:CLIB_DEFAULT_SCREEN_MODE=2 -subtype=rom -bn sin sin.c
c:/z88dk/lib/target/svi/classic/rom.asm:9: error: duplicate definition: CRT_ORG_CODE
  ^---- defc CRT_ORG_CODE = 0x0000
      ^---- defc CRT_ORG_CODE=0
```
Unfortunately it didn't work. and changing rom.asm didn't work.

<https://github.com/z88dk/z88dk/blob/9b164def8c4c395e8ac9c818aa0a83cbc42bbfd0/lib/target/svi/classic/rom.asm#L6-L9>
```
IF !DEFINED_CRT_ORG_BSS
    defc    CRT_ORG_BSS = 0x8000
ENDIF
    defc    CRT_ORG_CODE = 0x0000
```

suborb, 2024-12-05:
Ah, it was already documented: https://github.com/z88dk/z88dk/wiki/Platform---Spectravideo#16k-model

I wasn't clear, the ROM case is already available, it's the .cas subtype you need the CRT_ORG_CODE for. In summary you should be able to use one of:

```
zcc +svi -subtype=rom318- lm -pragma-define:CLIB_DEFAULT_SCREEN_MODE=2 bifurc.c -create-app
zcc +svi -lm -pragma-define:CRT_ORG_CODE=49200 -pragma-define:CLIB_DEFAULT_SCREEN_MODE=2 bifurc.c -create-app
```

fu-sen, 2024-12-05:
Oh, working on SV-318! I added information on SV-318. (I have already included the SV-328:
https://gist.github.com/fu-sen/4e079229d43a0aa6c9d40734c35fb2ab

Thank you for teaching me so much. These will be hints for those who want to run MSX and SVI.😆