z88dk/z88dk #2650: [msx][svi] SPR_XOR in MSX SCREEN 3 behaves like SPR_OR

fu-sen, 2024-12-14:
I created a simple test source. Displays a blinking monochrome sprite.:
```
#include <graphics.h>
#include <games.h>
#include <stdlib.h>

char testfont[] = { 32, 6,
  0x7e, 0x7e, 0x3e, 0x7e,
  0x18, 0x60, 0x60, 0x18,
  0x18, 0x78, 0x3c, 0x18,
  0x18, 0x60, 0x06, 0x18,
  0x18, 0x60, 0x06, 0x18,
  0x18, 0x7e, 0x7c, 0x18
};

void main(){
  clg();

  while (1) {
    putsprite(SPR_XOR, getmaxx()/2-16, getmaxy()/2-3, testfont);
    sleep(1);
  }
}
```

MSX SCREEN 2 works as expected:
```
zcc +msx -pragma-define:CLIB_DEFAULT_SCREEN_MODE=2 -create-app -subtype=rom -bn xortest2 xortest.c
```

but, in MSX SCREEN 3, the sprite remains lit and behaves like SPR_OR.:
```
zcc +msx -pragma-define:CLIB_DEFAULT_SCREEN_MODE=3 -create-app -subtype=rom -bn xortest3 xortest.c
```

I was able to reproduce the same issue with SV-328:
```
zcc +svi -create-app -pragma-define:CLIB_DEFAULT_SCREEN_MODE=3 -bn test test.c
```

suborb, 2024-12-17:
Thank you for a great example that shows the issue. 

I've pushed a fix for the issue - as you'd expect a silly bug.

closed 2024-12-17 by suborb

fu-sen, 2024-12-18:
I updated z88dk to z88dk-win32-20241218-9a5d529feb-23058.zip.
After that I checked it works on MSX SCREEN 3. It's perfect. Thank you! 😄