z88dk/z88dk #2651: Missing ansifont and oemascii

SM7I, 2024-12-14:
I have code for ZX81 that depends on oemansii and ansifont. We discovered that this is not avaliable in Z88DK 2.3 (Linux), but in 1.99 (Linux) and 2.3 (Win32) we can see the symbol present in the lib/clibs/zx81_clib.lib. Don´t know about Mac, but is it possible to fix this for Linux kind of asap ?

suborb, 2024-12-14:
I'm slightly puzzled by this, ansifont hasn't been defined by a library since about 2016!

It's worked out at link time by lib/crt/classic/crt_runtime_selection.asm where based on the number of columns in the screen it's assigned to a particular font.

I'm confused about the other symbol: `oemascii` or `oemansii`? Either way I'm not seeing anything in the library for 2.3.

SM7I, 2024-12-14:
This is the complete command line that I used to use in 1.99. Can it be modified to obtain same results ?

zcc +zx81 -subtype=_wrx -clib=wrxansi -pragma-redirect:ansifont=_oemascii -pragma-define:ansifont_is_packed=0 -pragma-define:ansicolumns=80 -lndos -lm81 -O3 -create-app -o test test.c

suborb, 2024-12-14:
Okay, so you're defining the 80 column font with your own symbol. I can run the same test on examples/console/ansitest:

```
zcc +zx81 -subtype=_wrx -clib=wrxansi -pragma-redirect:ansifont=_oemascii -pragma-define:ansifont_is_packed=0 -pragma-define:ansicolumns=80 -lndos -lm81 -O3 -create-app ansitest.c oemtest.asm
dom@zebedee console % zcc +zx81 -subtype=_wrx -clib=wrxansi -pragma-redirect:ansifont=_oemascii -pragma-define:ansifont_is_packed=0 -pragma-define:ansicolumns=80 -lndos -lm81 -O3 -create-app ansitest.c
/tmp/tmpzccVK4UheEj/zcc_opt.def:6: error: undefined symbol: _oemascii
  ^---- _oemascii
/tmp/tmpzccVK4UheEj/zcc_opt.def:6: error: undefined symbol: ansifont
  ^---- _oemascii
```

Which I guess is the error you're getting?

At some point z80asm became case sensitive, so I'm wondering if you're getting hit by that? Check the case of the symbol within your source file.

SM7I, 2024-12-14:
Exactly, I´m getting following error:

/tmp/tmpzccXXxHN5zs/zcc_opt.def:6: error: undefined symbol: _oemascii
  ^---- _oemascii
/tmp/tmpzccXXxHN5zs/zcc_opt.def:6: error: undefined symbol: ansifont
  ^---- _oemascii

So you´re saying that someway along the road they changed to capital letters och other casing ? I´´m using all small case letters. How can I find out how and if it´s big or small case in the clib ?

suborb, 2024-12-14:
I'm saying that the oemascii symbol (in some case) is defined in your code! I've had a search of the 1.99 code base and there's nothing there.

SM7I, 2024-12-14:
No, it´s not, but I can send you my code if you want to have a look ?

suborb, 2024-12-14:
That’s weird then! Please do, dom /at/ suborbital org uk

suborb, 2024-12-25:
I'm going to close this one, the font was nowhere to be found in the source sent or in the z88dk repo.

However, the font is used in other projects that used z88dk for example: https://github.com/FujiNetWIFI/fujinet-apps/blob/2ee1e015312d18d419fd018c9b01f258fec1e25a/netcat/adam/old/src/font.asm#L4

closed 2024-12-25 by suborb