z88dk/z88dk #2655: [question] How do I force z88dk assembler to compile Z80 assembly strictly using Z80 original instructions?

fitch, 2024-12-17:
I'm using [z88dk](https://z88dk.org/site/download) to compile Z80 assembly code into binary code to be used in a 80s computer.

If I use this kind of assembly code:

```
public _main            ; This entry point definition is required by
z88dk
_main:
    org 0x0000

    ld hl, 0x8000
    ld a, 1
    add hl, a           ; Not a correct Z80 instruction
```
... where I use an invalid Z80 instruction add hl, a, and compile it like:
```
z88dk-z80asm test.asm z88dk-z80asm -b test.o
```
... and then disassemble the test.bin binary back to assembly language, the assembler seems to have replaced the add hl, a instruction with a call to a subroutine that actually does "add hl, a" with standard Z80 instructions:
```
0000 ORG_0000:
0000              ld   hl,8000h     ; 32768, -32768
0003              ld   a,01h    ; 1
0005              call .org_0000_l  ; 0008h
0008 .org_0000_l:
0008              push af     
0009              add  a,l    
000A              ld   l,a    
000B              ld   a,h    
000C              adc  a,00h    ; 0
000E              ld   h,a    
000F              pop  af     
0010              ret         
```
How can I turn this behavior off in z88dk-z80asm or zcc?

feilipu, 2024-12-17:
See #2590

closed 2024-12-17 by feilipu

fitch, 2024-12-17:
Thanks for the explanation, I understand this design principle. Maybe I'm "using z88dk wrong" in here as - at least for now - using it as a vanilla assembler and also applying the "learning by doing" principle.

 In my use case, I learned from this specific example (the hard way of debugging it for 2 hours) that the compiler also accepts these synthetic instructions and doesn't warn about using them unintentionally.

feilipu, 2024-12-17:
I added to Paulo's comment on #2590 the thread.

Too late now, because you already understand what's going on, but using `-l` to read the assembled results will let you see what the assembler has done, directly.  That's the best feedback.

My personal hint is to keep the clrhome tables open while programming, as flag setting across the instruction set is not consistent and it always remains a trap.

pauloscustodio, 2025-05-01:
Closed through pull request #2707