z88dk/z88dk #2657: [bug][appmake/ti83papp] ROM builds fail to append DATA section

chandlerwr, 2024-12-27:
I discovered this issue trying to work with the graphics function [`putsprite`](https://github.com/z88dk/z88dk/blob/master/libsrc/target/ticalc/graphics/putsprite.asm), which uses self-modifying code for the or/and/xor modes.

For reproducibility, the following code works fine when compiled without `-subtype` (RAM program) but displays a strange cut-up sprite when compiled via `zcc +ti83p -create-app -startup=0 -subtype=app_one_page sprite.c` (flash app):
```C
#pragma string name Sprite
#pragma output nostreams

#include <stdio.h>
#include <games.h>
#include <graphics.h>

#define SCREENX 96
#define SCREENY 64

const unsigned char test_rmajor[] = { 16, 4,
    0xFF, 0xFF,
    0x00, 0x00,
    0xF0, 0xF0,
    0xFF, 0xFF,
};

main() {
    clg();
    putsprite(SPR_XOR, 0, 0, test_rmajor);
    getk();
}
```

## No creation of _DATA.bin file
The first problem is that [`ti83papp_crt0.crt`](https://github.com/z88dk/z88dk/blob/master/lib/target/ti83p/classic/ti83papp_crt0.asm) doesn't declare `CRT_MODEL = 1` to indicate that the program is being built for ROM. The entire app ends up in ROM and the self-modifying code cannot modify itself.

Please let me know if I should open this as a separate issue though since it's only half of the problem. I managed to work around it by using my own `cfg` and `crt` files (can provide if helpful) which *also* change/declare `CRT_ORG_BSS`, `__crt_org_bss`, and `__crt_model` (didn't work without the latter two for reasons unknown to me).

What got the job done were the lines:
```asm
defc CRT_ORG_BSS = appBackUpScreen	; = $9872
defc CRT_MODEL = 1
defc __crt_org_bss = CRT_ORG_BSS	; originally 8A3A but there's more space at 9872 and I don't use gray83p.asm
defc __crt_model = CRT_MODEL
```

## Appmake issue
However, making the above changes to the `crt` file causes the compiled app to crash the calculator (actually, not for the above example but it certainly doesn't work).

According to the `.map` file for the above code, `__ROMABLE_END_tail = $42CE` which is where the DATA segment (to be copied to RAM) begins according to the snip from `crt_section.inc`:
https://github.com/z88dk/z88dk/blob/b14985e5d24885d2ba88085f6309edab656a39fb/lib/crt/classic/crt_section.inc#L58-L64

The first few bytes of `sprite_DATA.bin` are `00 21 02 00 39 5e 23 56` compared to a ROM snippet from the calculator:
`42cd: 01 02 2d 40 2d 0f c3 60`

`02 2d` are actually the two starting bytes for the *app signature* from in `ti8xk.c`:
https://github.com/z88dk/z88dk/blob/b14985e5d24885d2ba88085f6309edab656a39fb/src/appmake/ti8xk.c#L718-L720

Which would indicate that `size` isn't including the _DATA.bin file. The file isn't even being read by `fopen_bin` since deleting `sprite_DATA.bin` and running `z88dk-appmake` separately doesn't throw the error in `appmake.c`:
https://github.com/z88dk/z88dk/blob/b14985e5d24885d2ba88085f6309edab656a39fb/src/appmake/appmake.c#L328-L333

All of that to say that the problem lines seem to be:
https://github.com/z88dk/z88dk/blob/b14985e5d24885d2ba88085f6309edab656a39fb/src/appmake/ti8xk.c#L561-L563
https://github.com/z88dk/z88dk/blob/b14985e5d24885d2ba88085f6309edab656a39fb/src/appmake/appmake.c#L302-L303

chandlerwr, 2024-12-27:
Originally I did try manually appending `sprite_DATA.bin`, but in testing just now I did find a workaround. Just appending `sprite_DATA.bin` to `sprite.bin` hits the following lines of `appmake.c`:
https://github.com/z88dk/z88dk/blob/b14985e5d24885d2ba88085f6309edab656a39fb/src/appmake/appmake.c#L293-L294
**But** if you combine the files into a different name like `sprite_COMB.bin` instead of `sprite.bin`, it works properly.
So on Windows:
`copy /B sprite.bin+sprite_DATA.bin sprite_COMB.bin` instead of `copy /B sprite.bin+sprite_DATA.bin sprite.bin`
Odd.

suborb, 2024-12-27:
The Ti83+ code was donated and I've never been able to get a working test environment.

I've just pushed a couple of changes which I think will fix the problem - that is "switching" to ROM mode with `CRT_MODEL` and a passing through crtfile to `fopen_bin` so that it will figure CRT_MODEL and append the data section.

chandlerwr, 2024-12-29:
**The fix certainly worked**, but only for my own CRT code. I dug into it a little bit, but I think this is probably a separate issue involving memory usage in TI programs.

[ti83papp_crt0](https://github.com/z88dk/z88dk/blob/master/lib/target/ti83p/classic/ti83papp_crt0.asm) includes [intwrap83p](https://github.com/z88dk/z88dk/blob/master/lib/target/ti83p/classic/intwrap83p.asm) which in turn includes [int83p](https://github.com/z88dk/z88dk/blob/master/lib/target/ti83p/classic/int83p.asm). This is a new project for me so I don't understand a lot of the interrupt code, but I see it uses `statVars` (`$8A3A`) to store a significant bit of information/code. Specifically, these lines end up overwriting the DATA section in a TI flash app: https://github.com/z88dk/z88dk/blob/5c4075d8e9b681cc298362df4d23e1ebda6e5651/lib/target/ti83p/classic/int83p.asm#L30-L37

The usage of `statVars` is a problem since the original CRT uses `statVars` for `CRT_ORG_BSS` and that's where the whole DATA section (100+ bytes or more) gets copied to for a flash app. My immediate thought would be to move `CRT_ORG_BSS`, but I'm not sure if that would cause issues with [some graphics code](https://github.com/z88dk/z88dk/blob/master/lib/target/ti83p/classic/gray83p.asm).

#
Feel free to close this issue though as the `copy /B` workaround isn't necessary anymore. I'd be happy to move the information above to a new issue, but I'll probably stick to using my own CRT with `CRT_ORG_BSS = appBackUpScreen` (`$9872`) now that I have it setup.

Oh and test-wise, Cemetech has a [great emulator](https://www.cemetech.net/projects/jstified/) that I've been running on my TI-84+ ROM. There are also downloadable ones like TilEm as well if you were curious.

suborb, 2025-01-14:
Apologies. I've been pondering on this one on and off.

Thanks for your research into this, I think moving CRT_ORG_BSS to appBackupScreen as you do does make the most sense. 

It looks like the greyscale display wasn't usable for apps, so there's no loss. And now we get an error if you try:

```
% zcc +ti83p -subtype=app_one_page graygfx.c
/Users/dom/z88dk/lib/config//../..//lib/target/ti83p/classic/ti83papp_crt0.asm:207: error: undefined symbol: GRAY_LIB_NOT_AVAILABLE_FOR_TI83P_APP
  ^---- defs GRAY_LIB_NOT_AVAILABLE_FOR_TI83P_APP
/Users/dom/z88dk/lib/config//../..//lib/target/ti83p/classic/ti83papp_crt0.asm:207: error: constant expression expected
  ^---- defs GRAY_LIB_NOT_AVAILABLE_FOR_TI83P_APP
```

Which is feels like a good conclusion.

closed 2025-01-18 by suborb