z88dk/z88dk #2658: Please test SDCC 4.5.0 RC1

spth, 2025-01-08:
Since z88dk uses SDCC, and SDCC is working on a SDCC 4.5.0 release (there is an RC1 since about a week ago), it probably would be good to test, and report (here or at the sdcc-user list) how bad/well SDCC 4.5.0 RC1 works for z88dk.

feilipu, 2025-01-08:
"My personal idea for 4.5.0 is for it to be mostly a better 4.4.0: fix bugs found in 4.4.0, and try to avoid regressions. In general, IMO, 4.4.0 was already a good release wrt. bugs."

Am I right in assuming this comment represents the general perspective on changes in 4.5.0?

Anything that needs to be specifically tested coming to mind?

feilipu, 2025-01-09:
Our patch (which is now quite minimal) applies cleanly, and there are no build issues.

I'll try some testing over the next few days, but prima facie there are no blockers from our side.

spth, 2025-01-15:
Since today, there is an RC2.

spth, 2025-01-19:
We recently found that z80 puts (at least some) nonbanked functions into` _HOME` instead of `_CODE` since 3.5.0, and are considering changing that back to `_CODE` for 4.5.0. Would that break anything for you?

feilipu, 2025-01-19:
I haven't seen any difference in behaviour from 4.4.0, so if using `_HOME` is new then I've not triggered it with my simple tests.
So, it is a bit difficult to comment knowledgeably.

However, I don't think we handle `_HOME` sections currently, so it would be best if the previous behaviour was restored.

spth, 2025-01-19:
Well, typically systems place _CODE next to _HOME, with both in ROM, so it tends to work wither way. Do you use the keyword __nonbanked?

feilipu, 2025-01-19:
@suborb thoughts?

feilipu, 2025-01-19:
> Do you use the keyword `__nonbanked`?

Yes, defined as `NONBANKED` throughout the Gameboy library header. But the actual functions are in assembly and are placed in sections more closely.

https://github.com/z88dk/z88dk/blob/master/include/arch/gb/gb.h

Also, I think that since the z88dk linker is different to sdcc,then the `_HOME` placement issue does not occur.

feilipu, 2025-01-20:
> Do you use the keyword __nonbanked?

Philipp @spth just following up on this with an example. In the case of Gameboy, which is the original use case, we're using sccz80 exclusively so issues with sdcc don't affect z88dk.

BUT. Building a contrived example using the CP/M target using sdcc it is possible to produce a problematic result. If a function is declared with `__nonbanked` then it will be put into the section `IGNORED` (which is [derived from `_HOME](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_opt.1#L318-L320)`) and it will be ignored by the z88dk linker, resulting in a failure.

So even though this is an unlikely scenario, there is an issue that we'll have to handle somehow.

```c
// zcc +cpm -SO2 --list -m bug-2658.c -o bug-2658.com

#include <stdint.h>
#include <stdlib.h>
#include <stdio.h>

#pragma printf = "%s %c %04X"     // enables  %s %c %X only

int32_t testBug( int16_t input) __nonbanked;

int32_t testBug( int16_t input)
{
    return (int32_t)input;
}

void main (void)
{
    for (uint16_t test = 0x0001; test != 0x0000; ++test) {
        fprintf(stdout, " Input: %04X  Output: %04X %04X\n", test, (uint16_t)(testBug(test)>>16), (uint16_t)testBug(test));
    }
}
```

listing excerpt

```
   521                          
   522                          ;--------------------------------------------------------
   523                          ; absolute external ram data
   524                          ;--------------------------------------------------------
   525                          	SECTION IGNORE
   526                          ;--------------------------------------------------------
   527                          ; global & static initialisations
   528                          ;--------------------------------------------------------
   529                          	SECTION code_crt_init
   530                          ;--------------------------------------------------------
   531                          ; Home
   532                          ;--------------------------------------------------------
   533                          	SECTION IGNORE
   534                          ;	---------------------------------
   535                          ; Function testBug
   536                          ; ---------------------------------
   537                          _testBug:
   538  0000  dde5              	push	ix
   539  0002  dd210000          	ld	ix,0
   540  0006  dd39              	add	ix,sp
   541  0008  dd6e04            	ld	l,(ix+4)
   542  000b  dd7e05            	ld	a,(ix+5)
   543  000e  67                	ld	h, a
   544  000f  07                	rlca
   545  0010  9f                	sbc	a, a
   546  0011  5f                	ld	e, a
   547  0012  57                	ld	d, a
   548  0013  dde1              	pop	ix
   549  0015  c9                	ret
   550                          	SECTION IGNORE
   551                          ;--------------------------------------------------------
   552                          ; code
   553                          ;--------------------------------------------------------
   554                          	SECTION code_compiler
   555                          ;	---------------------------------
[... 44 lines cut ...]
   600                          	SECTION rodata_compiler
   601                          ___str_0:
   602  0000  20496e7075743a20  	DEFM " Input: HL %04X  Output: %04X %04X"
              484c202530345820  
              204f75747075743a  
              2025303458202530  
              3458              
   603  0022  0a                	DEFB 0x0a
   604  0023  00                	DEFB 0x00
   605                          	SECTION IGNORE
```

map excerpt

```
_stderr                         = $0C13 ; addr, public, , cpm_crt_asm_m4, data_stdio, /home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/cpm/cpm_crt.asm.m4:1139
_stdin                          = $0BF1 ; addr, public, , cpm_crt_asm_m4, data_stdio, /home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/cpm/cpm_crt.asm.m4:901
_stdlst                         = $0C46 ; addr, public, , cpm_crt_asm_m4, data_stdio, /home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/cpm/cpm_crt.asm.m4:1444
_stdout                         = $0C02 ; addr, public, , cpm_crt_asm_m4, data_stdio, /home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/cpm/cpm_crt.asm.m4:1027
_stdpun                         = $0C35 ; addr, public, , cpm_crt_asm_m4, data_stdio, /home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/cpm/cpm_crt.asm.m4:1330
_stdrdr                         = $0C24 ; addr, public, , cpm_crt_asm_m4, data_stdio, /home/phillip/Z80/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/cpm/cpm_crt.asm.m4:1216
_testBug                        = $0000 ; addr, public, , bug_2658_c, IGNORE, bug-2658.c:537
acon_tbl_printf                 = $0766 ; addr, local, , cpm_crt_asm_m4, code_stdio, /home/phillip/Z80/z88dk/lib/config/../../libsrc/_DEVELOPMENT/target/cpm/../../stdio/z80/asm_vfprintf_unlocked_iy.asm:715
alpha                           = $05C9 ; addr, local, , l_num2char, code_l, l/util/9-common/ascii/l_num2char.asm:24
```

feilipu, 2025-01-22:
#2661 removes any further concern with `__nonbanked` declarations used with sdcc. They're now handled safely.

spth, 2025-01-25:
We have RC3 today, containing multiple fixes (most were about the documentation, but one was a codegen bug that could affect z80).

spth, 2025-01-28:
SDCC 4.5.0 was released today.

feilipu, 2025-01-28:
#2659 has been revised to r15248. Once the test suite passes we will merge and migrate to v4.5.0 too.

PR #2659 (2025-01-29): zsdcc - update to 4.5.0 r15248
A patch to v4.5.0-rc1 (r15212) for zsdcc, for testing.

Will do a final version once v4.5.0 is released.

Closes #2658
Closes #2635
closed 2025-01-29 by suborb