z88dk/z88dk #2660: zsdcc - make __nonbanked function declaration safe

feilipu, 2025-01-21:
Related to #2658.

If a function is declared with the modifier `__nonbanked` and compiled with zsdcc, the resulting function will be placed into a section `_HOME`, which is [redefined to section `IGNORED`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/sdcc_opt.1#L318-L320). At this point the assembly related to that function will be ignored by the z88dk linker, leading to a silent failure (restart from `0x0000`.

This has been the situation for some (24?) years with sdcc, so I guess people are aware of it. But, possibly either a warning message could be generated, or something sensible could be done with the functions defined as `__nonbanked`, rather than just silent failure.

One suggestion for improvement would be to modify the `sdcc_opt.1` redefinition of `_HOME` sections into a section called `code_compiler_home` or `code_home` or similar, and then add that section to the default memory map.

```
%"[ \t]*"0.%"(AREA|area)"9%"[ \t]+"1_HOME
=
%0SECTION%1code_home
```

Is this worth doing?

Thoughts on the best way to do it?

feilipu, 2025-01-21:
I've done a PR for a simple fix for zsdcc.

I can't find where sccz80 handles `__nonbanked`.
For consistency sccz80 should probably also handle `__nonbanked` if it doesn't already.

suborb, 2025-01-24:
Odd, sccz80 handles `__nonbanked` here: https://github.com/z88dk/z88dk/blob/master/src/sccz80/declparse.c#L696

I thought I remember you mentioning a parsing failure though?

feilipu, 2025-01-24:
> I thought I remember you mentioning a parsing failure though?

Only with zsdcc, where someone used the `__nonbanked` modifier and it would ignore code thus described.

With sccz80 the `__nonbanked` modifier is ignored, which means that (for the GB target specifically) if someone wants their code to be placed in the home section, there is no control provided by sccz80. That is the issue (enhancement).

In the thread on the linked sdcc svn the discussion from the GB side on `__banked` vs. `__nonbanked` was "heated" and it seems being able to place code into home or banked memory via the home section is important.

Code developed with dbdk-2020 will have these modifiers to aid placement into memory, so  it should be possible to compile that code with sccz80 and get similarly working results.

suborb, 2025-01-24:
Gotcha. I can sort that out easily I think now you’ve defined the sections.

feilipu, 2025-01-25:
Thanks. That looks good.

I'm tempted to learn a little more on how the GB banking and sections are assigned by gbdk-2020. To see whether it is possible to do something similar here.

closed 2025-01-25 by feilipu