z88dk/z88dk #2662: ticks - implement 8085 k bit jumps

feilipu, 2025-01-27:
Currently implementing sccz80 `l_` comparison improvements for the 8085 using`sub hl,bc` and using the `jp k,nnnn` and jp nk,nnnn` instructions.

But ticks doesn't implement these instructions, so I'm testing on 8085 hardware.
Before I can merge the results, ticks needs to have this support to enable the tests to pass.

The implementation can be derived from the information provided by [Ken Shirriff's reverse engineering blog](https://www.righto.com/2013/02/looking-at-silicon-to-understanding.html). Specifically:

> One mystery was the purpose of the K flag: "It does not resemble any normal flag bit."[[1]](https://www.righto.com/2013/02/looking-at-silicon-to-understanding.html#ref1) Its use for increment and decrement is clear, but for arithmetic operations why would you want the exclusive-or of the overflow and sign? It turns out the the K flag is useful for signed comparisons. __If you're comparing two signed values, the first is smaller if the exclusive-or of the sign and overflow is 1__. This is exactly what the K flag computes.

suborb, 2025-01-29:
This is implemented in #2663

feilipu, 2025-01-29:
I was wondering how the `inc nn` instructions had been passing tests using `jp k,nnnn`, then I realised the 8085 `crt0` is not tested...

Anyway, that's enough to get this over the line. Thanks.

feilipu, 2025-01-30:
Just looking at the test results, we're not getting failures, but also not successes.
Is that expected when using the "thin wedge" `k` flag solution?

```sh
zcc +test -vn -clib=8085  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c rshift.c -o test_8085_rshift.bin   -m
 00:31:16  Starting 8085 test 
Starting suite Right shift Tests (6 tests)
6 run, 0 passed, 0 failed
Ticks: 154769
 00:31:16  8085 test completed 
zcc +test -vn -clib=8085  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c lshift.c -o test_8085_lshift.bin   -m
 00:31:16  Starting 8085 test 
Starting suite Left shift Tests (4 tests)
4 run, 0 passed, 0 failed
Ticks: 152828
 00:31:16  8085 test completed 
zcc +test -vn -clib=8085  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c compare.c -o test_8085_compare.bin   -m
 00:31:17  Starting 8085 test 
Starting suite Compare Tests (11 tests)
11 run, 0 passed, 0 failed
Ticks: 178985
 00:31:17  8085 test completed 
zcc +test -vn -clib=8085  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c compare0.c -o test_8085_compare0.bin   -m
 00:31:18  Starting 8085 test 
Starting suite Compare0 Tests (6 tests)
6 run, 0 passed, 0 failed
Ticks: 151514
 00:31:18  8085 test completed 
zcc +test -vn -clib=8085  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c compare_const.c -o test_8085_compare_const.bin   -m
 00:31:19  Starting 8085 test 
Starting suite Compare1 Tests (6 tests)
6 run, 0 passed, 0 failed
Ticks: 151514
 00:31:19  8085 test completed 
zcc +test -vn -clib=8085  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c compare_mconst.c -o test_8085_compare_mconst.bin   -m
 00:31:19  Starting 8085 test 
Starting suite Compare-1 Tests (6 tests)
6 run, 0 passed, 0 failed
Ticks: 152599
 00:31:19  8085 test completed 
zcc +test -vn -clib=8085  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c mult.c -o test_8085_mult.bin   -m
 00:31:20  Starting 8085 test 
Starting suite Multiplication Tests (5 tests)
5 run, 0 passed, 0 failed
Ticks: 157596
```

suborb, 2025-01-30:
Good spot. That’s not good.

feilipu, 2025-01-30:
> In this case thin wedge means I just implemented it for sub hl,bc and didn’t touch what’s been done for other instructions.

The `sub hl,bc` `jp k,nnnn` `jp nnnn` set is being done for every signed comparison now.  For unsigned comparison it is `sub hl,bc` `jp c,nnnn` `jp nnnn`.  There's a `jp z,nnnn` used where appropriate to catch specific equality cases where appropriate.

I suspect that there's a comparison operation being done at the start of every test that is now not working as expected, which leads to the test bailing but not actually failing.

feilipu, 2025-01-30:
I've done some tracing with my little test program, and it behaves similarly to the test suite programs.

In summary, the first `l_le_hlbc` instruction from the first `for` loop is returning `false`, and this is wrong.
It causes the program to bail (`jp nc,i_3`) without doing anything.

```c
// zcc +test -clib=8085 -O2 --list -m sub_comparison.c -o subcomp.bin
// z88dk-ticks -m8085 subcomp.bin

#include <stdint.h>
#include <stdlib.h>
#include <stdio.h>

#pragma printf = "%s %c %d"     // enables  %s %c %d only

void main (void)
{
    for (int16_t testA = -4; testA <= 4; ++testA) {
        for (int16_t testB = -4; testB <= 4; ++testB) {
            fprintf(stdout, " Input: A %2d B %2d  A<B %d  A<=B %d  A>=B %d  A>B %d\n", testA, testB, (testA < testB), (testA <= testB), (testA >= testB), (testA > testB));
        }
    }
}
```

Excerpt from the listing.

```asm
    18                          ._main
    18  0000  21fcff            	ld	hl,65532	;const
    18  0003  e5                	push	hl
    18  0004  c30a00            	jp	i_4	;EOS
    18                          .i_2
    18  0007  e1                	pop	hl
    18  0008  23                	inc	hl
    18  0009  e5                	push	hl
    18                          .i_4
    18  000a  e1                	pop	hl
    18  000b  e5                	push	hl
    18  000c  444d              	ld	bc,hl
    18  000e  210400            	ld	hl,4
    18  0011  cd0000            	call	l_le_hlbc	;bc<=hl
    18  0014  d28000            	jp	nc,i_3	;
    18  0017  21fcff            	ld	hl,65532	;const
    18  001a  e5                	push	hl
    18  001b  c32100            	jp	i_7	;EOS

    18                          .i_6
    18  007c  c1                	pop	bc
    18  007d  c30700            	jp	i_2	;EOS
    18                          .i_3
    18  0080  c1                	pop	bc
    18  0081  c9                	ret
```
The important parts of the trace look like this.

```
                    ld        b,h                           ;[028a] 44

af $0000   bc $FF01   de $0000   hl $FFFC   ix $0000    S:0 Z:0 H:0 P/V:0 N:0 C:0
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $028B  [pc]$  4D   sp $FEB6  [sp]$FFFC
CLOCKH $C103   CLOCKL $C200

                    ld        c,l                           ;[028b] 4d

af $0000   bc $FFFC   de $0000   hl $FFFC   ix $0000    S:0 Z:0 H:0 P/V:0 N:0 C:0
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $028C  [pc]$  21   sp $FEB6  [sp]$FFFC
CLOCKH $C103   CLOCKL $C200

                    ld        hl,$0004                      ;[028c] 21 04 00

af $0000   bc $FFFC   de $0000   hl $0004   ix $0000    S:0 Z:0 H:0 P/V:0 N:0 C:0
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $028F  [pc]$  CD   sp $FEB6  [sp]$FFFC
CLOCKH $C103   CLOCKL $C200

                    call      $0256                         ;[028f] cd 56 02

af $0000   bc $FFFC   de $0000   hl $0004   ix $0000    S:0 Z:0 H:0 P/V:0 N:0 C:0
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $0256  [pc]$  08   sp $FEB4  [sp]$0292
CLOCKH $C103   CLOCKL $C200

                    sub       hl,bc                         ;[0256] 08

af $0111   bc $FFFC   de $0000   hl $0008   ix $0000    S:0 Z:0 H:1 P/V:0 N:0 C:1
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $0257  [pc]$  FD   sp $FEB4  [sp]$0292
CLOCKH $C103   CLOCKL $C200

                    jp        k,$023b                       ;[0257] fd 3b 02

af $0111   bc $FFFC   de $0000   hl $0008   ix $0000    S:0 Z:0 H:1 P/V:0 N:0 C:1
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $023B  [pc]$  AF   sp $FEB4  [sp]$0292
CLOCKH $C103   CLOCKL $C200

                    xor       a                             ;[023b] af

af $0146   bc $FFFC   de $0000   hl $0008   ix $0000    S:0 Z:1 H:0 P/V:1 N:1 C:0
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $023C  [pc]$  6F   sp $FEB4  [sp]$0292
CLOCKH $C103   CLOCKL $C200

                    ld        l,a                           ;[023c] 6f

af $0146   bc $FFFC   de $0000   hl $0000   ix $0000    S:0 Z:1 H:0 P/V:1 N:1 C:0
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $023D  [pc]$  67   sp $FEB4  [sp]$0292
CLOCKH $C103   CLOCKL $C200

                    ld        h,a                           ;[023d] 67

af $0146   bc $FFFC   de $0000   hl $0000   ix $0000    S:0 Z:1 H:0 P/V:1 N:1 C:0
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $023E  [pc]$  C9   sp $FEB4  [sp]$0292
CLOCKH $C103   CLOCKL $C200

                    ret                                     ;[023e] c9

af $0146   bc $FFFC   de $0000   hl $0000   ix $0000    S:0 Z:1 H:0 P/V:1 N:1 C:0
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $0292  [pc]$  D2   sp $FEB6  [sp]$FFFC
CLOCKH $C103   CLOCKL $C200

                    jp        nc,$02fe                      ;[0292] d2 fe 02

af $0146   bc $FFFC   de $0000   hl $0000   ix $0000    S:0 Z:1 H:0 P/V:1 N:1 C:0
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $02FE  [pc]$  C1   sp $FEB6  [sp]$FFFC
CLOCKH $C103   CLOCKL $C200

                    pop       bc                            ;[02fe] c1

af $0146   bc $FFFC   de $0000   hl $0000   ix $0000    S:0 Z:1 H:0 P/V:1 N:1 C:0
af'$0040   bc'$0000   de'$0000   hl'$0000   iy $0000
pc $02FF  [pc]$  C9   sp $FEB8  [sp]$0106
CLOCKH $C103   CLOCKL $C200

                    ret                                     ;[02ff] c9

```

suborb, 2025-01-31:
Looks like I used the lsb sign, not the msb sign which should have been used.

feilipu, 2025-02-01:
Test results look normal now, so I think this is done!

closed 2025-02-01 by feilipu