z88dk/z88dk #2664: [g800] PC-G815 graphics

fu-sen, 2025-01-28:
Sharp PC-G series graphics also support PC-G815.
(This is why there is a PC-G815 selection for emulator g800)
[image]
But it seems that the current z88dk monochrome graphics does not support this. (Confirmed with Nightly build 20250127-08765eb48b-23073)

Note: PC-G815 has a graphics size of **144x32**. (Text 24x4)

suborb, 2025-01-29:
I don't think I've every worked on these models, but looking at the emulator source code and our code to drive the LCD it looks like we'd need to change the ports.

Eg: https://github.com/z88dk/z88dk/blob/master/libsrc/target/g800/graphics/plotpixel.asm

The `in a,(0x41)` would be `in a,(0x57)` on the G815. I suspect that there will be other ports to change as well.

fu-sen, 2025-02-11:
I learned that PC-E200 and PC-G series, excluding PC-G815 and PC-G850, can use graphic images in machine language.
[image]

Therefore, it will be necessary to change the target to all PC-E200 and PC-G series.

fu-sen, 2025-02-11:
PC-E220 documentation:
https://wwwhomes.uni-bielefeld.de/achim/pc-e220.html

Japanese documentation (E200, G815 and G850)
https://web.archive.org/web/20221020084355/https://ha9.seikyou.ne.jp/home/akihi/pdf/C64_PocketBook.pdf

PC-G815 has different LED access. Therefore, it will be necessary to add options.

fu-sen, 2025-02-16:
Since I understood the specifications of the LED port, I created a sample program.

PC-E200 / G siries (Other than PC-G815, PC-G850)
[image]
0x58 setting, 0x59 busy, 0x5a writing, 0x5b reading
https://gist.github.com/fu-sen/7bca13a20bee5ebca75f3c416f720ed9

PC-G815
[image]
0x54 setting, 0x55 busy, 0x56 writing, 0x57 reading
0x58 setting, 0x59 busy, 0x5a writing, 0x5b reading
https://gist.github.com/fu-sen/1850fa2f22f7ffde9a3d65a625f4d565

The right half is flipped horizontally. (Note the ribbon on the balloon in the image above)
Furthermore, with PC-G815, you need to change the port number depending on the location.
Port access is much more troublesome than on the PC-G850...

zx70, 2025-02-26:
Hello fu-sen, thankyou for reporting the technical detail and two clear working examples !
To recap, the two models require their unique way to deal with the display ports.    The graphics functions support for those two models does not exist in z88dk.

If I remember correctly, the other models do not have a mirrored picture on the right half of the display.   It means it is easier to provide specific code than extend the existing core functions.

I'm thinking at a way to limit the code redundancy first, we have this interesting trick in {z88dk}/libraries/Makefile:
    $(MAKE) -C gfx TARGET=xxx FLAVOUR="narrow" SUBTYPE=yyyy
,I wonder if there is a way to use SUBTYPE on conditional code.

By the way, your suggestion is too good, I'm trying to get something working, first.

zx70, 2025-03-07:
I'm not sure this is the best path, but I chose to start making the G850 code more flexible.
'locate_850.asm' now holds the display addressing core for the G850, 'locate.asm' is buggy and does not work yet.
Hopefully providing two new 'locate' function sets should now suffice to port the graphics on the G815 and on the E200.

zx70, 2025-03-08:
I got to a first conclusion, to drive the G815 display there are too many variables to make it with just the registers, I will try to get some nasty SMC code with an even nastier part on top split into 4 sections, depending on which portion of the display we are writing on.
Hopefully the E200/G family is a little simpler, a clean implementation would require modulus 5 and modulus 7 computations, but we'll first see what comes up keeping the things simple.

zx70, 2025-03-08:
Ok I pulled a first experimental implementation for the E200 series, the display boundaries are not protected yet, so it is very unstable

zx70, 2025-03-10:
On the G815 example you wrote:

/*
scroll (0-63, line*8?)
outp (0x56, 0xc0+y);
outp (0x5a, 0xc0+y);

busy check
while (inp(0x55) & 0x80);
while (inp(0x59) & 0x80);

read
d = inp(0x59):
d = inp(0x5b):
*/

..there is something strange here, 
considering that the output ports are 0x56 and 0x5a, shouldn't it be:

busy check
while (inp(0x54) & 0x80);
while (inp(0x59) & 0x80);

read
d = inp(0x55):
d = inp(0x5b):

?

fu-sen, 2025-03-10:
You may not have noticed the mistake in my source:
https://gist.github.com/fu-sen/1850fa2f22f7ffde9a3d65a625f4d565

```
/*
scroll (0-63, line*8?)
outp (0x56, 0xc0+y);
outp (0x5a, 0xc0+y);

busy check
while (inp(0x55) & 0x80);
while (inp(0x59) & 0x80);

read
d = inp(0x57):
d = inp(0x5b):
*/
```

Correctly:
```
d = inp(0x59):
d = inp(0x5b):
```
|port|left & right|center|
|---|---|---|
|setting|0x54|0x58|
|busy(& 0x80)0x55|0x59|
|data read|0x56|0x5a|
|data write|0x57|0x5b|

fu-sen, 2025-03-10:
I'm ready for testing...
[image]
I have prepared these ROM images.

zx70, 2025-03-10:
Wonderful !!

>  By the way I think I nailed it...
errm, not totally, I just understood what is going on with the scroll register, It's also necessary to move the scroll reset on clg();

Coming soon

fu-sen, 2025-03-11:
For PC-E200 (other than G815 and G850), out 0x58, 0x7c changes the top and bottom status display.
(For PC-G815, out)0x58,0x7f / For PC-G850, X position 0xc0)
Is this what you're concerned about?

fu-sen, 2025-03-12:
Since I was creating source code that used monochrome sprites, I tried the latest nightly build:
https://gist.github.com/fu-sen/4e079229d43a0aa6c9d40734c35fb2ab

```
zcc +g800 -lm -create-app -clib=g850b -bn sin sin.c
```
[image]

```
zcc +g800 -lm -create-app -clib=g815 -bn sin sin.c
```
[image]

```
zcc +g800 -lm -create-app -clib=e200 -bn sin sin.c
```
[image]

Wonderful!! 😄
Some sources are already working as expected.

fu-sen, 2025-03-12:
Maybe you are doing something unintended when you use scrolling.
0x790d contains the current scroll value (in lines)
However, since the value is actually specified in dots, it is multiplied by 8.
- PC-G850: 0x40, 0x40-0x7f
- PC-G815: 0x54, 0xc0-0xff and 0x58, 0xc0-0xff 
- Other models: 0x58, 0xc0-0xff 

In fact, you can scroll in dot increments, but the g800 only supports the PC-G850.
I have informed the author of the g800, so I think they will take action. 😆

zx70, 2025-03-12:
DSTAR looks on The PC-G815 looks interesting, here's the binary content in HEX:

:10010000ED731701CD1B0121C0FF39F9ED73591AA9
:10011000CDB201CD2901310000C9E9AF218B190110
:10012000D300118C1977EDB0C9C97B2F5F7A2F5797
:1001300013C97D2F6F7C2F6723C9EB1DF8CB2CCB08
:100140001DC33B016069C3910C3923237E6F079F58
:1001500067E3E93923237E23666FE3E9EBE14E236E
:10016000462378B1280C7E23BB7E2320F1BA20EEF3
:100170006069E92323237E6F079F67C92323237EBA
:1001800023666FC9210600397E23666FC9210800E6
:10019000397E23666FC9210A00397E23666FC9E95B
:1001A0007CB5210000C0372CC9C1D1C57D12137C9C
:1001B00012C92100007D329019CDED02CD66047C7C
:1001C000B5C8CDC901C3BC01C9C5C53A8F19A7CAF5
:1001D000D801218E19C3DB01218D19D1C1E5D5CDFF
:1001E0006D06CD5C0126024100260261003B0251F2
:1001F000003B027100520250005202700067024F31
:100200000067026F007E0220009F0247009F026786
:1002100000A5022B00C9022D00E7024800E7026892
:10022000000000C3EA02210200CD5301210000E5D5
:1002300023E5CDD204C1C1C1C3EA02210200CD53DE
:1002400001210000E521FFFFE5CDD204C1C1C1C3FA
:10025000EA02210200CD5301210100E52BE5CDD2B8
:1002600004C1C1C1C3EA02210200CD530121FFFF35
:10027000E5210000E5CDD204C1C1C1C3EA02218F4E
:1002800019CD76017DEE016F7DCD77017D328F191D
:10029000CD6D06112000A7ED52CA9002C3EA0221DB
:1002A0000000CD6E0C219019CD7601237D3290197E
:1002B000FE19C2C302219019CD76012B7D3290190F
:1002C000C3EA02CDED02C3EA02219019CD76012BDB
:1002D0007D329019FEFFC2E702219019CD760123ED
:1002E0007D329019C3EA02CDED02C1C1C9C5C5C5B1
:1002F000AF328F19219119C1E521D515E52190194A
:10030000CD7601112600CD910CD119D1C1E5D5D101
:10031000E123E5D52BCD76017D328D19D1E123E5A1
:10032000D52BCD76017D328E1921040039AF77238C
:1003300077C33E0321040039347E23200134CD8469
:1003400001112400A7ED52CAB903E123E52BE5CD45
:100350008401CD7601110600CD3B017DE6036F26B9
:1003600000CD7701D17D12E123E52BE5CD8401CDD0
:100370007601110400CD3B017DE6036F2600CD77A9
:1003800001D17D12E123E52BE5CD8401CD7601CBB2
:100390002CCB1DCB2CCB1D7DE6036F2600CD77012A
:1003A000D17D12E123E52BE5CD84017EE6036FD1FB
:1003B00012D1E123E5D5C33403219119E5218D192B
:1003C000CD7601D119545D3E0312219119E5218E9C
:1003D00019CD7601D11936046E2600CDE203C1C1D4
:1003E000C1C9C5C5219119E5CDC60C210000D1C1F7
:1003F000E5D5C3FA03D1E123E5D5C1E1E5C511098E
:1004000000A7ED52CA620421040039AF772377C3F5
:100410001C0421040039347E23200134CD840111D1
:100420001000A7ED52CAF50321B600E52E0639CD1E
:100430007F01292901280009E5CD8D0129292BE516
:1004400021BC15E521080039347E23200134666F74
:100450002BCD7601444D292909D119E5CDD80CC3FE
:100460001204C1C1C1C9219119E5210100E5C3747C
:1004700004E123E5E1E5119000A7ED52CA9104D112
:10048000E123E5D52B7EFE02C27104210100C1C12A
:10049000C9219019CD7601237D329019FE19C2A78A
:1004A00004210000C1C1C9CDED02210100C1C1C9B3
:1004B000218F19CD7601CDA001D2CA0421020039C5
:1004C0007EFE02C2CA04210000C921020039CD7695
:1004D00001C9C5C5C5C521050039E5218F19CD76EE
:1004E000012323237DD11221040039E5210E00CD03
:1004F0004901210E0039CD760129292929D1197DFB
:10050000CD7701D17D1221060039E521100039CDCA
:100510007F01CD7601EB21911919D1CDAC012106D6
:1005200000CD530121060039CD7601D119CD7601D8
:10053000E5CDB004C17CB5CA3F05C1C1C1C1C92167
[... 314 lines cut ...]
:1018E000008085401005614220022955555555005C
:1018F00000000067695555555564191811401000CE
:10190000016010608151800215484056014004403A
:1019100081555555550000000067691555555444CB
:10192000191811402040015010218141810A21588D
:101930004052014004408115555554000000006795
:1019400069555555554004000545880165460024F4
:101950001542091401462100514500456955555568
:101960005500000000676955555555400000114865
:1019700058080140608141480141814602402145AB
:10198000840001555555550000000000000000007E
:101990000000000000000000000000000000000047
:1019A0000000000000000000000000000000000037
:1019B0000000000000000000000000000000000027
:1019C0000000000000000000000000000000000017
:1019D0000000000000000000000000000000000007
:1019E00000000000000000000000000000000000F7
:1019F00000000000000000000000000000000000E7
:101A000000000000000000000000000000000000D6
:101A100000000000000000000000000000000000C6
:101A200000000000000000000000000000000000B6
:101A300000000000000000000000000000000000A6
:101A40000000000000000000000000000000000096
:0F1A500000000000000000000000000000000087
:00000001FF

fu-sen, 2025-03-12:
> DSTAR on The PC-G815 looks nice, here's the binary content in HEX:

Yeah, that feels good.

[image]

[image]

[image]

zx70, 2025-03-21:
Trying to get the same text resolution on the G815 and the G850 I did some experiment.
Considering that in UPPERCASE it is slightly more readable, do you think it is a valid option ?
Writing a scroll routine with a 4 pixels height would be is much simpler.

On the G850 it would push the text resolution to 12 rows, could it be useful as an extra option ?

[image]

zx70, 2025-03-22:
I'll try to use the same font without blanks between the rows, it should balance the resolution (6 rows) and the speed. 
The ultra small font can still stay as an option.

zx70, 2025-03-24:
I still haven't understood what's happening with the first pixel column, a simpe workaround is to shift right the text, to test the look and feel  :P

It is slightly more readable, but the speed on the real hardware could be concerning.   I suppose the emulator runs faster than the real machines, right?

[image]

zx70, 2025-03-25:
I havent solved the graphics bug of the first pixel (yey), but the workaround helps.
Could anybody test the new text driver on the PC-G815 ?
Is the speed acceptable ?

zx70, 2025-04-08:
[image]

zx70, 2025-04-08:
[image]

fu-sen, 2025-04-11:
We may still need to make some tweaks, but that should be enough to close this. Thank you for the great response!

closed 2025-04-11 by fu-sen