z88dk/z88dk #2666: Developer Documentation -- What would be required to add support for a new target board?

cogburnd02, 2025-02-09:
I'd like to add support for a new target board to z88dk. Is there something like a to-do list that describes what would be needed to add support for a new board, and if such a list doesn't exist yet then can we put one together?

feilipu, 2025-02-09:
Probably good add some pages in the wiki for this.

The way I'd start is to find a similar machine, particularly for cpu type, and grep all references to that machine which will allow you to copy existing work for the new machine.

Another kickstart is to search for the PR  where another machine was added to see all the files that were touched.

But neither of these methods are going to be perfect because they won't cover the intent of the author. So some actual instructions with intent (the why) would be good.

cogburnd02, 2025-02-09:
How would one define a similar machine? How is similarity determined? The machine I'd like to get added is the Cidco Mailstation 100.

feilipu, 2025-02-09:
You need to know what ancillary devices are used. Also, if you have a ROM dump or listing to understand which ports are used by the I/O devices like the parallel port and the LCD screen.

Can you get access to the ROM and do you have a device to reprogram it?

Is there any technical info or a user group for the mivo 100 device? 

If not it is very much a reverse engineering project, which is mainly unrelated to z88dk except that `z88dk-dis` can be used to disassemble the ROM, once you have it extracted.

feilipu, 2025-02-09:
There is a [page summarising some basic information here](https://jcs.org/2019/05/03/mailstation).

zx70, 2025-02-20:
well Dominic did a lot of work in documenting the kit,  here's acouple of links.
https://github.com/z88dk/z88dk/wiki/Classic--Retargetting

look around in the wiki, there is a lot more,  in example:
https://github.com/z88dk/z88dk/wiki/Classic--Homebrew

jagarsoft, 2025-03-01:
In case my experience is of any use to anyone, these are the steps I'm taking to create a new target for +Zux on [Zux](https://github.com/jagarsoft/Zux) for my [Z80Emulator](https://github.com/jagarsoft/Z80Emulator) I am developing...

For this I'm giving "baby-steps" applied in the TDD methodology.

1. Starting from scratch, run
zcc +Zux -v -no-cleanup -z80-verb test.asm
where test.asm is a empty file.
2. He complains that 
Can't open config file C:/.../z88dk/lib/config/Zux.cfg
 doesn't exist. I create it, empty. I rerun,
No CRT0 defined in configuration file <C:/.../z88dk/lib/config/Zux.cfg>
3. Add
CRT0 DESTDIR/lib/target/Zux/classic/Zux_crt0
4. PROCESSING C:/.../z88dk/lib/config\..\..\/lib/target/Zux/classic/Zux_crt0.asm.m4
The system cannot find the file specified.
5. I create that file, empty
6. Now zcc crates zcc_proj.lst with
C:/.../z88dk/lib/config\..\..\/lib/target/Zux/classic/Zux_crt0.asm.m4
test.asm

Well, this is my current step... I'm working on it ;-)

My next step, when I have time, will be:

Copy the following paths to target\zux\ changing as necessary. Add includes for each section. Start with a minimum:
C:\...\z88dk\libsrc\_DEVELOPMENT\target\z80\startup\z80_crt_0.asm.m4
to
C:\...\z88dk\lib\target\Zux\classic\Zux_crt0.asm.m4

Stay tuned, if you're interested ;-)

jagarsoft, 2025-04-15:
Here you are the result in my own brand-new target: [Zux](https://github.com/jagarsoft/z88dk/tree/master/lib/target/Zux)

More information about it in #2695

closed 2025-12-02 by suborb