z88dk/z88dk #2672: Starting point for supporting Rabbit 6000 (r6k)?

tomlogic, 2025-03-01:
I was very happy to find this project when I was searching for an Open Source Z80 disassembler I could use as a starting point for a Rabbit 2000 disassembler for some work I'm doing.  I didn't need much, and `z88dk-dis -mr2ka` gave me exactly what I needed!

I'm wondering if I can help out with adding support for the Rabbit 6000.  I'm the support engineer for Rabbit products at Digi International, and with the recent announcement that all Rabbit products were going EOL, I'm confident that I can convince Digi to Open Source some of the files the closed-source Dynamic C used for Rabbit assembly/disassembly.  IIRC there are a series of CSV files describing the opcodes.

They did have me Open Source the libraries and sample programs (device code) from both Dynamic C 9 (Rabbit 2000/3000) and 10 (Rabbit 4000/5000/6000) in 2015.

- [DCRabbit_9](http://github.com/digidotcom/DCRabbit_9/)
- [DCRabbit_10](http://github.com/digidotcom/DCRabbit_10/)

It's been a while, but I've also coded in Rabbit 4000 assembly and might be able to help out there as well.  I worked on some Dynamic C 10 compiler assembly optimizations while an employee from 2008-2014.  For example, when I did a search trying to find opcode definitions in this project, I saw a `ld hl, 0` used on Rabbit 4000 that you could replace with the new `clr hl` opcode (0xBF).

feilipu, 2025-03-03:
Hi Tom,

AFK currently, and will provide thoughts when back online.

I think we emailed about rcm2024 at some stage. The pieces (design, layout, etc) are still on my desktop.

P.

suborb, 2025-03-07:
I'm reviewing the spreadsheet from:  http://ftp1.digi.com/support/documentation/Rabbit%206000%20Instruction%20Spreadsheet.zip

And I'm collating the new instructions and turning it into something more usable for our purposes (we did a similar exercise for the KC160 here: https://github.com/z88dk/z88dk/issues/2402#issuecomment-1746350820). It looks like it's row 544 onwards with attention paid to 516 -> 524.

I'll then add to the disassembler and stub out ticks as appropriate.

The assembler side looks trickier with the new `alts` and `altsd` prefixes

suborb, 2025-03-08:
Is that spreadsheet actually correct? It looks like the 8 bit operations which were put into page 0x7f for the 4000 have been brought back into the main page and the new Rabbit instructions pushed back into the 0x7f page.

I can see that `ld r1,r2` where r1=r2 have been pushed into the 0xcb page which allows the `alts` and `altsd` prefixes to work.
However `rl hl` is main-page 0x42 in the spreadsheet which would be `ld b,d`

suborb, 2025-03-10:
Just a status update:

- The disassembler supports the `-mr6k` flag and should disassemble the new instructions
- `z88dk-ticks` now has `-mr6k`. We're down to about 100 unsupported instructions (mostly block ones)
- `z88dk-ticks` has partial support for `alts` - this will be an ongoing background task to retrofit it
- Generated code is now lightly optimised for the 6000, `mul(u) hl,de` and `alu hl,(sp+n)` are substituted in as appropriate.

suborb, 2025-05-06:
Sorry I didn't get back to you on this one.

That looks like a general purpose structure for both disassembly and assembly. Although we wouldn't use it directly it could be of use as a reference.

pauloscustodio, 2025-05-07:
> I'm reviewing the spreadsheet from: http://ftp1.digi.com/support/documentation/Rabbit%206000%20Instruction%20Spreadsheet.zip

The spreadsheet includes duplicate opcodes for the same instructions, i.e. some ld r,r are duplicated in the main page and in the 0x7F page. Some instruction involving the pw,px,py,pz registers are duplicated in the main page and in the 0x6D page. It includes conditional jumps on a a new flag (bit x) without a corresponding mnemonic in Rabbit6000_Delta4000Instructions.xlsx.

Is there a complete Rabbit Instruction Reference.pdf including the r6k mnemonics that I can refer to?

suborb, 2025-05-07:
> The spreadsheet includes duplicate opcodes for the same instructions, i.e. some ld r,r are duplicated in the main page and in the 0x7F page. Some instruction involving the pw,px,py,pz registers are [...]

See the edit here: https://github.com/z88dk/z88dk/issues/2672#issuecomment-2707788086 - I think it's a partial documentation of the undocumented operating modes. From the sdcc wiki:

> The Rabbit 4000 and Rabbit 5000 have 4 modes, set via 2 bits in the internal I/O register EDMR. The pages ed, fd, dd and cb pages are the same in all modes. The 6d page is the same in all modes that have it.
00 - "Default instruction set" / "Rabbit 3000 mode" - binary compatible with Rabbit 2000 and Rabbit 3000A.
11 - "Enhanced instruction set" - assembler compatible with Rabbit 2000 and Rabbit 3000A, but different opcodes. Has 6d and 7f pages. Many instructions removed from main page, most of them replaced by new instructions. The 7f page contains instructions that would be on the main page in mode 00, but have been replaced by new instructions.
01 - Undocumented. ld l, l instruction removed from main page, new 6d page added. Otherwise the same as 00.
10 - Undocumented. ld l, l and ld a,a instructions removed from main page, new 6d and 7f pages added. The 7f page contains the instructions that would be on the main page in mode 11, but not mode 00.

suborb, 2025-05-22:
The toolchain should now support R6K. There's not much in terms of board support unfortunately.

closed 2025-05-22 by suborb