z88dk/z88dk #2683: classic - add a second serial (tty) interface

feilipu, 2025-03-14:
It is possible in the newlib to add a second (or further) serial interfaces. Specifically a "tty" interface has been added for rc2014 SIO/2 and UART devices.

But, classic library doesn't (to my knowledge) support a second or further serial interfaces. This is relevant for RC2014 8085 builds with `uart85` subtype.

I've [added the basic `ttyin`, `ttyout`, `ttyerr` definitions to `stdio.h`](https://github.com/z88dk/z88dk/commit/cd17002278c16bd2dc2d955fbaec6d77e8b44c8a) to enable compilation to proceed, but it would be nice to have these ports actually connected properly.

Not sure of the best way to proceed with this. Clearly something has to be added to the startup definitions, but I'm sure there's more.

suborb, 2025-03-14:
To hook up to FILE you should be able to write an freopen trampoline and place it at the right point in the FILE array from the crt. 

I’m open to having a new general purpose serial interface - the current one was borrowed from cc65 years ago for compatibility reasons.

feilipu, 2025-03-14:
It would be good to have a general solution for multiple ports, parallel to or derived from Alvin's abstract newlib solution.

I think not only CP/M but other targets can use several ports. CP/M has `crt` `tty` and several special ports like `punch`. I think it would be sufficient to use the CP/M example as a best case.

suborb, 2025-03-14:
Can you point me to the bit you're referring to? The one I'm thinking of is the device table layout in the crt which doesn't match what I thought you were thinking of.

The freopen trampoline is fairly similar to that device handling, though I don't think there's anyway to drive down an IOCTL as yet.

suborb, 2025-03-14:
Okay, I think I understand now, working on this:

```
// Header file
extern FILE *stdpun;
```

```
crt:

    MACRO setup_static_fp  name, trampoline
        SECTION crt_bss
        PUBLIC  _name
_name:   defs    10

        SECTION code_crt_init
        ld      hl,+(__sgoioblk + 2 +  (n * 10))
        ld      (hl),33         ;_IOEXTRA|_IOUSE (stdio.h)
        inc     hl              ;ungetc
        inc     hl
        ld      (hl), +( trampoline % 256)
        inc     hl
        ld      (hl), +( trampoline / 256)
    ENDM

    ; After sections have been setup
    EXTERN _stdpun_trampoline
    setup_static_fp(punch,_stdpun_trampoline)
```

```
// stdpun_trampoline.c
#define STDIO_ASM
#include <stdio.h>
#include <arch/cpm/cpm.h>

#asm
        SECTION code_clib
        PUBLIC  _stdpun_trampoline

_stdpun_trampoline:
        cp      __STDIO_MSG_PUTC
        jr      z,handle_putc
        cp      __STDIO_MSG_WRITE
        jr      z,handle_write
        scf     ; error
        ret

; Entry: ix = fp
;        bc = character
; Exit: hl = result
handle_putc:
        ld      e,c
        ld      c,CPM_WPUN
        call    bdos
        and     a
        ret

; ix = fp
; de = buf,
; bc = length
handle_write:
        ex      de,hl
loop:
        push    bc
        push    hl
        ld      e,(hl)
        ld      c,CPM_WPUN
        call    bdos
        pop     hl
        inc     hl
        pop     bc
        dec     bc
        ld      a,b
        or      c
        jr      nz,loop
        and     a
        ret
#endasm

suborb, 2025-03-14:
I pushed a change that provides stdpun, stdrdr, stdlst for +cpm - hopefully it's easy to follow how it can be done.

One caveat is that this isn't supported on on 808x since I got lazy back when I added. I don't think it's too onerous to add, just a bit of register shuffling.

feilipu, 2025-03-14:
Thanks. Assuming you're still working on the solution second serial `tty`, so I'll hold off for time being.

For future me, a rough sketch of how I understand the newlib solution to work. Noting that newlib stdio doesn't handle file access, and that has been a limitation from the start.

1. Based on the provided subtype a [configuration](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/crt_config.inc) number and [startup code](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/target/rc2014/startup) are defined.
2. The subtype startup code is written in `.m4` so I don' t really understand it 100%, but here there are driver instantiation instructions. The first and second [serial port configuration](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/startup/rc2014_crt_8.asm.m4) is launched from here.
3. The subtype specific startup code is also useful to connect interrupts, build special memory maps, and do anything not found in the generic startup code.
4. In the [terminal directory](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/target/rc2014/driver/terminal) the specific `.m4` terminal configuration files are found, together with an assembly file linking the specific implementations of `getc`, `interrupt`, `reject`, and `flush`. The `rc_01_*_uarta.*` files are an implementation for the first `crt` port. Similarly the `rc_01_*_uartb.*` does the same for the second `tty` port.
5. I believe the newlib [cpm implementation](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/cpm/startup/cpm_crt_4.asm.m4) __does not__ implement the second `tty` serial port. It just does the `rdr`, `pun`, and `lst` devices. Really, it needs to have its second port implemented too, [as described here](https://www.seasip.info/Cpm/iobyte.html).

suborb, 2025-03-15:
It's probably worth having a go at describing the classic `FILE` setup:

```
typedef struct filestr FILE;

struct filestr {
        union f0xx {
                int         fd;
                uint8_t    *ptr;
        } desc;
        uint8_t   flags;
        uint8_t   ungetc;
        intptr_t  extra;
        uint8_t   flags2;  
        uint8_t   reserved;
        uint8_t   reserved1;
        uint8_t   reserved2;
};

#define _IOUSE          1
#define _IOREAD         2
#define _IOWRITE        4
#define _IOEOF          8
#define _IOSYSTEM      16
#define _IOEXTRA       32
#define _IOTEXT        64
#define _IOSTRING     128
```

#### Direct console interaction:

_Calls `fgetc_cons()` and `fputc_cons()`_

`filestr->flags |= _IOSYSTEM`

This is the shortcut code path, the console driver can be changed by `-pragma-define:fputc_cons=XXX` etc

#### File I/O

Calls the <fcntl.h> instructions

`filestr->flags = _IOUSE|_IOREAD` or `filestr->flags = _IOUSE|_IOWRITE`.`filestr->dest.fd` = handle from `open()`

File handling, usually fd holds a handle from the ROM, or an index into an array of FCBs that z88dk owns.

#### String I/O

Used for `scanf` and friends. 

`filestr->flags |= _IOSTRING`. `filestr->extra` = string length, `file->desc.ptr` = pointer to next character in string.

#### ZSock/`funopen()`

`filestr->flags |= _IOEXTRA` `filestr->extra` = address of trampoline

When `_IOEXTRA` is set, all file operations are pushed down via the trampoline. It's used by [`funopen()`](https://man.openbsd.org/funopen.3) and allows users to create a `FILE` with any implementation at all.

The parameters to the trampoline functions are "documented" here: [zsock_trampoline.c](https://github.com/z88dk/z88dk/blob/master/libsrc/stdio/zsock/zsock_trampoline.c)

It's this mechanism that I used to implement the `stdpun` etc devices yesterday. I just wrapped it up a z80asm macro to allow a simple drop-in into the crt.

With that feature, it's possible to add `FILE` that points to something I've not thought about either in the crt or in user code. So you should be able to add those `stdtty` points for rc2014 (with a bit of glue code).

----

Regarding the second tty in CP/M. There's not a BDOS call to access it directly - the following are available:

* BDOS 1 = Console input (`fgetc_cons`)
* BDOS 2 = Console output (`fputc_cons`)
* BDOS 3 = auxiliary input (`stdrdr`)
* BDOS 4 = auxiliary output (`stdpun`)
* BDOS 5 = list output (`stdlst`)

And where they end up is controlled by the `IOBYTE` so you possibly want to fiddle with the `IOBYTE` to set `PTP` and `PTR` to point to `TTY` and we could create a trampoline that combines the functionality of `stdrdr` and `stdpun`

----

There's another way to access serial ports in classic which is  [rs232.h](https://github.com/z88dk/z88dk/blob/master/include/rs232.h) - this has a restriction to a single implementation which is restrictive - for example there's multiple libraries for the +cpc that implement it.

It would be nice to support multiple devices of multiple implementations but I need some thinking time.

feilipu, 2025-03-15:
> It would be nice to support multiple devices of multiple implementations, but I need some thinking time.

It is an enhancement request. 🙏  

The current practical limitation is very minor. It only relates to the 8085 based RC2014 builds, which is something I've cooked up, with a Dual UART Module and Spencer is just launching the UART Module. So it is not a big deal. And, the limitation is only in the boot shell to determine which port of the two is being used.

Once inside CP/M there is no further issue, as both ports are supported, selected by the [`IOBYTE` setting](https://www.seasip.info/Cpm/iobyte.html) which is done by `BDOS 7` & `BDOS 8`, or via the `STAT` program.

In the interim, I've just removed the shell [port selection code](https://github.com/RC2014Z80/RC2014/blob/master/ROMs/CPM-IDE/8085-cf-uart/main.c#L626-L653) in the 8085 builds. Not "mission critical" as they say. 😄