z88dk/z88dk #2695: Why does z88dk translate \n to 0x0D instead of 0x0A?

jagarsoft, 2025-04-11:
Hi all!!

printf("Kernel\n");

```
defb      $4b
defb      $65
defb      $72
defb      $6e
defb      $65
defb      $6c
defb      $0d <- is not 0x0a
defb      $00
```

\r=CR=0x0D
\n=LF=0x0A
isn't it?

Thanks in advance!!

suborb, 2025-04-14:
A long time ago \n was mapped to 0x0d, however that's not been the case for many years.

Which target are you using? It's possible that we might be missing the flag from one of them.

suborb, 2025-04-15:
Yes, there's a flag used by all the other targets!  `-Cc-standard-escape-chars`

https://github.com/z88dk/z88dk/blob/72cc3007512b4674a10597cab5973d1c171630c5/lib/config/z80.cfg#L15

feilipu, 2025-04-15:
Pretty much all targets respect `STANDARDESCAPECHARS`,

like in this example (that I just fixed a few days ago).

https://github.com/z88dk/z88dk/blob/master/libsrc/target/rc2014/stdio/fputc_cons_uarta.asm#L25

closed 2025-04-15 by suborb

jagarsoft, 2025-04-15:
Excuse me, but with or without -Cc-standard-escape-chars in Zux.cfg the result is the same.

With this code:

```C
void __FASTCALL__ putck(unsigned char c)
{
  // https://www.z88dk.org/forum/viewtopic.php?p=22863&sid=95e6655ce9bac030f0007ae5cbcafc4e#p22863

  __asm
		ld	a, l
		rst	0x10
IF STANDARDESCAPECHARS
		cmp	0x0a
		jr	nz, putsk_L0
		ld	a, 0x0d
ELSE
		cmp	0x0d
		jr	nz, putsk_L0
		ld	a, 0x0a
ENDIF
		rst	0x10
.putsk_L0:
   __endasm;
}
```

gives this opcodes:

```ASM
_putck:
                    push      hl                            ;[4846] e5
                    ld        a,l                           ;[4847] 7d
                    rst       $10                           ;[4848] d7
                    cp        $0d                           ;[4849] fe 0d
                    jr        nz,putsk_L0                   ;[484b] 20 03
                    ld        a,$0a                         ;[484d] 3e 0a
                    rst       $10                           ;[484f] d7
putsk_L0:
                    pop       bc                            ;[4850] c1
                    ret                                     ;[4851] c9
```

It's as if STANDARDESCAPECHARS were never defined.

When I use -Cc-standard-escape-chars in Zux.cfg, must I do something else to define STANDARDESCAPECHARS?

Thanks in advance!!

feilipu, 2025-04-16:
If you're using an assembly source, like in your library, you need to define with `-D`, as the C compiler is not involved.

See the [Makefile](https://github.com/z88dk/z88dk/blob/master/libsrc/Makefile) for examples of usage.