z88dk/z88dk #2696: (ez80) invalid instruction in library

pauloscustodio, 2025-04-12:
While developing the z80asm strict option, I found the following errors in the library:

```
../_DEVELOPMENT/l/sccz80/4-ez80/lp_gchar.asm:14: error: syntax error
  ^---- ld.lis hl,2 ; point to pushed HLU by loading 16 bit 0x0002 to HL, HLU is written as 0x00.
      ^---- ld.lis hl,2
../_DEVELOPMENT/l/sccz80/4-ez80/lp_guchar.asm:14: error: syntax error
  ^---- ld.lis hl,2 ; point to pushed HLU by loading 16 bit 0x0002 to HL, HLU is written as 0x00.
      ^---- ld.lis hl,2
../_DEVELOPMENT/l/sccz80/4-ez80/lp_gint.asm:14: error: syntax error
  ^---- ld.lis hl,2
../_DEVELOPMENT/l/sccz80/4-ez80/lp_glong.asm:14: error: syntax error
  ^---- ld.lis hl,2
../_DEVELOPMENT/l/sccz80/4-ez80/lp_gptr.asm:14: error: syntax error
  ^---- ld.lis hl,2
../_DEVELOPMENT/l/sccz80/4-ez80/lp_pchar.asm:16: error: syntax error
  ^---- ld.lis hl,2
../_DEVELOPMENT/l/sccz80/4-ez80/lp_pint.asm:16: error: syntax error
  ^---- ld.lis hl,2
../_DEVELOPMENT/l/sccz80/4-ez80/lp_plong.asm:17: error: syntax error
  ^---- ld.lis hl,2
../_DEVELOPMENT/l/sccz80/4-ez80/lp_pptr.asm:17: error: syntax error
  ^---- ld.lis hl,2
```

I think ```ld.lil``` was intended, as ```ld.lis rr,Mmn``` does not exist. ```ld.lil``` loads a 3-byte value while in ADL=0 mode.

Can a ez80 expert please confirm this?

suborb, 2025-04-12:
This change has some info here: https://github.com/z88dk/z88dk/issues/2423

pauloscustodio, 2025-04-14:
Duplicate of https://github.com/z88dk/z88dk/issues/2423

closed 2025-04-14 by pauloscustodio