z88dk/z88dk #2699: How do I define a programme entry point?

Reymon-Real, 2025-04-16:
I need to set an entry point.

I read the documentation and didn't see anything related to how to define it from the command line or with a directive.

Maybe I didn't read it right, but I really need to know how to set it up.

The project uses object files instead of assembling everything in one file.

The problem is that always the point I have in mind as the program input is put at the end of the binary file instead of the beginning.

suborb, 2025-04-16:
I don’t understand what you’re trying to do - do you have a repo you can point me towards?

For C based programs the entry point to user code is main. The actual entry point is usually at the start of the program and is located in the crt0 file.

Reymon-Real, 2025-04-16:
> I don’t understand what you’re trying to do - do you have a repo you can point me towards?
> 
> For C based programs the entry point to user code is main. The actual entry point is usually at the start of the program and is located in the crt0 file.

An apology that the question is not very clear.
I will improve it.

First, the program is an assembler program and I need to define an entry point since my code in the main tag is assembled at the end of the binary file instead of at the beginning.

dbolli, 2025-04-16:
You can use the z80asm -r option to set the ORG address e.g. -r$((0x6100))

Regards,
Derek.

Reymon-Real, 2025-04-17:
@dbolli 
It seems that the problem is more about how z80asm assembles.

I did exactly what you said and the problem persists. It doesn't persist in the sense of where it starts the code, but it seems that it arranges the binary differently from what I want.

For example:

init.asm
```
ORG $0000
init:
XOR A
LD BC, $0000
JP BOOT
```

boot.asm:
```
BOOT:
JR BOOT
```

Disassembler:

```
JR 00                 ;; [00] 18 00
XOR A               ;; [03] af
LD BC, 0000    ;; [04] 01 00 00
JP 0000            ;; [07] c3 00 00
```

dbolli, 2025-04-17:
Unless you have INCLUDE "boot.asm" in your init.asm file, or you are declaring PUBLIC BOOT in boot.asm and EXTERNAL BOOT in init.asm and linking the files using the linker via ZCC, z80asm won't know the address of BOOT.

Regards,
Derek.

Reymon-Real, 2025-04-17:
@dbolli 

An apology for the code I put as an example, but that code was only to exemplify how the machine code is generated in my project.

I didn't want to make it open source yet (it's in a very early stage), but as I need that help.

So here is the link to the GitHub repository [My project](https://github.com/Pexnidi/Nexos).

The code folder is the énosis folder located at src/z80

dbolli, 2025-04-17:
If code order in the .bin file is the issue, I suggest you have a main.asm file that INCLUDEs init.asm and boot.asm in the order you want them and assemble that.

Regards,
Derek.

Reymon-Real, 2025-04-17:
@dbolli 

I feared as much. Well, thanks for the help. Had to get by the old-fashioned way with proper memory management.

closed 2025-04-17 by Reymon-Real

pauloscustodio, 2025-04-22:
> If code order in the .bin file is the issue, I suggest you have a main.asm file that INCLUDEs init.asm and boot.asm in the order you want them and assemble that.

z80asm links the object files in the order supplied in the command line.