z88dk/z88dk #2709: [pv1000] Cannot import custom font / -pragma-define:PV1000_CUSTOM_TILESET=1 only searches for a cpc binary

Fabrizio-Caruso, 2025-05-05:
I would like to use a custom .asm font file (e.g. one I may have generated with font2pv1000).
Including the .asm file (**not** the original binary) seems to have no effect (see attachment).

Using `-pragma-define:PV1000_CUSTOM_TILESET=1` produces an error about the inability to access the binary file. 

```
zcc.exe +pv1000      -O3 \
-Icross_lib -Icross_lib/sleep -Icross_lib/display -Icross_lib/display/graphics_mode -Icross_lib/display/alt_print -Icross_lib/include -Icross_lib/sound -Icross_lib/sound/cc65/atmos -Icross_lib/sound/cc65/c264 -Icross_lib/sound/cc65/pokey -Icross_lib/sound/cc65/sid -Icross_lib/sound/cc65/vic20 -Icross_lib/sound/cc65/gamate -Icross_lib/sound/cc65/creativision -Icross_lib/sound/cc65/atari_lynx -Icross_lib/sound/cc65/atari7800 -Icross_lib/sound/cc65/pet -Icross_lib/sound/cc65/supervision -Icross_lib/sound/z88dk/bit_bang -Icross_lib/sound/z88dk/gb -Icross_lib/sound/z88dk/psg -Icross_lib/sound/lcc1802/comx -Icross_lib/sound/cmoc/mo5 -Icross_lib/sound/generic -Icross_lib/sound/gcc4ti99 -Icross_lib/sound/vbcc/bbc -Icross_lib/text -Icross_lib/input -Icross_lib/rand -Icross_lib/display/redefine_characters -Icross_lib/display/tiles -Icross_lib/display/init_graphics/vbcc/bbc -Igames/bomber/generated_assets -Igames/bomber  -pragma-define:PV1000_CUSTOM_TILESET=1 \
-D__PV1000__ \
-DZ88DK_JOYSTICK -DYSize=23 \
-D__VT52 -D_XL_NO_COLOR -D_XL_NO_UDG \
-D_XL_SLOW_DOWN_FACTOR=10*40 \
-D__CONIO_GRAPHICS \
-D__NO_INIT_GRAPHICS \
-lndos \
-clib=default -pragma-redirect:fputc_cons=fputc_cons_generic \
cross_lib/display/display_macros.c cross_lib/input/input_macros.c cross_lib/sleep/sleep_macros.c games/bomber/main.c  \
games/bomber/generated_assets/z88dk_pv1000.asm \
-create-app -o../build/Xbomber_pv1000.bin
tileset.asm:1: error: syntax error
  ^---- Cannot open input file </cygdrive/c/z88dk/libsrc/_DEVELOPMENT/font/font_8x8/font_8x8_cpc_system.bin>
make: *** [makefiles.common/targets/z88dk/Makefile_pv1000:26: pv1000_gfx] Error 1
```

[z88dk_pv1000.zip](https://github.com/user-attachments/files/20047910/z88dk_pv1000.zip)

suborb, 2025-05-05:
I think this is a combination of an include order issue and me choosing a common name for the custom tiles.

From pv1000_crt0.asm:

```
    ; Character map - TODO, redefining it
IF PV1000_CUSTOM_TILESET
    INCLUDE     "tileset.asm"
ELSE
    INCLUDE     "target/pv1000/classic/font.asm"
ENDIF
```

And that's finding a `tileset.asm` for more sane targets somewhere else in the include path.

If you rename `z88dk_pv1000.asm`  to `tileset.asm` and put that directory in the include path does that  work?

Fabrizio-Caruso, 2025-05-05:
If I just rename my custom asm file to `tileset.asm` and keep in the declared location, it fails with the very same error.
`tileset.asm` is in `src/games/bomber/generated_assets`, which is in an included path.

It should be possible to just pass an .asm file as it is done with sprites. I don't understand why we need need a special pragma.

```
zcc.exe +pv1000      -O3 \
-Icross_lib -Icross_lib/sleep -Icross_lib/display -Icross_lib/display/graphics_mode -Icross_lib/display/alt_print -Icross_lib/include -Icross_lib/sound -Icross_lib/sound/cc65/atmos -Icross_lib/sound/cc65/c264 -Icross_lib/sound/cc65/pokey -Icross_lib/sound/cc65/sid -Icross_lib/sound/cc65/vic20 -Icross_lib/sound/cc65/gamate -Icross_lib/sound/cc65/creativision -Icross_lib/sound/cc65/atari_lynx -Icross_lib/sound/cc65/atari7800 -Icross_lib/sound/cc65/pet -Icross_lib/sound/cc65/supervision -Icross_lib/sound/z88dk/bit_bang -Icross_lib/sound/z88dk/gb -Icross_lib/sound/z88dk/psg -Icross_lib/sound/lcc1802/comx -Icross_lib/sound/cmoc/mo5 -Icross_lib/sound/generic -Icross_lib/sound/gcc4ti99 -Icross_lib/sound/vbcc/bbc -Icross_lib/text -Icross_lib/input -Icross_lib/rand -Icross_lib/display/redefine_characters -Icross_lib/display/tiles -Icross_lib/display/init_graphics/vbcc/bbc -Igames/bomber/generated_assets -Igames/bomber  -pragma-define:PV1000_CUSTOM_TILESET=1 \
-D__PV1000__ \
-DZ88DK_JOYSTICK -DYSize=23 \
-D__VT52 -D_XL_NO_COLOR -D_XL_NO_UDG \
-D_XL_SLOW_DOWN_FACTOR=10*40 \
-D__CONIO_GRAPHICS \
-D__NO_INIT_GRAPHICS \
-lndos \
-clib=default -pragma-redirect:fputc_cons=fputc_cons_generic \
cross_lib/display/display_macros.c cross_lib/input/input_macros.c cross_lib/sleep/sleep_macros.c games/bomber/main.c  \
games/bomber/generated_assets/tileset.asm \
-create-app -o../build/Xbomber_pv1000_gfx.bin
tileset.asm:1: error: syntax error
  ^---- Cannot open input file </cygdrive/c/z88dk/libsrc/_DEVELOPMENT/font/font_8x8/font_8x8_cpc_system.bin>
make: *** [makefiles.common/targets/z88dk/Makefile_pv1000:26: pv1000_gfx] Error 1
```

suborb, 2025-05-05:
The tiles have to go at a fixed place in the ROM - that’s why it’s done this way. 

Let me figure out a way to change the filename of the file included.

Fabrizio-Caruso, 2025-05-05:
@suborb sure, but could this be done by something like a directive in the .asm that tells the linker where the assembly code has to go in memory?

Fabrizio-Caruso, 2025-05-06:
@suborb I also have a question about the number of characters I can redefine.
The doc says 224 (256-32) but the font2pv1000 only generates up to character 127.
Can I use refine all 224 characters by manually adding more characters to the asm file?

suborb, 2025-05-06:
I took the easiest path to unblock you and changed the filename that it automatically includes.

I've been musing over adding something to assemble binary blocks into a single ROM file which would address this case.

Yes, you should be able to redefine all the character codes - be careful of memory though, it'll take about 8k of ROM space to do that and the max supported ROM size seems to be 16k!

Fabrizio-Caruso, 2025-05-06:
I don't understand what I have to do. Can I link my own file with my any name?

Fabrizio-Caruso, 2025-05-06:
Changing the file name did not work for me yesterday. So I don't understand why having a different default name should work. I will give a try. Can I keep the file where I want? Having it on an included path did not work yesterday. No need to put it in the command line?

suborb, 2025-05-06:
The problem that I saw yesterday was that you already have a file called `tileset.asm` in your project. I've changed the default name to `pv1000_tileset.asm` which should be unique enough for you not to have multiple files named that.

I'll be honest, I can't remember what gets added the include path at link time, to be safe put it in whatever directory you like add `-Ca-I[path]` option to the link line.

Fabrizio-Caruso, 2025-05-06:
I will give it a try after the latest snapshot has been made available. 
Just out of curiosity, couldn't this be done by specifying the memory segment in the .asm file? 
Can't the linker have named segment areas (something like `.segment fonts` at the beginning of the .asm file?

suborb, 2025-05-06:
Actually yes. I’m being a bit of an idiot - it could be made a section.

I’ll do that later on this week.

suborb, 2025-05-07:
Try this method with the next nightly: https://github.com/z88dk/z88dk/wiki/Platform---Casio-PV1000#managing-using-sections

It's all section based (with a pragma to disable the default tileset).

Fabrizio-Caruso, 2025-05-14:
It seems to work but including an .asm file makes the binary exceed the 16k limit even if I just put include a binary with as little characters as the default ones. So I must be doing something wrong. Or less probably, something is wrong in Z88DK.

Z88DK manual (https://github.com/z88dk/z88dk/wiki/Platform---Casio-PV1000)  says:
"By default, a z88dk ROM builds with the following character set (starting at code 0xa and ending at 0x7f):"

If I include the .asm generated as the example of the manual, I lose a lot of memory. 
The example does `font2pv1000 -f 6 -b 0 -c 32 {z88dk}/libsrc/_DEVELOPMENT/font/font_8x8/font_8x8_cpc_system.bin > tileset.asm`
and generates characters from 0x20 to 0x7F.

The manual says z88dk `"by default" is including characters from 0xa to 0x7f. How can I stay within the same limit memorywise?

suborb, 2025-05-22:
The by default refers to the regular 96 glyph font that we use on all other targets. On the pv1000 each character ends up being 32 bytes so ROM space can vanish quite rapidly.

There's no automatic enforcement of a limit - the hardware supports up to 256 characters.

closed 2025-05-28 by Fabrizio-Caruso