z88dk/z88dk #2715: Endless loop with dir_move_next() on big CP/M 3 directories.

Ho-Ro, 2025-05-09:
When traversing big CP/M directories the repeated call of the function `dir_move_next()` does not provide all directory entries and stops finally but results in an endless loop (maybe related to #1251). 
The error is seen on CP/M 3 running on my Z80-MBC2 HW as well on my yaze-ag simulation.

See the detailed error description in the test file below:
```c
#include <stdio.h>
#include <stdlib.h> /* wcmatch, exit */
#include <cpm.h>    /* bdos */

/*
 * Minimal CP/M test program to show the "dir_move_next issue":
 * When parsing (too?) big directories it displays:
 * 1.) directory entry #1 (result of dir_move_first())
 * 2.) followed by the next 128 directory entries #2..#129
 * 3.) followed by 128 repetitions of entry #2
 * 4.) then it loops to number 2.) above
 * Stop the loop with ^C
 *
 */

void chk_ctrl_c( void ) {
    if ( bdos( 6, 0xff ) == 3  ) { /* Ctrl C was typed */
        fprintf( stderr, "\n^C\n" );
        exit( -1 );
    }
}

/* ---------------------------------------------- */
/* -------------------- MAIN -------------------- */
/* ---------------------------------------------- */
int main( int argc, char *argv[] ) {
    int i = 0, c, x;
    char *filename;

    x = dir_move_first();
    while ( x == 0 ) {
        chk_ctrl_c(); /* kill loop with ^C */
        filename = dir_get_entry_name();
        printf( "   %-12s", filename );
        if ( ++i == 8 ) {
            printf( "\n" );
            i = 0;
        }
        x = dir_move_next();
    }
    return 0;
}
```
Directory structure:
```
Total Space     =      3540k  Total Bytes =  1569456  Total Records =    24552
Total 1k Blocks =      3161   Files Found =      193  Dir entries     317/ 384
```
Output of DIR
```
C>dir
C: HTC      ZIP : DDIR     COM : ASSERT   H   : C309-17  COM : C309     COM : CGEN     COM : CONIO    H   : CPM      H   
C: CPP      COM : CREF     COM : CRTCPM   OBJ : CTYPE    H   : DEBUG    COM : DRTCPM   OBJ : EXEC     H   : FLOAT    H   
C: HITECH   H   : LIBC     LIB : LIBF     LIB : LIBOVR   LIB : LIBR     COM : LIMITS   H   : LINQ     COM : MATH     H   
C: NRTCPM   OBJ : OBJTOHEX COM : OPTIM    COM : OPTIONS      : OVERLAY  H   : P1       COM : RRTCPM   OBJ : SETJMP   H   
C: SIGNAL   H   : STAT     H   : STDARG   H   : STDDEF   H   : STDINT   H   : STDIO    H   : STDLIB   H   : STRING   H   
C: SYMTOAS  COM : SYS      H   : TIME     H   : UNIXIO   H   : ZAS      COM : ARGTEST  C   : CC       COM : ARGTEST  BAK 
C: ARGTEST  COM : OUT      LST : ARGS     LST : HTC-DOC  TAR : UNTAR    BAK : UNTAR    COM : DEBUGMAN TXT : TESTTIME C   
C: UNTAR    C   : FILETIME C   : TESTTIME COM : OBJCODE  TXT : FILETIME BAK : CFT      BAK : FILETIME COM : CFT      C   
C: CREF     TMP : CFT      COM : TESTTIM0 C   : TT       C   : CFT      OBJ : TAR      C   : ANSITAR  C   : CFT      AS  
C: ANSITAR  BAK : CAT      TAR : ANSITAR  COM : MKTAR    C   : TEST80   TAR : MKTAR    COM : TEST80   BAK : CAT          
C: MKTAR    BAK : TXT80    TXT : TEST80   TXT : TXT80    TAR : TAR      BAK : TST      TAR : DDIR0000 COM : GUNZIP   C   
C: LS-L0000 GZ  : BIG      GZ  : ZCAT-ASM GZ  : LS-LST   GZ  : ZCAT-AS0 GZ  : LS-L     GZ  : TINYTAR  C   : Z        GZ  
C: TT000001 COM : Z        Z   : WSPRINT  OVR : NIX          : TT000002 COM : TT000003 COM : DMN      COM : SYMFILE  TXT 
C: TEST     ZIP : TEST280  LOG : TEST280  SUB : TESTAES  C   : TESTARGS C   : TESTBDOS C   : TESTBIOS C   : TESTFILE C   
[... 33 lines cut ...]
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM       DDIR.COM    
   DDIR.COM       DDIR.COM       ASSERT.H       C309-17.COM    C309.COM       CGEN.COM       CONIO.H        CPM.H       
   CPP.COM        CREF.COM       CRTCPM.OBJ     CTYPE.H        DEBUG.COM      DRTCPM.OBJ     EXEC.H         FLOAT.H     
   HITECH.H       LIBC.LIB       LIBF.LIB       LIBOVR.LIB     LIBR.COM       LIMITS.H       LINQ.COM       MATH.H      
   NRTCPM.OBJ     OBJTOHEX.COM   OPTIM.COM      OPTIONS        OVERLAY.H      P1.COM         RRTCPM.OBJ     SETJMP.H    
   SIGNAL.H       STAT.H         STDARG.H    
^C

C>
```
Martin

suborb, 2025-05-10:
This is probably the cause of it: https://github.com/z88dk/z88dk/blob/master/libsrc/target/cpm/fcntl/dir_move_first.c#L16

suborb, 2025-05-11:
I've been staring at the code for quite a while and I think there's a limit of 128 files for these functions. I've just pushed a change which should support 32k entries.

I suspect that it's going to be pretty slow with large directories since it scans from the start of the directory for the next entry - basically a pre-requisite of the bdos API that "find next" can only be called after "find first" - I'm guessing the sector cache will be blasted away if you do any other disc operations.

Ho-Ro, 2025-05-11:
> I think there's a limit of 128 files for these functions

Yes, also because of storing the number as `char` (aka. `int8_t`) in `dirbuf[133]` it is limited to 128 instead of 256 :) - Now the root cause is found (nevertheless I saw two more lines to fix, see my comments in 3660010) - and even if it is slow, it will work.
But while browsing `libsrc` I found an even simpler solution for my intention to do a wildcard expansion with a quite high number of possible matches, which I need for my implementation of [tar for CP/M](https://github.com/Ho-Ro/CPM-tools/blob/main/tools/tinytar.c). Instead of iterating through all files in a directory and comparing each name with argv[i] (as shamelessly stolen from `examples/osca/wc.c` at first), I modified the functions `dir_move_first()` and `dir_move_next()` from `libsrc/target/cpm/fcntl` to search for my wildcard argument(s) only - instead of `"*.*"`, which speeds up the search quite a bit.

**BTW** - shouldn't you change `char dirbuf[134]` to `char dirbuf[128]` now, since it's just exactly one CP/M record copied by `parsefcb()`? Browsing through the entire project, I have not found any use of the higher addresses 128..134.

Thank you for your quick reaction
Martin

closed 2025-05-11 by Ho-Ro