z88dk/z88dk #2717: SDCC build fails on Fedora 42

jorgegv, 2025-05-11:
There is an open upstream bug (https://sourceforge.net/p/sdcc/bugs/3846/) due to a mismatched `elf()` function declaration in SDCC sources. It compiles fine with older compilers, but on Fedora 42, GCC 15 defaults to compiling in C23 mode, and SDCC build fails.

As a consequence, Z88DK's sdcc build fails as well.

The proposed upstream workaround is to compile SDCC with `-std=c17` option, but since this would also require patching the SDCC Makefile, I have opted for a more direct fix with a patch that directly fixes the mismatched declaration in  SDCC sources when preparing for building.

Pull request is here:

jorgegv, 2025-05-12:
Wel, so far I'm building it with C23 on Fedora 42, and this one is the only error, so it seems it will not be very much affected...?

feilipu, 2025-05-12:
closed by #2718.

closed 2025-05-12 by feilipu