z88dk/z88dk #2725: (classic) CP/M file io performance

suborb, 2025-05-17:
Pulling out an issue from the thread here: https://github.com/z88dk/z88dk-ext/pull/6#discussion_r2089616650

In summary CP/M file performance has always been an issue.

* Before 2023, when writing a byte we read the sector, modified the byte, rewrote the sector
* In 2023, a sector cache was added to reduce the number of read/write cycles. On real hardware this showed a significant performance improvement
* In the linked thread, byte by byte writing is being used, I was surprised that a userland sector cache made such a significant improvement and I've been digging to understand why.

To investigate this issue I've added support for CPM random file access to ticks in 0b7d9f00a9e6749c3871a0a272e52de71a9b82d1. This allows us to use the debugging tools with ticks (hotspots etc) to analyse what's going on.

There's 3 operating modes in test tool:

* Unbuffered (-U) which relies on buffering within the library, calling via stdio to fcntl on a byte-by-byte basis
* Buffered, userland buffering, calling via stdio to fcntl in 128 byte blocks 
* Direct, calling to fcntl on a byte-by-byte basis

The tool is basically reading a 1 megabyte file and then writing it to a different file.

[buf.c](https://github.com/user-attachments/files/20264538/buf.c.txt)

CP/M file io is fairly complex, so I've baselined against `+test` which is just a thin wrapper around the host, so it's ideal for it. Some measurements:

| Mode | Tick count (10e6)  |
|-|-|
| Unbuffered | 1441 |
| Buffered | 510 |

And for CP/M we have:

| Mode | Tick count (10e6)  |
|-|-|
| Unbuffered (before changes) | 14545 |
| Unbuffered (after changes)  | 10595  |
| Direct (after changes) | 9238 |
| Buffered (after changes) | 672 |

Summary of changes (not yet committed):

| Mode | Tick count (10e6)  |
|-|-|
| 1 Unbuffered (before changes) | 14545 |
| 2 Return &fcb rather than index | 12986 |
| 3 Cache file position -> record number | 12708 |
| 4 Optimise size==1 case to avoid memcpy | 12628 |
| 5 Reorder local variables to make fcb easier to access in write | 12569 |
| 6 Move read fcb as above | 11435 |
| 7 Optimisation rule + inline cache contents check| 10595 |

2. Changing the CP/M fcntl library to return an `(int)&fcb` rather than index into an array of FCBs. This saves a 16 bit multiplication on every entry to `read` and `write`
3. Removes a 32 bit division (demoted to >>7) to map between file position and sector number.
4. Just doing a byte assignment
5. Relocating fcb so it's at sp+0
6. Ditto for `read()`
7.  Inlining the check as to what sector is currently cached

suborb, 2025-05-17:
Given that I'm using ticks, the random/sequential discussion isn't part of this - wall clock time doing host io isn't counted in the numbers.

This test really is deliberately the worst case, but is realistic if a file is opened in text mode (rather than binary).

It looks like the single call overhead of the CP/M implementation of `read()` or `write()` is about 4500T with my changes (down from 6000T).

Ho-Ro, 2025-05-17:
> This test really is deliberately the worst case, but is realistic if a file is opened in text mode (rather than binary).

In my case (expanding a gzip'ed file) it seems also realistic to write it as binary.

suborb, 2025-05-17:
> > This test really is deliberately the worst case, but is realistic if a file is opened in text mode (rather than binary).
> 
> In my case (expanding a gzip'ed file) it seems also realistic to write it as binary.

This was referring to how text files have to be handled - reading byte by byte looking for a ^Z.

Anyway. I was puzzled about how HTC could be faster when I know that its `write()` implementation is very much like our pre 2023 version.

HTC does buffering at the stdio level rather than the fcntl as we do. This means that if you were to use the -D option I added to the test program I suspect it wouldn't be quite so good.

I think I've squeezed pretty much all I can out of `write()` optimisations, so I'm going to add an optional (for most targets) buffer at the stdio level.

suborb, 2025-05-17:
> I think I've squeezed pretty much all I can out of `write()` optimisations, so I'm going to add an optional (for most targets) buffer at the stdio level.

That wasn't true. By having a special case for a read/write of 1, I've got it down to 6149 - it's at a cost of a bit more code memory, whereas a stdio buffer would be heap/bss and code memory. So it's a trade off.

Ho-Ro, 2025-05-19:
Sorry to be a nuisance, but there are still two problems:

1. `ftell()` no longer works as expected, it returns `-1` at the end of the file.

```c
    fpos_t bytecount;
    if ( ( infile = fopen( argv[ 1 ], "rb" ) ) == NULL ) {
        fprintf( stderr, "cannot open infile %s\n", argv[ 1 ] );
        return -1;
    }
    fseek( infile, 0, SEEK_END ); /* go to EOF */
    bytecount = ftell( infile );  /* get position */
    fseek( infile, 0, SEEK_SET ); /* rewind */
    fprintf( stderr, "%s: %lu bytes\n", argv[ 1 ], bytecount );
```

2. `fgetc()` returns random character when executed in the emulators [cpm](https://github.com/jhallen/cpm) and [tnylpo](https://github.com/SvenMb/gbrein_tnylpo).
It's ok in real CP/M 3 as well as `yaze-ag`. The HTC version works as expected.

[copy.c.txt](https://github.com/user-attachments/files/20282190/copy.c.txt)

```sh
$ cpm copy assert.h assert-c.out  
ASSERT.H: -1 bytes
Z88DK: file copy using standard fgetc/fputc

$ cpm copy assert.h assert-c.out
ASSERT.H: -1 bytes
Z88DK: file copy using standard fgetc/fputc

$ cpm copy-htc assert.h assert-c.htc
ASSERT.H: 640 bytes
HTC: file copy using standard fgetc/fputc

$ tnylpo copy assert.h assert-t.out
ASSERT.H: -1 bytes
Z88DK: file copy using standard fgetc/fputc

$ tnylpo copy-htc assert.h assert-t.htc
ASSERT.H: 640 bytes
HTC: file copy using standard fgetc/fputc

$ ls -l assert*.*
-rw-r--r-- 1 horo horo  640 19. Mai 12:24 assert-c.htc
-rw-r--r-- 1 horo horo 4096 19. Mai 12:23 assert-c.out
-rw-r--r-- 1 horo horo  521 19. Mai 11:59 assert.h
-rw-r--r-- 1 horo horo  640 19. Mai 12:24 assert-t.htc
-rw-r--r-- 1 horo horo 4096 19. Mai 12:24 assert-t.out

$ hd -v assert-c.out | head -4
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|

$ hd -v assert-t.out | head -4
00000000  76 76 76 76 76 76 76 76  76 76 76 76 76 76 76 76  |vvvvvvvvvvvvvvvv|
00000010  76 76 76 76 76 76 76 76  76 76 76 76 76 76 76 76  |vvvvvvvvvvvvvvvv|
00000020  76 76 76 76 76 76 76 76  76 76 76 76 76 76 76 76  |vvvvvvvvvvvvvvvv|
00000030  76 76 76 76 76 76 76 76  76 76 76 76 76 76 76 76  |vvvvvvvvvvvvvvvv|

$ hd -v assert-c.htc | head -4
00000000  23 69 66 6e 64 65 66 20  5f 48 54 43 5f 41 53 53  |#ifndef _HTC_ASS|
00000010  45 52 54 5f 48 0d 0a 23  64 65 66 69 6e 65 20 5f  |ERT_H..#define _|
00000020  48 54 43 5f 41 53 53 45  52 54 5f 48 0d 0a 0d 0a  |HTC_ASSERT_H....|
00000030  2f 2a 0d 0a 20 2a 09 41  73 73 65 72 74 69 6f 6e  |/*.. *.Assertion|

$ hd -v assert-t.htc | head -4
00000000  23 69 66 6e 64 65 66 20  5f 48 54 43 5f 41 53 53  |#ifndef _HTC_ASS|
00000010  45 52 54 5f 48 0d 0a 23  64 65 66 69 6e 65 20 5f  |ERT_H..#define _|
00000020  48 54 43 5f 41 53 53 45  52 54 5f 48 0d 0a 0d 0a  |HTC_ASSERT_H....|
00000030  2f 2a 0d 0a 20 2a 09 41  73 73 65 72 74 69 6f 6e  |/*.. *.Assertion|
```

 z88dk version: 2.3-1382-g134b9b7368 

```
commit 134b9b7368a4678a709d8edf3583c3ab4a79fa31
Author: suborb <dom@suborbital.org.uk>
Date:   Sun May 18 19:26:12 2025 +0100

    (snap) icu needs a bump as well
```

suborb, 2025-05-19:
Thanks for spotting these. 

First issue was caused by remove the index into the fcb array. Second was a badly written optimisation rule resulting in the high word of an int not being reset.

feilipu, 2025-05-20:
For comparison, I've pushed a very simple C file copy program that emulates the CP/M PIP program in its logic. It is essentially the same as the assembly version already uploaded a few years ago.

The resulting real world performance is similar to the previous state. That's to be expected as I understand most of the improvements here relate to single byte read and write.

The reference machine is an RC2014 Z80 using an IDE 16bit Disk on Module (DOM) and the DRI PIP program.
The test is to copy a 1MB random file within the same CP/M disk.

Here's some numbers:

CP/M PIP (standard DRI written in PLM. [PIP.PLM](https://github.com/user-attachments/files/20327310/PIP.PLM.TXT) attached)
`pip 1mbp.txt=1mb.txt` 46.9 Seconds

Assembly [`cp.asm` ](https://github.com/z88dk/z88dk-ext/blob/master/os-related/CPM/cp.asm)
`cp 1mba.txt 1mb.txt` 43.7 seconds

In C [`cp.c`](https://github.com/z88dk/z88dk-ext/blob/master/os-related/CPM/cp.c)
2023 Testing
SDCC 4.2 60.2 seconds

2025 repeated test with 2023 binary
SDCC 4.2  57.3 seconds

2025 new test 
SDCC 4.5  59.7 seconds / 63.1 seconds
SCCZ80    60.0 seconds / 60.0 seconds

suborb, 2025-05-20:
I think that makes sense - the more times that read/write are called the higher the relative improvement. 

I’m guess your tool reads in 50k or so chunks so only calls 40 times. A 128 byte buffer should show significant changes.

feilipu, 2025-05-20:
Yes. Just to check no regression for large files. And, I see none.

`PIP` opens 32kB buffers when it can and uses a fast copy mechanism when it recognises that it doesn't need to do byte by byte activity.

I can do that too, with an assembly program calling BDOS directly. But using the library calls now, I don't think that is possible.

I think there is a way to pass between sequential and random calls in the BDOS, but not sure exactly how at the minute. Further research needed.

feilipu, 2025-05-21:
> I think there is a way to pass between sequential and random calls in the BDOS, but not sure exactly how at the minute.

Yes, it is using the [BDOS function 36](https://www.seasip.info/Cpm/bdos.html#36) which updates the random access pointer to allow RRAN and WRAN calls following the use of the sequential calls.

The BDOS random access functions are nothing more than setting the correct position for the random access pointer, and then calling the standard sequential function. Unfortunately, there's a lot of code to set the `POSITION` every time the random access call is made.

```
;
;   Read a random record.
;
READRAN:
    LD      C,0FFH          ;set 'read' status.
    CALL    POSITION        ;position the file to proper record.
    CALL    Z,RDSEQ1        ;and read it as usual (if no errors).
    RET
;
;   Write to a random record.
;
WRITERAN:
    LD      C,0             ;set 'writing' flag.
    CALL    POSITION        ;position the file to proper record.
    CALL    Z,WTSEQ1        ;and write it as usual (if no errors).
    RET
```

The BDOS function 36 simply computes the random record number, and then stuffs these into the FCB. So if it were possible to do this in the `read.c` and `write.c` functions, either using the BDOS 36 or in C, then perhaps the much faster sequential calls could be used?

```
;
;   Function to return the random record position of a given
;   file which has been read in sequential mode up to now.
;
SETRAN:
    LD      HL,(PARAMS)     ;point to fcb.
    LD      DE,32           ;and to last used record.
    CALL    COMPRAND        ;compute random position.
    LD      HL,33           ;now stuff these values into fcb.
    ADD     HL,DE
    LD      (HL+),C         ;move 'r0'.
    LD      (HL+),B         ;and 'r1'.
    LD      (HL),A          ;and lastly 'r2'.
    RET
```

```
;
;   Compute the random record number. Enter with (HL) pointing
;   to a fcb an (DE) contains a relative location of a record
;   number. On exit, (C) contains the 'r0' byte, (B) the 'r1'
;   byte, and (A) the 'r2' byte.
;
;   On return, the zero flag is set if the record is within
;   bounds. Otherwise, an overflow occured.
;
COMPRAND:
    EX      DE,HL           ;save fcb pointer in (DE).
    ADD     HL,DE           ;compute relative position of record #.
    LD      C,(HL)          ;get record number into (BC).
    LD      B,0
    LD      HL,12           ;now get extent.
    ADD     HL,DE
    LD      A,(HL)          ;compute (BC)=(record #)+(extent)*128.
    RRCA                    ;move lower bit into bit 7.
    AND     80H             ;and ignore all other bits.
    ADD     A,C             ;add to our record number.
    LD      C,A
    LD      A,0             ;take care of any carry.
    ADC     A,B
    LD      B,A
    LD      A,(HL)          ;now get the upper bits of extent into
    RRCA                    ;bit positions 0-3.
    AND     0FH             ;and ignore all others.
    ADD     A,B             ;add this in to 'r1' byte.
    LD      B,A
    LD      HL,14           ;get the 's2' byte (extra extent).
    ADD     HL,DE
    LD      A,(HL)
    ADD     A,A             ;and shift it left 4 bits (bits 4-7).
    ADD     A,A
    ADD     A,A
    ADD     A,A
    PUSH    AF              ;save carry flag (bit 0 of flag byte).
    ADD     A,B             ;now add extra extent into 'r1'.
    LD      B,A
    PUSH    AF              ;and save carry (overflow byte 'r2').
    POP     HL              ;bit 0 of (L) is the overflow indicator.
    LD      A,L
    POP     HL              ;and same for first carry flag.
    OR      L               ;either one of these set?
    AND     01H             ;only check the carry flags.
    RET
```

feilipu, 2025-05-21:
>  either using the BDOS 36 or in C, then perhaps the much faster sequential calls could be used for most cases?

Bare with me while I think through the logic here.

- BDOS 15 F_OPEN and BDOS 22 F_MAKE both actively clear the `S2` byte except for `S2` bit 7 which will be set to indicate that the file is unmodified. This bit is cleared by any file write, and is checked by BDOS 16 F_CLOSE allowing an early exit for unmodified files.
- BDOS 36 F_RANDREC only updates the `R0`, `R1`, and `R2` bytes for the FCB. It does nothing else.
- The internal BDOS `MODE` variable is set by all read and write functions, with 0 for random access and 1 for sequential.
- On exit, all read and write functions call `SETNREC` which will check the `MODE` and only if it is 1 (sequential) the record number is incremented. So sequential read and write are auto-increment on record number. Random (mode 0) and special random (mode 2) do not.
- During sequential read and write `R0`, `R1`, and `R2` bytes are not touched. So they remain as zero. Just the record number is updated by increment.

So it looks like sequential read or write can be used if  the size of a read or write is a full 128 byte record. `R0`, `R1`, and `R2` bytes remain zero so long as reading is done sequentially. The next record is automatically prepared by increment.

Any function that deviates from a sequential process needs to have BDOS 36 F_RANDREC function called in advance to prepare for random read and write. This will make sure that `R0`, `R1`, and `R2` bytes are non-zero and are properly prepared. Also the record count will need to be decremented, if the current record is needed, as it will have been pre-incremented.

tbc...

feilipu, 2025-05-27:
Yes. In theory there is no benefit to using random at all. Sequential operations just has a side benefit of precalculating the next record for performance.

For sequential write the next record is opened in advance, I believe. Random write duplicates this space check and record opening.

Nevertheless, I'm still thinking about how to do this.

I discarded the idea of mixing the two access types. Can only lead to confusion.

A neat assembly routine is needed to convert library file pointer to EX S2 and CR bytes, before calling a sequential bdos routine. The conversion is regular and fairly simple.

What I am missing is any small side effects, which need to be managed.

For reference the [BDOS](https://github.com/feilipu/yaz180/blob/master/cpm22/cpm22.asm) for yaz180 is well commented any it is easy to see what is done behind the curtain.

I've nothing concrete coded to date.

feilipu, 2025-05-28:
I've created a branch to play with this. #2738

Currently not working.

It relies on the `fc->rwptr` being accurate and updated. And, I'm not sure it is currently.
Of course, there can and will be other issues.

suborb, 2025-05-30:
The single byte read is probably the bit that gets the "ungot" byte in `fread()` which falls through into reading a single byte from the file if there's an "ungot" byte.

feilipu, 2025-05-30:
> falls through into reading a single byte from the file if there isn't an "ungot" byte.

Ouch. That makes my futzing around for a few percent with BDOS calls kind of irrelevant.

The double BDOS read call, and writing into a cache twice is much more time consuming.

Is there some way to avoid that situation?

suborb, 2025-05-30:
Yeah we can bypass it. Needs a flag to indicate whether it’s set since it looks a wrong atm. I‘ll fiddle this evening. 

This probably explains why the read/write benchmark was better than fread/fwrite - we end up with a double call down to read()

feilipu, 2025-05-31:
Switched to the `rework_ungetc` branch to check using the current random access BDOS. I'm getting a slight performance regression of 68 seconds versus 60 seconds previously.

Never the less, the logic is performing correctly now, with only direct reads being activated.
I could put the performance change down to having the directory and extents on the drive I'm using being dirtied. It may be a free extent searching issue, but I don't know exactly.

2023 Testing
SDCC 4.2 60.2 seconds

2025 repeated test with 2023 binary
SDCC 4.2 57.3 seconds

2025 new test
SDCC 4.5 59.7 seconds / 63.1 seconds
SCCZ80 60.0 seconds / 60.0 seconds

2025 `rework_ungetc` fixes
SCCZ80 68.0 seconds / 68.0 seconds

feilipu, 2025-06-11:
I've put (probably too much) time into this investigation. In summary it is difficult to write a clean C path for using the sequential BDOS API more efficiently than the `_putoffset.c` and BDOS `POSITION` function associated with the random BDOS API.

So I've optimised the `_putoffset.c` function to [use assembly for sccz80](https://github.com/z88dk/z88dk/commit/a2b4fdb58414536d5d311d71d1d92913babdc873) which reduces its impact from 220 cycles per BDOS call down to 110 cycles per call. It won't make too much difference overall, but it is a small help.

commit 2c7404e7 (2025-05-18): Merge pull request #2727 from z88dk/Issue_2725

Fix #2725: CP/M file performance
closed 2025-06-11 by feilipu