z88dk/z88dk #2726: [pv1000] No rom images work on real hardware. Not even a helloworld tiny program.

Fabrizio-Caruso, 2025-05-17:
Roel Shuring (TinnedTrace.wordpress.com) has reported to me about his own tests on real hardware+multicard.
None of my 9 games and none of several simple test programs I gave him, worked. Nor the 16k standard format, nor the 32k subtype.

All my games and test cases work fine under Takeda's emulator (both 16k and 32k formats).

Problem reported by him: "None of the games work, they all give blue and yellow screens which is the same screen I get with no game put into the system. I did get a few stray corrupted pictures with random letters and what might be sprites or hex data, but restarting got me back to the blue and yellow screen."

Roel Shuring reports that all roms by Inufuto (http://inufuto.web.fc2.com/8bit/)  and KanedaFr (https://kanedafr.itch.io/unk) work fine for him.
Neither Inufuto nor KanedaFr use Z88DK nor appmake.

Maybe we have a problem with the rom format.

I have attached mazy2 by Inufuto and unk by Takeda (not build with Z88DK) and my failing helloworld example.
Inufuto provides source code of his Mazy2 game here: http://inufuto.web.fc2.com/8bit/mazy2/

[OK.zip](https://github.com/user-attachments/files/20265494/OK.zip)

[Xhelloworld_pv1000.zip](https://github.com/user-attachments/files/20265497/Xhelloworld_pv1000.zip)

My helloworld example is built with:
```
make pv1000 USE_TOOLS=1  ZSDCC_MAKEFILE_THREADS_OPTS='-j 24'  ZSDCC_MAKEFILE_COMPILATION_OPTS='-SO3 --max-allocs-per-node200000 ' Z88DK_MAKEFILE_COMPILATION_OPTS=' -O3' GCC4TI99_MAKEFILE_COMPILATION_OPTS='-O2 -fno-peephole2 -fno-function-cse' VBCC_MAKEFILE_COMPILATION_OPTS='-O1' NATIVE_MAKEFILE_COMPILATION_OPTS='-Wpedantic -Wall' CMOC_MAKEFILE_COMPILATION_OPTS='-O2' CC65_MAKEFILE_COMPILATION_OPTS='-Or -Cl' LCC1802_MAKEFILE_COMPILATION_OPTS='"-Wf-volatile" -O "-Wp-D nofloats" "-Wa-D LCCNOLONG" "-Wf-mulcall"' _NATIVE_CC=gcc TOOL_CC=gcc GNU_MAKE=make -f examples/helloworld/Makefile.helloworld
zcc.exe +pv1000      -O3 \
-Icross_lib -Icross_lib/sleep -Icross_lib/display -Icross_lib/display/graphics_mode -Icross_lib/display/alt_print -Icross_lib/include -Icross_lib/sound -Icross_lib/sound/cc65/atmos -Icross_lib/sound/cc65/c264 -Icross_lib/sound/cc65/pokey -Icross_lib/sound/cc65/sid -Icross_lib/sound/cc65/vic20 -Icross_lib/sound/cc65/gamate -Icross_lib/sound/cc65/creativision -Icross_lib/sound/cc65/atari_lynx -Icross_lib/sound/cc65/atari7800 -Icross_lib/sound/cc65/pet -Icross_lib/sound/cc65/supervision -Icross_lib/sound/z88dk/bit_bang -Icross_lib/sound/z88dk/gb -Icross_lib/sound/z88dk/psg -Icross_lib/sound/lcc1802/comx -Icross_lib/sound/cmoc/mo5 -Icross_lib/sound/generic -Icross_lib/sound/gcc4ti99 -Icross_lib/sound/vbcc/bbc -Icross_lib/text -Icross_lib/input -Icross_lib/rand -Icross_lib/display/redefine_characters -Icross_lib/display/tiles -Icross_lib/display/init_graphics/vbcc/bbc -Iexamples/helloworld/generated_assets -Iexamples/helloworld  \
-D__PV1000__ \
-DZ88DK_JOYSTICK -DYSize=23 \
-D__VT52 -D_XL_NO_COLOR -D_XL_NO_UDG \
-D_XL_SLOW_DOWN_FACTOR=10*0 \
-D__CONIO_GRAPHICS \
-D__NO_INIT_GRAPHICS \
-lndos \
-clib=default -pragma-redirect:fputc_cons=fputc_cons_generic \
cross_lib/display/display_macros.c cross_lib/input/input_macros.c cross_lib/sleep/sleep_macros.c examples/helloworld/main.c  \
-create-app -o../build/Xhelloworld_pv1000.bin
# rm -rf ../build/Xhelloworld_pv1000.bin
rm -rf ../build/Xhelloworld_pv1000_DATA.bin
rm -rf ../build/Xhelloworld_pv1000_BSS.bin
```

You can build it with Cross-Lib with `xl helloworld pv1000`.

suborb, 2025-05-17:
I can see some differences in the boot up of the ROM for Mazy2 - possibly a timing delay and initialising the hardware.

Unk just shows some hardware initialisation at $069c. I'll add that into the crt - hopefully you can get it retested?

Fabrizio-Caruso, 2025-05-17:
@suborb Sure Roel is willing to help us and test the games and examples.

suborb, 2025-05-17:
Great, I've pushed a change - one of the settings is the base address of the VRAM so that could explain the screen not changing!

TinnedTrace, 2025-05-18:
I'm Roel Schuring, made a github account for this. If there are any rom images I could/should test, feel free to send them to me.

Fabrizio-Caruso, 2025-05-18:
@TinnedTrace Thanks for your support!

I have attached some games and simple test programs that I have recompiled with the latest Z88DK version that may have fixed the issue. 

[build.zip](https://github.com/user-attachments/files/20273377/build.zip)

TinnedTrace, 2025-05-18:
[image]

They all give similar pictures to this, which looks like random ROM or RAM data.
This is the hello world program. At least now something is happening on screen.

suborb, 2025-05-18:
Thank you - definitely progress. The one attached here has the same register settings as Mazy. Do you mind trying it out?

[pv1000_roms.zip](https://github.com/user-attachments/files/20273851/pv1000_roms.zip)

TinnedTrace, 2025-05-18:
Sure.
Ansitest gave me the same results as above.

suborb, 2025-05-18:
That's awkward. Makes me think that it's crashing in someway and the RAM just has random data in it so isn't cleared.

Let's try disabling interrupts...

[sorry, this is going to be a painful iterative experience - both Mame and pv1000.exe aren't showing up any issues]

[ansitest2.zip](https://github.com/user-attachments/files/20274101/ansitest2.zip)

TinnedTrace, 2025-05-18:
[image]

Something different showed.

And no worries it being iterative, kinda expected it and have some experience with it in hardware projects.

suborb, 2025-05-18:
That's what it's meant to show!

I think this means that an interrupt is happening before we've initialised RAM, given that the start is of the ROM is:

```
jp label
...
label: 
im 1
di
```

It's pretty quick. The interrupt handler picks up vectors from RAM and so it just goes off into space.

Which has given me something to work with. The attached disables interrupts at the start of the ROM but doesn't re-enable them. If this fails then I'll have to think of another way to handle this early interrupt.

[ansitest3.zip](https://github.com/user-attachments/files/20274301/ansitest3.zip)

KashtPR, 2025-05-18:
Glad to see there is progress being made! :)

Just to add some additional info, Inufuto has 18 homebrew games that all work, so whatever he did seems to be the most consistent way to make it work, for now. UNK is the only other example from someone else to get homebrew to run in an actual console, but I haven't been able to get a hold of him yet.

The most exact info about how the hardware works can be found here: https://obscure.nesdev.org/wiki/Casio_PV-1000 (page is down atm)

lidnariq did most of the emulation work to get it as exact as it is in MAME, so as far as I am concerned, you can use his site (NesDev) as reference. 

Looking forward to getting Fabrizio's games working, so thanks everyone for all the work, seems like you guys are very close!

suborb, 2025-05-18:
Great, let's try something different. I've attached dstar and this should read input from the joystick.

[dstar.zip](https://github.com/user-attachments/files/20274437/dstar.zip)

TinnedTrace, 2025-05-18:
That gives me the random RAM data again, like the image from 3 hours ago.

suborb, 2025-05-18:
Ok, that's told me it's the interrupt handler causing the problem and that it seems to fire before program start, so I need to rewrite/remove that feature.

I've got some info from the nesdev link that will help.

suborb, 2025-05-18:
Let's give that another go...

[dstar2.zip](https://github.com/user-attachments/files/20275049/dstar2.zip)

TinnedTrace, 2025-05-18:
[image]

Now it's half the screen of RAM data.

suborb, 2025-05-19:
I'm not liking this interrupt! Let me rewrite it and the joypad scanning so they work together.

KashtPR, 2025-05-19:
The UNK developer replied to my question about his compiler here: https://kanedafr.itch.io/unk

Not sure if it helps any but sharing just in case.

KanedaFr, 2025-05-19:
Hi there ;)
Is there anything I can do to help ?

KanedaFr, 2025-05-19:
What I did is

At boot
- im
- init I/O
- clear screen
- read joypad first values
- ei

at 0x38 int
- inc Vcount
- if vcount %16 ==0 -> first vint so vblank + psg_update
- else read joy step

suborb, 2025-05-19:
Apart from the joypad that's what I thought I was doing - I need to do the joypad since clearing the interrupt will muck around with reading outside of the interrupt

But, having just disassembled the last ROM I've pointed the interrupt to the routine which relies on RAM data.

Thanks for the nudge!

TinnedTrace, 2025-05-19:
Both work! Only difference between them is that Dstar4 has some symbol going crazy in the top right corner.

suborb, 2025-05-19:
The crazy symbol was my debug for the interrupt triggering! 

Thank you so much for testing. It’s amazing how much the emulators can differ from the actual hardware.

commit 60b48659 (2025-05-19): Fix #2726: Fix crt0/interrupts so that it works on real hardware

Many thanks to @tinnedtrace for testing.

This change also ensures that joysticks are numbered the same way
as the other machines, i.e. 1 and 2, rather than 0, 1