z88dk/z88dk #2728: (ticks r6k) "alts ld de',bc" incorrect execution

pauloscustodio, 2025-05-18:
The ALTS prefix is not being parsed by ticks.

```
pc=0000, [pc]=76,    bc=0000,  de=0000,  hl=0000,  af=0040, ix=0000, iy=0000
sp=0000, [sp]=0176, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
                    altd ld        bc,$1234                 ;[0000] 76 01 34 12
pc=0001, [pc]=01,    bc=0000,  de=0000,  hl=0000,  af=0040, ix=0000, iy=0000
sp=0000, [sp]=0176, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
clockh=AD29, clockhl=60B0
                    ld        bc,$1234                      ;[0001] 01 34 12
pc=0004, [pc]=40,    bc=0000,  de=0000,  hl=0000,  af=0040, ix=0000, iy=0000
sp=0000, [sp]=0176, bc'=1234, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
clockh=AD29, clockhl=60B0
                    ld        b,b                           ;[0004] 40
pc=0005, [pc]=ED,    bc=0000,  de=0000,  hl=0000,  af=0040, ix=0000, iy=0000
sp=0000, [sp]=0176, bc'=1234, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
clockh=AD29, clockhl=60B0
                    ld        de',bc                        ;[0005] ed 59
pc=0007, [pc]=C3,    bc=0000,  de=0000,  hl=0000,  af=0040, ix=0000, iy=0000
sp=0000, [sp]=0176, bc'=1234, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
clockh=AD29, clockhl=60B0

```

suborb, 2025-05-18:
You need to change the EMDR mode first of all:

```
    ;; Enable R4K instruction mode on the R4K
    ld      a,$c0
    ioi ld  ($0420),a       ;EDMR register (p299 in R4000UM.pdf)
```

Otherwise it runs in r2k mode.

pauloscustodio, 2025-05-18:
You're right with EDMR!

But it still does not work:
```
$ z88dk-ticks.exe -mr6k -trace test.bin
pc=0000, [pc]=3E,    bc=0000,  de=0000,  hl=0000,  af=0040, ix=0000, iy=0000
sp=0000, [sp]=C03E, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
                    ld        a,$c0                         ;[0000] 3e c0
pc=0002, [pc]=D3,    bc=0000,  de=0000,  hl=0000,  af=C040, ix=0000, iy=0000
sp=0000, [sp]=C03E, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
clockh=A3AC, clockhl=0150
                    ioi ld        ($0420),a                 ;[0002] d3 32 20 04
pc=0003, [pc]=32,    bc=0000,  de=0000,  hl=0000,  af=C040, ix=0000, iy=0000
sp=0000, [sp]=0000, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
clockh=A3AC, clockhl=0150
                    ld        ($0420),a                     ;[0003] 32 20 04
pc=0006, [pc]=76,    bc=0000,  de=0000,  hl=0000,  af=C040, ix=0000, iy=0000
sp=0000, [sp]=C03E, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
clockh=A3AC, clockhl=0150
                    altd ld        bc,$1234                 ;[0006] 76 01 34 12
pc=0007, [pc]=01,    bc=0000,  de=0000,  hl=0000,  af=C040, ix=0000, iy=0000
sp=0000, [sp]=C03E, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
clockh=A3AC, clockhl=0150
                    ld        bc,$1234                      ;[0007] 01 34 12
pc=000A, [pc]=40,    bc=0000,  de=0000,  hl=0000,  af=C040, ix=0000, iy=0000
sp=0000, [sp]=C03E, bc'=1234, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
clockh=A3AC, clockhl=0150
                    alts ld        de',bc                   ;[000a] 40 ed 59
pc=000B, [pc]=ED,    bc=0000,  de=0000,  hl=0000,  af=C040, ix=0000, iy=0000
sp=0000, [sp]=C03E, bc'=1234, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
clockh=A3AC, clockhl=0150
                    ld        de',bc                        ;[000b] ed 59
pc=000D, [pc]=C3,    bc=0000,  de=0000,  hl=0000,  af=C040, ix=0000, iy=0000
sp=0000, [sp]=C03E, bc'=1234, de'=1212, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0 (8085: K=0)
clockh=A3AC, clockhl=0150

```

pauloscustodio, 2025-05-18:
ticks.c line 4364 has a typo.

commit a9866955 (2025-05-19): Fix #2728,#2729: R6K LD rr,rr2 with alts
closed 2025-05-19 by suborb