z88dk/z88dk #2731: error message points at wrong line

Ho-Ro, 2025-05-21:
This error description was already used for #815 - but here the error message comes from `z88dk-sccz80` (or `z88dk-zsdcc`, which stops after the 1st error) triggered by a simple syntax error (missing semicolon) in source code line number **12 and 21**.
(`int main( void ) {` is the 1st line.)

```c
int main( void ) {

#if TEST
/*
 *
 * here comes a long comment
 *
 */
#endif

    int test = TEST;
    int a = 12 /*  <--- this is an error in line 12 */
    char b = 10;

/*
 *
 * here comes another long comment - but if'ed out
 *
 */

    int x = 21 /* <--- this is an eror in line 21 */
    char y = 22;
    return 23;
}
```
Depending on the value of TEST it reports line numbers **8, 14** or **9, 15**.

```sh
$ zcc -v +cpm -DTEST=0 err.c 

PROCESSING err.c
z88dk-ucpp -iquote"." -D__CPM -DCPM -D__CPM__ -DTEST=0 -D__Z80__ -D__Z80 -D__Z88DK  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/usr/local/share/z88dk//include"    "err.c" "/tmp/tmpXXjOX9ta.i2"
z88dk-zpragma -sccz80 -zcc-opt="/tmp/tmpzccXXPtikt3/zcc_opt.def" < "/tmp/tmpXXjOX9ta.i2" > "/tmp/tmpXXjOX9ta.i"
z88dk-sccz80  -ext=opt -mz80 -zcc-opt="/tmp/tmpzccXXPtikt3/zcc_opt.def" -standard-escape-chars   "/tmp/tmpXXjOX9ta.i" -o "/tmp/tmpXXjOX9ta.opt"
err.c:8:5: error: Missing token, expecting ; got c
err.c:14:5: error: Missing token, expecting ; got c

$ zcc -v +cpm -DTEST=1 err.c 

PROCESSING err.c
z88dk-ucpp -iquote"." -D__CPM -DCPM -D__CPM__ -DTEST=1 -D__Z80__ -D__Z80 -D__Z88DK  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/usr/local/share/z88dk//include"    "err.c" "/tmp/tmpXX7oZPGj.i2"
z88dk-zpragma -sccz80 -zcc-opt="/tmp/tmpzccXXYHF4zn/zcc_opt.def" < "/tmp/tmpXX7oZPGj.i2" > "/tmp/tmpXX7oZPGj.i"
z88dk-sccz80  -ext=opt -mz80 -zcc-opt="/tmp/tmpzccXXYHF4zn/zcc_opt.def" -standard-escape-chars   "/tmp/tmpXX7oZPGj.i" -o "/tmp/tmpXX7oZPGj.opt"
err.c:9:5: error: Missing token, expecting ; got c
err.c:15:5: error: Missing token, expecting ; got c
```
The result is independent of option `-compiler=`; the file `err.c.i` is identical for both compilers (sdcc stops after the 1st error). `-DTEST=1` adds one more empty line.
The line numbers of `err.c.i` do match.
The file starts with an empty line, `#line 1 "err.c"` is in the 2nd line.)
These line numbers show the correct error location, e.g. the 1st error comes from the 7th code line (`int a = 12`) and is detected in line 8 when the token 'c' from "char" is seen instead of ';' (or an operator, e.g. '+',..)
`-DTEST=0`
```c

#line 1 "err.c"
int main( void ) {

 

    int test =  0 ;
    int a = 12  
    char b = 10;

 

    int x = 21  
    char y = 22;
    return 23;
}

```
`-DTEST=1`
```c

#line 1 "err.c"
int main( void ) {

 

    int test =  1 ;
    int a = 12  
    char b = 10;

 

    int x = 21  
    char y = 22;
    return 23;
}

```

This works also for large files, e.g. `gunzip.c:429:3: fatal error: Expected ';'`
1. create `gunzip.c.i`
2. find the last line telling `#line nn "gunzip.c" with `nn <= errline`,e.g `#line 62 "gunzip.c"`

suborb, 2025-05-22:
This seems to be fairly recent, it looks like it was providing sensible line numbers at 49287be1a39df2da4d34bd28de7c1219c85271c6

suborb, 2025-05-22:
I got the wrong issue number on it - change b3c1960764f061e8be4fde79b17b6cc24bf0b2ac restores line numbers so that they're more sensible (the change revert was collapsing blank lines, but failed to output a `#line` directive).

suborb, 2025-05-27:
Closing, the regression in line numbers has been resolved.

closed 2025-05-27 by suborb