z88dk/z88dk #2732: Compiling C++ / ez80-clang error

Manello, 2025-05-22:
Hey there!
I've found an issue with the z88dk while trying to compile for cpp (Using the latest build of today on both windows and Debian):

`zcc +z80 firmware/main.cpp -compiler=ez80clang -create-app -m -v`

Triggers the following output:

`PROCESSING firmware/main.cpp
C:\Users\Manue\Desktop\Tools\/z88dk/bin/z88dk-ucpp -iquote"." -D__EMBEDDED_Z80 -D__SDCC_IY -D__Z88DK_NEWLIB -isystem"C:\Users\Manue\Desktop\Tools\/z88dk/bin/../\include\_DEVELOPMENT\sdcc" -D__Z88DK  -D__SDCC  "firmware/main.cpp" "C:\Users\Manue\AppData\Local\Temp\zcc00000B480CD52.i2"
C:\Users\Manue\Desktop\Tools\/z88dk/bin/z88dk-zpragma -zcc-opt="C:\Users\Manue\AppData\Local\Temp\zcc00000B480CD53/zcc_opt.def" < "C:\Users\Manue\AppData\Local\Temp\zcc00000B480CD52.i2" > "C:\Users\Manue\AppData\Local\Temp\zcc00000B480CD52.i"
Only -compiler=ez80clang supports c++`

No matter if the -compiler=ez80clang flag is present or not, this error shows up. This makes it impossible to compile .cpp files.
This errors seems to originate from the zcc.c file.

Manello, 2025-05-22:
After compiling my files by just using ez80-clang without z88dk with success I assume that the arguments of ez80-clang got renamed / the layout of them changed.

suborb, 2025-05-22:
I’m not a windows person but I’m wondering if it’s a mingw/cygwin clash.

Manello, 2025-05-23:
So after adding 

`-clib=classic`

the application complains under pure Windows that it can't find the file/command ez80-clang.
As ez80-clang is not available for Windows I am switching completely to WSL with a Debian image + manually installed ez80-clang 
(I guess this is the way to do this. Unless z88dk has an ez80-clang integrated which somehow isn't found?).

On WSL the status is the same as in my [previous comment](https://github.com/z88dk/z88dk/issues/2732#issuecomment-2901537068).

suborb, 2025-05-23:
I found a Linux binary in a random directory. I've put it up for download here: http://nightly.z88dk.org/misc/clang/linux/

This run is on Ubuntu 22.04 in a container:

```
zcc +z80 test.cpp -compiler=ez80clang -clib=classic  -v

PROCESSING test.cpp
ez80-clang -iquote"." -D__EMBEDDED_Z80 -D__Z80 -D__Z88DK  -E -D__CLANG -isystem"/home/dom/z88dk/lib/config//../..//include"    "test.cpp" > "/tmp/tmpXX1b7Ddm.i2"
z88dk-zpragma -zcc-opt="/tmp/tmpzccXXE58ntJ/zcc_opt.def" < "/tmp/tmpXX1b7Ddm.i2" > "/tmp/tmpXX1b7Ddm.i"
ez80-clang  -cc1 -x c++ -triple z80 -S -O3  < "/tmp/tmpXX1b7Ddm.i" -o "/tmp/tmpXX1b7Ddm.opt"
z88dk-copt -mz80  /home/dom/z88dk/lib/config//../..//lib/clang_rules.1 < "/tmp/tmpXX1b7Ddm.opt" > "/tmp/tmpXX1b7Ddm.asm"
cat "/tmp/tmpXX1b7Ddm.asm" >> "/tmp/tmpXX1b7Ddm.tmp"
cat "/tmp/tmpXX1b7Ddm.tmp" >> "/tmp/tmpXX1b7Ddm.asm"
z88dk-z80asm -I"/tmp/tmpzccXXE58ntJ"  -s  -mz80 -I"/home/dom/z88dk/lib/config//../..//lib" "/tmp/tmpXX1b7Ddm.asm"

PROCESSING /home/dom/z88dk/lib/config//../..//lib/target/z80/classic/z80_crt0.asm
cat "/home/dom/z88dk/lib/config//../..//lib/target/z80/classic/z80_crt0.asm" >> "/tmp/tmpXXLVz4zv.asm"
cat "/tmp/tmpXXLVz4zv.asm" >> "/tmp/tmpXXLVz4zv.tmp"
cat "/tmp/tmpXXLVz4zv.tmp" >> "/tmp/tmpXXLVz4zv.asm"
WILL ACT AS CRT

GENERATING OUTPUT

PROCESSING CRT
z88dk-z80asm -I"/tmp/tmpzccXXE58ntJ"  -s  -mz80 -I"/home/dom/z88dk/lib/config//../..//lib" -I"/home/dom/z88dk/lib/config//../..//lib/target/z80/classic"  "/tmp/tmpXXLVz4zv.asm"

z88dk-z80asm -mz80 -b -d  -o"a.bin" -L.  -L"/home/dom/z88dk/lib/config//../..//lib/clibs/z80"   -I"/tmp/tmpzccXXE58ntJ" -L"/home/dom/z88dk/lib/config//../..//lib/clibs" -I"/home/dom/z88dk/lib/config//../..//lib" -lz80_clib -lndos -l"z80_crt0"     "/tmp/tmpXXLVz4zv.o" "/tmp/tmpXX1b7Ddm.o"
```

suborb, 2025-05-23:
I've got this running natively on windows now. The ingredients:

* git-bash
* Latest nightly z88dk (or anything recent since clang was added)
* v11.2 of CE-programming toolkit: https://github.com/CE-Programming/toolchain/releases/tag/v11.2

Note, later version of the CE programming toolkit seem to fail on Windows.

Setting up is fairly simple:

1. unzip z88dk
2. unzip CEprogramming
3. Copy CEDev/bin/ez80-clang into z88dk/bin
4. Setup path to point to z88dk/bin
5. export ZCCCFG={z88dk}/lib/config/

Create a cpp file and compile:

[image]

[image]

[image]