z88dk/z88dk #2747: sccz80 - 8085 issues ldir instruction

feilipu, 2025-06-11:
With this C code, sccz80 issues an `ldir` instruction

`zcc +test -clib=8085 --list -S _putoffset.c`

```C
struct x {
    unsigned char buf[3];
};

void _putoffset(unsigned char *where,long offset) {
    *(struct x *)where = (struct x)&offset;
}
```

```asm
; Function _putoffset flags 0x00000280 __smallc 
; void _putoffset(unsigned char * where, long offset)
; parameter 'long offset' at sp+2 size(4)
; parameter 'unsigned char * where' at sp+6 size(2)
	C_LINE	18,"_putoffset.c::_putoffset::0::0"
._putoffset
	GLOBAL	__putoffset
.__putoffset
	GLOBAL	____putoffset
.____putoffset
	ld	de,sp+6
	ld	hl,(de)	;l_gint
	push	hl
	ld	hl,4	;const
	add	hl,sp
	pop	de
	ld	bc,3
	ldir
	ret
```

feilipu, 2025-06-11:
Ok. I was reading the `-S` and listings stage. So the substitution happens after that?

pauloscustodio, 2025-06-11:
In the list file you see ldir translated as CD 00 00. The actual address is filled at link time.

closed 2025-06-11 by feilipu