z88dk/z88dk #2758: Problem with integer type bigger than int16_t

Vagant1, 2025-06-29:
Hi,

I am trying to use ZCC to compile code for Amstrad PCW (cpm) and I hit the problem with int. Everything is working fine while using simple int or int16_t but neither int32_t or log works - I am getting 0 for any operations.

Is there a library I should be adding as part of compilation?

Cheers
Michal

aralbrec, 2025-06-29:
Can you show a compile line and a code snippet to go along with it?

Unsigned is going to be better code on the z80 but int32_t should be no problem.  Log requires float math so you need to link against the float library.

suborb, 2025-06-29:
To print a 32 bit value you have to use the format `%ld`

feilipu, 2025-06-29:
I think to print 32-bit variables you need to use the `%ld` format converter.

And, probably you need to use the `#pragma` to enable these in your source.

The `-lm` for classic cp/m target will link the slowest math library, `genmath`. And you should include the `math.h` header too.

Vagant1, 2025-06-29:
Thanks again! I did test with float math and -lm was definitely slower than --math32 - is this the fastest option?
Trying to get fixed point working but the standard library and Accum has bit too low accuracy.

Thanks for your help! I would much rather ask such questions on the forum but I can't see any way to register.

feilipu, 2025-06-29:
`genmath` linked with `-lm` is definitely the slowest, but works on more targets.

`math48` is most accurate, but that accuracy is only useful with `sccz80` supporting 6 byte floats.

Either `math32` or Microsoft `mbf32` are the fastest general float solutions.

[Classic--Maths-Libraries](https://github.com/z88dk/z88dk/wiki/Classic--Maths-Libraries)

But for best speed play with fixed point math or `math16`.

closed 2025-06-29 by Vagant1

suborb, 2025-06-29:
> And you are right - it does work - big thanks! Is this specific to the ZCC? I tested the same code with GCC and it did work with simple %d.

That's because sizeof(int)==sizeof(long) for gcc, whereas for z88dk (and other compilers for 8 bit processors), sizeof(int)=2, sizeof(long)=4.

> Thanks for your help! I would much rather ask such questions on the forum but I can't see any way to register.

Unfortunately, sometimes I get overloaded with cleaning up after spammers from St Petersburg and just turn off registration for a break.

Don't get me started on the trouble that abusive AI scraping bots cause.

suborb, 2025-06-30:
> > Unfortunately, sometimes I get overloaded with cleaning up after spammers from St Petersburg and just turn off registration for a break.
> 
> Is there any way around this to get registered?

They're open again. Hopefully we've dropped off the target list for a while.

Vagant1, 2025-06-30:
Superb! Registered and just waiting for activation.