z88dk/z88dk #2767: How to reserve a memory block

zx70, 2025-07-27:
as for the tiki-100, the Apple II needs a memory block such is usually in the middle of a CP/M program, what's the right way to reserve it?

suborb, 2025-07-27:
If memory is banked out eg for vram access then HIMEM is the way to go - the code is copied up to just below the bios.

zx70, 2025-07-28:
Good point, I'll try to figure out how to page it in/out.

zx70, 2025-08-04:
I think the classic Z80 SoftCard did not provide extra memory, it simply shuffles the memory locations to preserve the 6502 "page zero", the I/O section and the display text memory (which is used also for the lo-rez graphics.

The HI-RES graphics use a different memory portion which, due to the SoftCard circuitry, the Z80 sees at $1000 (the 6502 sees it at $2000).
The text page, which originally was at $400 is shifted at $F400 it should be directly accessible by the Z80, but I'm not sure it is a good idea, the 40 columns display had an odd arrangement and we'd loose the compatibility with different display types (mostly the 80 columns option)

Different, later and less popular Z80 cards had their own RAM and probably allowed the graphics to stay untouched together with a full CP/M user area.

The Apple II allowed pointing to a second "page" both in text/lorez and hires mode, so we could alternatively just put the graphics page at $3000 (seen as $4000 on the 6502 side), map the BSS and the user area at $5000, and limit the actual code size at 12K when running in hirez.