z88dk/z88dk #2769: Spectrum examples throw errors

thearduinoguy, 2025-07-29:
A lot of the Spectrum examples don't work and throw errors.

dbolli, 2025-07-29:
I compiled the spectrum examples from a nightly build from 3 days ago and they all compile and run

```
cd /path/to/z88dk/examples/spectrum/
make clean
make
```

Regards,
Derek.

thearduinoguy, 2025-07-29:
I just tried running the cmandel.c example and I get:

`PS D:\z88dk\examples\spectrum> zcc +zx cmandel.c -create-app -bn frog.bin
cmandel.c::main::0::9:39: error: undefined symbol: dsub
  ^---- dsub
cmandel.c::main::0::9:39: error: undefined symbol: ddiv
  ^---- ddiv
cmandel.c::main::0::9:39: error: undefined symbol: dsub
  ^---- dsub
cmandel.c::main::0::9:39: error: undefined symbol: ddiv
  ^---- ddiv
cmandel.c::main::0::9:39: error: undefined symbol: dmul
  ^---- dmul
cmandel.c::main::0::9:39: error: undefined symbol: dadd
  ^---- dadd
cmandel.c::main::0::9:39: error: undefined symbol: dmul
  ^---- dmul
cmandel.c::main::0::9:39: error: undefined symbol: dadd
  ^---- dadd
cmandel.c::main::0::9:39: error: undefined symbol: dsub
  ^---- dsub
cmandel.c::main::0::9:39: error: undefined symbol: dadd
  ^---- dadd
cmandel.c::main::0::9:39: error: undefined symbol: l_f48_ldexp
  ^---- l_f48_ldexp
cmandel.c::main::0::9:39: error: undefined symbol: dmul
  ^---- dmul
cmandel.c::main::0::9:39: error: undefined symbol: dadd
  ^---- dadd
cmandel.c::main::0::9:39: error: undefined symbol: dmul
  ^---- dmul
cmandel.c::main::0::9:39: error: undefined symbol: dmul
  ^---- dmul
cmandel.c::main::0::9:39: error: undefined symbol: dadd
  ^---- dadd
cmandel.c::main::0::9:39: error: undefined symbol: dlt
  ^---- dlt
../_DEVELOPMENT/l/sccz80/9-common/f48/fa.asm:12: error: undefined symbol: init_floatpack
  ^---- init_floatpack
../_DEVELOPMENT/l/sccz80/9-common/f48/int2long_s_float.asm:23: error: undefined symbol: float
  ^---- float
../_DEVELOPMENT/l/sccz80/9-common/f48/int2long_s_float.asm:27: error: undefined symbol: float
  ^---- float
../_DEVELOPMENT/l/sccz80/9-common/f48/int2long_u_float.asm:14: error: undefined symbol: float
  ^---- float`

Xalior, 2025-07-29:
Looks like you didn’t link the mathlib.  Check the makefile in the same
folder for the correct libraries to include for each example :)

-Dx

thearduinoguy, 2025-07-29:
My bad. Thanks.

closed 2025-07-30 by suborb