z88dk/z88dk #2788: [feature][g850] Support for 4 pixel tall fonts

Fabrizio-Caruso, 2025-09-30:
I see that the similar g815 sub-target can use 4 pixel tall fonts to get 6 rows of text.

@suborb Could you please add support for this font to the g850 target in order to get 12 rows of text?

In general, it would be great to have this on all targets with screen with heights lower than 64 pixels, in order to be able to cram enough text or tiles on the screen.

Fabrizio-Caruso, 2025-10-07:
@zx70 
I get this errors if I change `-clib=g850c`:
```
...
C:/z88dk/bin/../lib/crt/classic/crt_runtime_selection.inc:379: error: undefined symbol: ansicharacter_pixelwidth
C:/z88dk/bin/../lib/crt/classic/crt_runtime_selection.inc:380: error: undefined symbol: ansifont
C:/z88dk/bin/../lib/crt/classic/crt_runtime_selection.inc:381: error: undefined symbol: ansifont_is_packed
make: *** [makefiles.common/targets/z88dk/Makefile_g800:56: g850] Error 1
```

I have tried with and without the pragmas.

zx70, 2025-10-08:
It helps a lot !

`
IF NEED_ansi4x4
	PUBLIC		fputc_cons
	EXTERN		fputc_cons_ansi
	EXTERN		puts_cons_ansi
	defc DEFINED_fputc_cons = 1
	defc DEFINED_puts_cons = 1
	defc fputc_cons = fputc_cons_ansi
	PUBLIC ansi_attr
	PUBLIC ansi_cls
	PUBLIC ansi_CHAR
	PUBLIC ansi_del_line
	PUBLIC ansi_SCROLLUP
	EXTERN __gencon_ansi_BEL
	EXTERN clg
	EXTERN __ansichar_4x4
	EXTERN ansi_del_line_generic
	EXTERN scrollup_4px
	defc ansi_attr = __gencon_ansi_BEL
	defc ansi_cls = clg
	defc ansi_CHAR = __ansichar_4x4
	defc ansi_del_line = ansi_del_line_generic
	defc ansi_SCROLLUP = scrollup_4px
	; Don't include an ANSIfont
	UNDEFINE TAR__no_ansifont
	defc TAR__no_ansifont = 1
ENDIF
`

[image]

Fabrizio-Caruso, 2025-10-08:
So if just use `clib=g850c` I get:
```
C:/z88dk/bin/../lib/crt/classic/crt_runtime_selection.inc:379: error: undefined symbol: ansicharacter_pixelwidth
C:/z88dk/bin/../lib/crt/classic/crt_runtime_selection.inc:380: error: undefined symbol: ansifont
C:/z88dk/bin/../lib/crt/classic/crt_runtime_selection.inc:381: error: undefined symbol: ansifont_is_packed
make: *** [makefiles.common/targets/z88dk/Makefile_g800:58: g850_12] Error 1
```

@suborb @zx70 I assume you are still working on it.